aegean-odyssey/mpmd_marlin_1.1.x #3: COM Port Issue

ramin3, 2020-03-29:
I flashed the mini delta with mpmd_marlin_1.1.x-119r08-SM1111-PCfan-10Alimit.bin. The reason I chose this version was because I have altered the extruder and had to flip the cable to make it work properly. Anyway, after having flashed the board with the new firmware, I can no longer connect to it via USB because it is not recognized by my Windows 10 computer. When I connect via USB, the printer now shows up as "STM32 Virtual ComPort in FS Mode" and as "USB Serial Device(COM4)". Because of this, I can no longer connect to the printer via USB using PronterFace or Repetier-Host.

aegean-odyssey, 2020-03-29:
First, thank you for explaining the SM1111 -- it immediately caught my eye.

Full disclosure: I've been "Microsoft-free" for about two decades, now, so I've virtually no experience with Windows of late. Perhaps experienced Windows users will comment as well.

Though not Windows, I've tested with Pronterface (linux) and chosing whatever serial port the machines shows up on allowed Pronterface to connect with no problems. `STM32 Virtual ComPort in FS Mode` comes directly from the firmware, so it seems the USB's intial handshaking is successful, and the correct USB "mode" is as a virtual serial comport (com1, com2, etc. on Windows). As far as PronterFace (perhaps Repetier-Host, I've zero experience) is concerned, the program should be connecting through a serial port. The upshot is: Can you select the com port and have the program (or Repetier-Host) function? -- though it says com4, it really is USB at USB speeds.

If you chose the com port (i.e. virtual com port) and Pronterface does not connect, please report back. I'll work to track down the problem.

ramin3, 2020-03-29:
COM 4 does not appear in Pronterface, so I cannot select it. Also, I have now changed to firmware version mpmd_marlin_1.1.x-119r08-SM1110-ACfan-10Alimit.bin. and the COM problem remains. I am no expert and I am relatively new to Marlin, I did notice that in the mpmd_marlin_1.1.x-master\marlin_changes\configuration.h file, the Serial Port is set to #define SERIAL_PORT 0 and the baud rate is set to #define BAUDRATE 250000. I was looking at the Marlin configuration.h website and it says "Serial port -1 is the USB emulated serial port, if available." So, perhaps this should be set to -1? Also, the baud rate may explain COM port issues others have been having? I am just guessing here and I have never done a complete Marlin configuration and compiled it before or I would have changed the serial port configuration and tried it out. 

Unrelated to the COM port issue, and further explanation for the firmware change, the very first thing I did to this printer was change the extruder to an aluminum MK8. However, the new extruder ran the filament on the bottom of the gear, rather than the top of the gear as in the original. Because I had not changed the firmware to Marlin, this required flipping the wire to the stepper motor. When I changed to Marlin, I assumed that the extruder would be running backward because I had flipped it. That is why I chose SM1111 to begin with. However, when I was running the printer and calibrating/first layer, the extruder was running backwards. So I have changed it to SM1110 now and the extruder is working properly. 

Thanks for your help and your quick response.

aegean-odyssey, 2020-03-29:
The configuration you refer to is how the serial ports are identified on the printer's controller board. It's not the designation of the ports on the computer that is running pronterface. I'd like close this issue, and "lump  it" with [Issue #5](https://github.com/aegean-odyssey/mpmd_marlin_1.1.x/issues/5), since it seems that [@CScott](https://github.com/cscott) in [Issue #5](https://github.com/aegean-odyssey/mpmd_marlin_1.1.x/issues/5) is connecting the printer via pronterface on a PC laptop, but he has identified a USB issue that clearly needs further investigation. Hopefully in the process, I'll come up with a better answer for you, as well.

closed 2020-03-29 by aegean-odyssey