aegean-odyssey/mpmd_marlin_1.1.x #4: Crashes (both hardware and software) running AUTO_CALIBRATION on 119r08

cscott, 2020-03-29:
Two crashes when running the AUTO_CALIBRATION script on mpmd_marlin_1.1.x-119r08-SM0001-ACfan-10Alimit.bin

My MPMD is an indiegogo model (ie, first or second production run).

First, if you have either of the heaters enabled (usually calibration is done with the bed and nozzle heaters enabled to allow them to thermally expand to their "while printing" sizes), the script crashes and resets the controller.

If you run the auto calibration gcode script w/o a heater enabled, it tries to move the head off the bed on the fifth point tested.

cscott, 2020-03-29:
Same thing happens with 119r07.

Video at https://www.youtube.com/watch?v=jQgvQ8DaixY

cscott, 2020-03-29:
The bed crash seems to be because the steps/mm setting is off.
Before upgrade:
>echo:Steps per unit:
>echo:  M92 X57.14 Y57.14 Z57.14 E48.50
After:
>echo:Steps per mm:
>echo:  M92 X114.29 Y114.29 Z114.29 E97.01

Perhaps the same microstepping issue discussed here:
https://mpminidelta.com/firmware/motion_controller

Things got a little more normal after I adjusted M92:
```
>>> M92 X57.14 Y57.14 Z57.14 E48.50
SENDING:M92 X57.14 Y57.14 Z57.14 E48.50
ok P63 B7
>>> M851 Z0
SENDING:M851 Z0
ok P63 B7
>>> G28
SENDING:G28
ok P63 B6
>>> G33 V3 T
SENDING:G33 V3 T
G33 Auto Calibrate
Checking... AC
.Height:130.00  Ex:+0.00  Ey:+0.00  Ez:+0.00  Radius:63.35
Error:Probing failed
[ERROR] Error:Probing failed

Correct delta settings with M665 and M666
ok P63 B6
>>> G28
SENDING:G28
ok P63 B6
>>> M666 X-0.04 Y-0.64 Z0.00
SENDING:M666 X-0.04 Y-0.64 Z0.00
ok P63 B7
>>> M665 L123.00 H133.79 R64.00 S120.00 A0.00 B0.00 C0.00 D0.00 E0.00 F0.00 X0.00 Y0.00 Z0.00
SENDING:M665 L123.00 H133.79 R64.00 S120.00 A0.00 B0.00 C0.00 D0.00 E0.00 F0.00 X0.00 Y0.00 Z0.00
ok P63 B7
>>> G33 V3 T
```
.... and that tried to calibrate, but doesn't move the head from the center, and eventually spit out a useless calibration and hung:
```
G33 Auto Calibrate
Checking... AC
.Height:133.79  Ex:-0.04  Ey:-0.64  Ez:+0.00  Radius:64.00
.      c:-0.23   x:-0.23   y:-0.23   z:-0.24
.               yz:-0.23  zx:-0.23  xy:-0.23
Iteration : 01                                std dev:0.231
.Height:-4294967295.21474836472147483647  Ex:2147483647.21474836472147483647  Ey:2147483647.21474836472147483647  Ez:-4294967295.21474836472147483647  Radius:4294967295.21474836472147483647
```

aegean-odyssey, 2020-03-29:
Wow, thank you for the all of the effort. 

It does look like something is terribly off. And as you've discovered, M92 is configured by default for the (presumedly) later models of the printer. This affects not only M92, but also M665 L and H.

It looks like the auto-calibrate failed when testing the height -- with H130 the machine "thinks" it's below Z=0 and halts the calibration. Also, L needs to be correct for your machine (auto-calibrate does not change this) -- with L128, I think the auto calibration algorithm does not converge.

Please, could you try:

```gcode
M502;  reset to factory defaults
M92 X57.14 Y57.14 Z57.14 E48.50 ; for your machine
M665 L123 H134  ; instead of factory default L128 H130
G33 V3 T  ; auto-calibrate
```

aegean-odyssey, 2020-04-28:
I wanted to spend some checking out the issues you raised here before getting back to you. We've no 1/8 micro-stepping machines to test with, but in the limited testing we have been able to do, I'm "reasonably" sure that the firmware can work with the 1/8 micro-stepping machines. 

Also, we don't have much experience with the -10Amp variants of the firmware. There is an outside chance, I think, that with steppers running plus nozzle and bed heating, the printer demands more from the power supply than it can deliver. If this is really the issue in your case, then (as a test) using the -05Amp variant of the firmware should eliminate the crashing.

I broke out the three (3) main issues of your bug report and registered them as separate issues #8, #9, and #10 to help others zero in on your observations and the possible solutions.

Again, thank you for the bug report. It has been helpful.

I'll close this issue, but please feel free to follow up, if necessary.

closed 2020-04-28 by aegean-odyssey