aegean-odyssey/mpmd_marlin_1.1.x #5: USB serial broken in 119r07 and 119r08

cscott, 2020-03-29:
I couldn't get pronterface to reliably connect to the MPMD from my machine (thinkpad T460p, SS USB port) on 119r07 or 119r08. It would appear to connect, but drop half or more of the characters sent from the mini delta.  Reverting back to 119r06 made the serial work properly (at 115200 baud).

aegean-odyssey, 2020-03-29:
(sigh) I though I nailed all of the USB issues. Thank you for your reporting. I'll try to track it down.

ramin3, 2020-04-15:
Just some more information regarding the USB / Com Port situation: Because cscott said that he was able to get his USB port working using 119r06, I figured I would give that a try, so I flashed the board with 119r06. But, it did not work for me. I still got "STM32 Virtual ComPort in FS Mode" and an "Unknown device" on Com 4. So, I flashed the board with the original firmware that this printer came with (version 44). It fixed the issue and Windows 10 now recognizes the printer as "Serial GCode 3D Printer" on Com 3. Pronterface recognizes that the printer is on Com 3 and I can now connect with Pronterface at 250000 and send and receive to the printer. I am going to try flashing 119r06 now and see what happens....

PurpleHullPeas, 2020-04-15:
Have any of you tried the Windows COM port drivers from Marlin4MPMD? https://github.com/mcheah/Marlin4MPMD/tree/master/windows%20drivers

cscott, 2020-04-15:
I'm on Linux.  Not a driver problem, in my case at least.

ramin3, 2020-04-15:
@PurpleHullPeas 

I found those drivers when doing a google search on STM32F0's HAL USB drivers, but I did not install them because I didn't want to create an account and sign up. I had no idea they were in any way related to Marlin4MPMD. I will go get them and install them now and see what happens. Thank you for the suggestion!

aegean-odyssey, 2020-04-17:
Yes, thank you @PurpleHullPeas. I'll include your tip in the wiki.

@cscott's issue(s) are a bit more insidious -- I'm still trying to track them down.

nathanielstenzel, 2021-08-07:
I am getting this issue from Pronterface. I think it might be related to the buffer size or the chip quality, but I may be wrong.

> Exception in thread Thread-2:
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/serial/serialposix.py", line 500, in read
    raise SerialException(
serial.serialutil.SerialException: device reports readiness to read but returned no data (device disconnected or multiple access on port?)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/printrun/printcore.py", line 260, in _readline
    line = self.printer.readline().decode('ascii')
  File "/usr/lib/python3/dist-packages/serial/serialposix.py", line 509, in read
    raise SerialException('read failed: {}'.format(e))
serial.serialutil.SerialException: read failed: device reports readiness to read but returned no data (device disconnected or multiple access on port?)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3.8/threading.py", line 932, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.8/threading.py", line 870, in run
    self._target(*self._args, **self._kwargs)
  File "/usr/lib/python3/dist-packages/printrun/printcore.py", line 347, in _listen
    line = self._readline()
  File "/usr/lib/python3/dist-packages/printrun/printcore.py", line 281, in _readline
    if 'Bad file descriptor' in e.args[1]:
IndexError: tuple index out of range

aegean-odyssey, 2021-08-09:
created a new issue, #80, closing this issue