aegean-odyssey/mpmd_marlin_1.1.x #6: Stepper Motor Currents

ramin3, 2020-04-08:
I was wondering how I would go about altering the stepper motor currents. I have done some research and was thinking perhaps an M907 command would work, but it didn't. I pulled the heatsink off the board to check if there were pots to alter the current (spoiler alert, there aren't) I suppose if this were regular Marlin, I would go into Configuration_adv.h and change the digipot I2C motor currents and then upload the new configuration to the board. However, for this branch of Marlin, I have no idea how to go about making such a change and then compiling and uploading the firmware. I have visual studio code and the latest version of Platform IO for VSCODE, but I don't know enough about programming to go about making the changes I would like. Any help on how to go about doing this would be great.

aegean-odyssey, 2020-04-17:
I do not think it is possible with the printer's stock controller board (mainboard).

As far as I can tell, the controller board in the Monoprice Mini Delta does NOT provide a mechanism to adjust the stepper motor currents (via software). There is no configuration on the stepper motor drivers other than issuing a reset. According to the A4988 datasheet, "The maximum value of current limiting is set by the selection of RSx and the voltage at the VREF pin." -- probably not a practical approach in your case. Also, the microstepping inputs on A4988 are hard-wired as well, meaning that features that rely on Marlin's "baby-stepping" will not work either.

The "quiet" version of the controller board supposedly uses TMC2208 drivers to control the stepper motors. This part does include a serial (uart) interface to set drive parameters, but it's not clear if the uart pins are actually connected to the board's CPU. Currently the firmware does not provision any serial connection to the stepper motor drivers.

Files HAL_stm32.(c|h) and pins_MALYAN_M300.h pretty much contain our current understanding the printer's controller board (M300-mainboard). Though the CPU is quite capable, the board itself does not seem to make use of much of the CPU's I/O.

This page offers a little info, too:  [mainboard mp mini delta](https://gigdigit.com/mainboard-mp-mini-delta/)

PurpleHullPeas, 2020-04-17:
Matthew at Gigdigit asks for stepper motor serial numbers before custom-making a board so that he can match the board hardware to the motor resistance (10 Ohm or 2 Ohm). There has been a lot of discussion about this in the Facebook Group, but it is not something I have bothered looking into too much.

ramin3, 2020-04-17:
@aegean-odyssey @PurpleHullPeas 

Thank you very much for the information on the stepper motors and the CPU. Getting this little printer up and running and printing to my satisfaction has been challenging and fun. It has been really nice that every time I run into a problem I don't know how to fix, someone out there has the answer and has been willing to help me out!. Michael Laws from Teaching Tech was right about this little printer and the community support behind it. 

I've thought about changing the main board but if I went that way I'd probably go the whole enchilada and get an SKR 1.4 Turbo, or the new SKR Mini for the E3. Of course, then I'd need some Moons stepper motors and a Mosquito hot end...upgrade the bed to a high temp bed make an enclosure so I can print PEEK.....Obviously I'd have to make some mods to the printer, but that's half the fun! For now I'm sticking with the M300-mainboard and seeing how far that takes me. 

Thanks again for your help.

closed 2020-04-18 by aegean-odyssey