aegean-odyssey/mpmd_marlin_1.1.x #7: Nozzle moves left during startup gcode

mulcmu, 2020-04-21:
Twice tonight, hot end has traveled in -x direction before starting a print.  Using the start g-code for Cura from wiki.  After probing bed, seems to home okay, but then the hot end doesn't lower to Z60, seems to wait the 4 second pause and then moves in -X direction.  I then removed power.  In both instances usb was not connected, printing from sdcard.

After power cycle same file from sdcard prints fine.

mpmd_marlin_1.1.x-119r09-SM0001-ACfan-05Alimit.bin was installed this morning.  I had about 20 hours of runtime with 119r08 before that.  Machine was originally V37.

```
; home axes, probe/adjust z-offset, and pause 4s
G29 P0
G0 X0 Y0 Z60 F3600 ; [DV: added feedrate]
G4 S4
```

aegean-odyssey, 2020-04-21:
Hmm, does sound like something is amiss. I will try to reproduce the behavior here. Just to be thorough (I don't think this is the problem), can you verify that your SD card is error-free (e.g. chkdsk, fsck, ...)?

mulcmu, 2020-04-21:
Used windows gui to check the sdcard.  It did make some repairs, but didn't give any details.  

Card is a Kingston 8GB, fat32 format.  It has been in use for a while and had lots for files from printing with the stock firmware.  I plan to reformat it and see if I can reproduce issue.

Otherwise firmware has been working great without issue.

mulcmu, 2020-04-23:
I got it figured out.  My end gcode was using G91 for relative positioning and didn't issue G90 to put back in in absolute positioning.  It was still in relative positioning when next print was started which accounts for the behavior reported.

closed 2020-04-23 by mulcmu

aegean-odyssey, 2020-04-23:
That is good news. So far, I've ran a half a dozen prints with no problem, so I'm glad you came up with an answer. FYI, feedrate is another parameter that is "remembered", which also can lead to odd, differing behavior when a print starts. Our fix was to explicitly set the feedrates in our start-up g-code.  Thank you for your diligence in tracking down a solution to this issue.