aegean-odyssey/mpmd_marlin_1.1.x #10: Calibration should be performed with a heated nozzle and bed.

aegean-odyssey, 2020-04-28:
In issue #4, @cscott comments that it is common practice to heat the nozzle and bed before performing the auto-calibration. The rationale being that whatever thermal expansion the machine experiences during printing should also present during the calibration process to improve the automated calibration results. @cscott makes a good point, here.

Of course this is perfectly reasonable and easy to do -- preheat the printer using the front panel controls. Once your desired temperatures are established, run the auto-calibrate command.

During our general testing, we've calibrated the printer with and without preheating the MP Mini Delta. Qualitatively, this does not seem to very much affect the calibration, but I hope to do more specific testing and report the results. I think it will be interesting to compare the two approaches
and put some numbers to it.

mulcmu, 2020-05-03:
I ran some tests of the G33 auto calibration on my machine.  Original firmware was V37, upgraded with adjustable binder clip bed clips and glass on thermal pad.  Gcode file created that opens txt file for output and performs ten auto calibrate cycles.  Before each cycle the default M665 radius is loaded and the tower angles are reset to 0.  Four batches were performed cold and four batches performed heated (200/55).  Mean of the cold was 63.194 and mean hot was 63.147.

[image]

Raw results and the 10x Auto Calibrate gcode:  [mpmd cal testing.zip](https://github.com/aegean-odyssey/mpmd_marlin_1.1.x/files/4571435/mpmd.cal.testing.zip)

It took about an hour to run each batch of 10.  Testing performed over 3 days with no other printing or changes made to setup.

aegean-odyssey, 2020-05-04:
Excellent work! Thank you. Looking at your results, it seems heated calibration produced a better (lower) std dev (~0.01 compared to ~0.03), but both seem quite good to me. On my machine, std dev has never been below ~0.05.  Interested in your thoughts on your test results and your experience with the printer.

 I, too, use a class bed, but lazily I attached it to the build plate with double-sided cellophone tape and print on masking tape on the 3mm glass. It is far from precision, but printing is largely trouble-free. I ran some tests using only the G29 (bed leveling command) with cold (ambient ~20&deg;C) and hot (70&deg;C) print bed. With the glass, I didn't expect much of a difference between the two tests -- pretty much confirmed. I hope to do similar testing with a machine more or less right out of the box (with mpmd_marlin_1.1.x installed).

I think your test, with the glass installed, minimizes the temperature effects on the bed itself, so your test results better capture what's going on with the machine's geometry. Again, good work.

[ Also, tickled that you used the custom M988/M989 commands to capture results. ]

mulcmu, 2020-05-04:
[CSV](https://github.com/aegean-odyssey/mpmd_marlin_1.1.x/files/4575921/out.txt) file with the extracted data.  I plotted some of the values, there seemed to be some temperature dependence trends but the ranges overlapped.  Could be other confounding factors too, like filament blobs on the glass when performing multiple repeated heated trials.

When I started this testing, I was mainly interested in how consistent the tower angle adjustments were.  Some bigger rectangular prints were coming out a bit skewed.  I noticed a trend that issuing G33 several times in a row would seem to still make some adjustments.  The std dev of the initial probe would be bigger than it was on the best probe of the prior run.

Saving data right to the card worked out pretty well unless it got canceled before reaching the M989.

see3d, 2020-07-13:
I have run tests in the past for heated vs non-heated G29.  I came to the conclusion that the bed should be heated, but not the hot end (at least not more than 100C).  The bed may warp a bit and grow with temp a little different in different positions, but the hot end is in the relative same position and should only make a constant offset.  That offset can be compensated on each print with the G29 P0 at the start of each print.  The real advantage of the G29 operation is to map out the mechanical variations in the arms vs position.  That will not change with temperature.  I also don't like doing a lot of probing with a hot nozzle onto my bed (PEI), which can get dents.  In fact, since I added an Energetic magnetic bed to my printer, I turn it upside down to the bare metal side when doing the initial G29 calibration.  Probing bare glass should not be an issue though.

aegean-odyssey, 2020-07-13:
I have to agree, @see3d. My own (albeit qualitative) experience with heated vs non-heated calibrations support your conclusions. 

I recently saw this video: [Home-Fried 3D Printing, MPMD Arm Length](https://www.youtube.com/watch?v=GYoeg-HAw0I) and was wondering if adjusting the individual arm lengths in software via the `M665 ABC` parameters would meaningfully improve the machine's calibration and bed level. If so, I'd like to find a way to automatically derive the `M665` `A`, `B`, and `C` values from results of the `G29` bed probe data. I've not had much success yet with the calculations, but I'm just getting started with it. Also, I think [@PurpleHullPeas](https://github.com/PurpleHullPeas) will have some useful insights if I make any progress.

PurpleHullPeas, 2020-07-13:
@aegean-odyssey I made the video tutorial you mentioned based off of a tutorial originally made by @see3d . He knows more about this than I do.

PurpleHullPeas, 2020-07-13:
Also, here's my latest script tutorial. I cannot remember whether or not I sent it to you on Reddit. 
https://github.com/PurpleHullPeas/MPMD-AutoBedLevel-Cal/blob/master/advanced/Advanced_Readme.md

aegean-odyssey, 2020-07-13:
I suspected the video was yours, @PurpleHullPeas. Yes, read your script tutorial a while back and again now. Your calibration results exceed Marlin's automated approach, and it's prodded me to improve the automatic calibration in the mpmd_marlin_1.1.x firmware. I find that the motion on my printer, if not accurate, is quite repeatable. Calibration after calibration produces very similar "heat" maps of the bed level, but the bed (or the motion across the bed) is not all that flat. I am hoping that there is enough information in the heat map, along with assumptions about the print bed, to work backwards to determine the machine parameters that produced it.

see3d, 2020-07-13:
@aegean-odyssey I could not see a way for calibrating the arm lengths via probe dimensions.  I have spent literally years grappling with the bed level and arm length accuracy questions.  If the arm pairs are matched, then the FW can compensate for different lengths on the 3 axis.  If the arm pairs are mismatched in length, then it will but a mechanical  twist in the effector plate between the position extremes.  That twist will cause the nozzle to shift positions in the plane perpendicular to the arm motion.  Do that on all three axis and the positional accuracy can get a bit warped.  

I use a carbon paper probe test across each delta axis and the X Cartesian axis also to see just how warped the accuracy is.  Transferring these numbers to a spreadsheet or program can be used to calibrate the M665 better overall.  Of course, getting the numbers requires a good scale and a magnifying glass.

I first made a measuring fixture and adjusted the arm lengths to a similar value, then pared them up to make them closer.  The M665 is the final tweak.  

Another thing.  M666 should be done at a 25mm radius.  The mechanical motion is pretty flat for the first 25mm.  Then it deviates in a non-linear curve from inside to outside along an axis.  This keeps the bed level separate from the cup/bowel of the M665.  However, you will get a much better result on M666 and M665 if you first shim the bed to be square to the vertical smooth rods.  This can make a huge difference depending on how bad the factory assembled the printer.  I believe that it may be possible to calculate the shim values from the results of M666 at the 25mm radius.  This needs more work to verify.

If you calibrate the dimensionally accuracy first, then the cup/bowl can be adjusted without affecting the dimensional accuracy.  Just adjust the M665 R,L values in a fixed relation to each other.

I am happy to share my research, most of which has been posted in the announcements for the MPMD FB group with links to details in other posts.

aegean-odyssey, 2020-07-13:
Thank you, @see3d. I have much to learn. How did you arrive at the 25mm radius? Is it related to a machine dimension? Do the locations of three bed level switches come into play? I'm still trying to wrap my head around how errors in the printer's kinematic model manifest into nozzle position errors.

see3d, 2020-07-13:
@aegean-odyssey I have learned much, but not enough yet.  the 25mm radius was from observation of the graph of the shape of the three delta axis curves overlaid on each other.  It seems to be related to the machine design, as it is very similar across machines.  I suspect that if the low level kinematic formula had a correction vector on a per delta axis, that corrected this design fault (manufacturing accuracy really), the machines would be much easier to align.  

I varied the various M665/6 parameters and the bed shimming to see what happened (my machine has the bed shimming adjusted by 3 screws and my bed switches have been converted to top of bed so only the slightest movement of the bed is detected.  This eliminates the switch throw variable.  This picture of my spreadsheet will give you an example of what I am talking about.  I do not believe the bed switch location are a factor (except for be bed rocking without better bed hold down clips).  I have the same results from stock switches and my top of bed switches which are at different positions.
[image]

aegean-odyssey, 2020-07-14:
Excellent! I see what you're saying.

mulcmu, 2020-08-04:
> Excellent work! Thank you. Looking at your results, it seems heated calibration produced a better (lower) std dev (~0.01 compared to ~0.03), but both seem quite good to me. On my machine, std dev [...]

My typical std dev seems to have increased after leveling the bed relative to the rails to around ~0.07.  Just an interesting observation for now.  Factory alignment was off quite a bit.  Bed leveling plots seem to show one tower is still a bit high.  I used a combination of layered HVAC and Kapton tape which worked well to get the desired shim thickness.

see3d, 2020-08-04:
@mulcmu I used layered HVAC tape years ago to level my bed.  The problem with it is that it did not hold calibration because the soft aluminum tape and adhesive became dented quite easily.  I switched to using index card or paper stock for the main shim buildup, then just put one top layer of HVAC tape to hold it all in place.  My bed needed about 1 mm of shim on one tower.  Since then, I changed to a top of bed switch arrangement with screw adjustments that let me level the bed square to the rails from the top side.  It is 10 times easier now.

aegean-odyssey, 2020-08-04:
Thank you for your report.

The firmware contains an experiment to correct the effects of the tilting bed that occurs during probing. It has no effect during printing, but it comes into play whenever the probe is used. I don't think it works very well and it appears to interfere with many calibration procedures. This probe compensation can be disabled (before calibrating) by setting the debugging option, `M111 S128` (a command to do this is provided in the SD card's `/setup_gcode` folder).

I'll probably disable the probe compensation as the default, but I'm just not ready to give up on it yet. I hope to get a better understanding of how it affects the calibration procedure the way it does -- I have a suspicion that there are ways to improve the machine's `G33` calibration process.

Superficially, I would think that the `G33` calibration would work best without the bed being heated, so that the bed is not distorted due to a temperature gradient. And then perform the `G29` bed leveling with the bed heated to capture any bed distortion in the bed level mesh. A bit too simplistic perhaps, since it seems the bed leveling is used to correct for the whole host of errors regardless of their source. I'm hoping to do a "formal-ish" analysis, but my skills may not be up to the task.

see3d, 2020-08-04:
I found that the probe compensation for the bed tilt did provide a better alignment on a stock bed switch arrangement, but only if the stock bed hold down clips were replaced with adjustable clips, and the clips were adjusted to 0.24mm throw.  In the screenshot of my spreadsheet above, the box of cells marked Bed Setup contains the 3 variables needed for the probe compensation.  In the case of my current machine with top of bed switches, there is no switch throw and therefore no bed tilt, so Clip Z Shim is zero instead of 0.24mm, which disables it.

In order to get a good bed alignment, I follow these steps in the order given (for a minimally upgraded printer):
1. Adjust all belts to the same tension (check periodically).
2. Shim bed to be square to the vertical rails (all switches fully depressed but just barely)
3. Adjust bed hold down clips for 0.24mm throw (adjustable bed clips or shims)
4. Adjust M666 and M665 to get into the ballpark
5. Calibrate M665 R (optional D,E,F to get 3 arms different), for 100mm distance commanded along each tower axis
6. Calibrate M666 and M665 L,R,A,B,C,D,E,F to minimize the bed asymmetry and deviation from a flat plane.
7. Generate compensation mesh with G29 for final bed virtual flatness compensation.
8. Test print one layer spiral as final check and to adjust M851 Z (adjust closer for small parts and higher for parts with a large first layer to make part removal better)

I use G29 P0 with a heated bed and nozzle to start printing to compensate for thermal expansion.  It will hold as long as no changes are made to the machine.

mulcmu, 2020-08-04:
@see3d I'll keep an eye on the tape to see if indents over time.  I was thinking the card stock would have similar issues with compression.  I just checked and it was still square after about 2 months.  Shims were 0.60 mm and 0.46 mm from my notes.  Did you have any links/posts showing all of your MPMD mods?  I liked the extension springs for the ball joint backlash posted in one of the other issues.

aegean-odyssey, 2020-08-04:
@see3d, the probe compensation you describe -- the minimum Z value at a given X, Y of three intersecting tilted planes? With the planes' tilt computed from the 3 Bed Setup parameters? 

Such a model makes a lot of sense, but I did not get good agreement with the probe at x,y = (0,0). At the center, it appeared that the actually Z offset was closer to the full switch travel; in complete disagreement with the model.

I na&iuml;vely adjusted the model to arrive at the far from satisfying result that I call, probe compensation. You can find a description at [Probe Compensation Note](https://github.com/aegean-odyssey/mpmd_marlin_1.1.x/wiki/Probe-Compensation-Note) -- a lot of figuring for little gain.

On the upside: I learned how to get LaTeX expressions into GitHub's wiki pages.

see3d, 2020-08-04:
@mulcmu I have made extensive posts of all my mods (which have changed and evolved over time), with pointers in the announcements section of: https://www.facebook.com/groups/mpminideltaowners.  It is a private group so you would have to ask to join if you are not already a member.

see3d, 2020-08-04:
@aegean-odyssey Yes, the bed center has no tilt, but the full switch travel.  All three switches are moving at the same time.  Here is the text explanation in my spreadsheet:

> This column is the compensation to the Z for the rocking of the bed off the binder clips when probed to actuate the bed switches.  The symmetry is such that only the TZ (Y axis) and the Z axis need to be calculated by formula.  The other axis are just duplicates of the results.
> 
> The formulas use a block of 4 variable cells that contain the binder clip radius, bed switch radius, Z movement at the binder clip to actuate the switch, and any additional offset caused by the pressure of actuating 1,2, or 3 switches at the same time from different mechanical advantages of differing radii.

Since I know the position I am probing for my spreadsheet, I could customize each cell point with its own formula based on pressing on each probe point and seeing if it tilted the bed.  I am only probing along the tower axis for my main alignments.  That make the formula easy because it is just a 2 dimensional problem.  Just a teeter-totter.  For my extended bed probing, I probe half way between two towers.  Then it pushes two switches at the same time. 

With a square matrix, the probe compensation becomes more complicated for a single formula.  Exactly which switch spring is the weakest and which way the bed ends up tilting can be ambiguous, but generally follows where the mechanical advantage is best.  Things got so much easier for me when I went to the zero throw switch arrangement.  

I had been considering the possibility of adding a paperclip spring under the bed clips and adjusting throw to 0.05mm.  That way the switches would actuate with hardly more than a touch.  The bed would still have to be manually shimmed square to the vertical rails first for a good alignment, but the bed tilt being so small could be all but ignored.  I just have not done any more research on it since I came up with the top of bed switches that killed two birds with one stone (zero switch throw and adjustable squaring to the rails).

aegean-odyssey, 2020-08-04:
Thanks. Our methods are quite similar, though your approach is emminently more practical. It turns out to be simple to implement and I think accurate _if it's possible to determine the switch configurations for a given X,Y_. And there's the rub. Your method makes an intelligent choice, my method is a guess.

I don't want to give up on it just yet because if there is a good model, then it may be possible to "shim" the switches in software using another set of trim values or deduce these errors and incorporate the knowledge into the calibration routine. Probably overreaching but I'll give it a try.

I've also been pondering some kind of force-feedback in the nozzle to serve as the bed probe and eliminate the switches all together. It could use the input from the bed switches for a digital input, but I think it would be possible to multiplex an analog input via the nozzle temperature input as well. A back-burner project, since my current mission to squeeze as much as possible out of the stock printer.

see3d, 2020-08-04:
I like the idea of being able to "shim" the bed switches in the FW.  The adjustable bed hold downs are still a must to make the bed rock consistently with probing.  If the bed was 1 mm out of square to the vertical rails, the M666 needs to be set so that the center of the bed is projected up perpendicular to the flat be plane will hit the nozzle at the home position.  I am not sure how to determine that from FW.  Perhaps give up getting the center of the bed and just make it the sensors parallel to the bed, like now.  Then we lose a bit of print area from a misplaced center.  It should not be too bad though.  The biggest problem with that is we lose the symmetry of the arms for leveling the bed errors.  I found that there is distortion in the prints without having arm symmetry.  

I have also pondered the force feedback idea.  As a simple case, suppose the mounting of one side of the hot end was a live hinge, and the other side was a spring pressing it down.  Then put a couple of metal contacts at that interface so that when the force of the spring is overcome, it will break the connection.  That is sort of what I have on my three top of bed switches.  The bed switch input could be routed to the hot end.  I might do such a thing if I were to put a better hot end on my machine and had to change the mounting anyway.  Then the bed is attached solid to the frame and aligned square to the rails when mounted.  Hmmm, seems like my top of bed switches solves all those problems with less work.

PurpleHullPeas, 2020-08-05:
@mulcmu I have a more [public guide](https://www.reddit.com/r/mpminidelta/comments/bzm1s2/updated_mpmd_calibration_guide_and_faq/) that includes the vast majority of the mods/calibrations by @see3d . The bed clips are included with Thingiverse links, but the more advanced bed switch replacement points back to Facebook. Regarding the springs on the arms, I have not seen him post those to the Facebook Group, yet. This GitHub and a private message between the two of us are the only places I've seen it. 

If you are going to mess with the arms, go ahead and check how parallel they are (i.e., the spacing on the effector versus the carriages). I recently found that mine were off by about ~1mm on my more-stock second MPMD. I would like to find some time to fix that with M3 washers along with before/after photos, but physically getting to my printers seems to be harder right now because of some life things. The arms not being perfectly parallel results in increased effector/nozzle tilt as you travel farther from the center of the bed - I accidentally demonstrated this to myself with an effector replacement mod on my first MPMD.

mulcmu, 2020-08-06:
@PurpleHullPeas Your guide, videos, and script have been a good references for those of us without facebook accounts.  I don't recall seeing arm parallel check anywhere else.  I checked my arms for parallel, two had the same spacing within 0.1mm.  The other was out by 0.5mm.  Getting measurements was pretty finicky.  I managed to find a small washer to shim the one carriage that was out.

I noticed the belt tension was loose while doing this adjustment, looks like that was a more likely cause of the increase in the std dev than bed levelness.

see3d, 2020-08-06:
It would be very interesting to have a simulation that shows how much deviation there is from any commanded X,Y location caused by non-parallel arms.  Sort of like a heat map or contour map.  It is a bit beyond me to do that at present, but it could be an independent effort by a math whiz to create that.

PurpleHullPeas, 2020-08-07:
@mulcmu It is something that I only recently (a few days ago) noticed. FWIW, I measured the distances with the arms and washers removed. I.E. I measured the distance of the innermost nuts with calipers. Of course, this assumes that the ball joints on the arms are all the same diameter, which may not be a good assumption.

aegean-odyssey, 2020-08-08:
The project might be a good place to start, [OpenSCAD-Delta-Simulator](https://github.com/PRouzeau/OpenSCAD-Delta-Simulator)

OpenSCAD is a great, simple (to use) design tool.

aegean-odyssey, 2021-01-27:
Placing this issue into a topic in the Discussions area.

closed 2021-01-27 by aegean-odyssey