aegean-odyssey/mpmd_marlin_1.1.x #9: Printer goes "haywire" during auto-calibration

aegean-odyssey, 2020-04-28:
In issue #4, @cscott reports crashes, erratic motion, and non-sensical output during the printer's auto-calibration process after installing firmware. In essence, the firmware appears to be badly broken.

From the photos @cscott supplied, it appears that the settings are suspect. In our own testing, occassionally after installing the firmware, the firmware does not properly detect and correct errors in the flash memory used to store its settings (the exact cause is not clear to us, yet). Bizarre or erratic motion is a strong indication that there is a problem with the settings. The `M502` (load factory settings) command followed by the `M500` (save settings) command should get the firmware operating from a known "ground" state.

Thanks to @cscott for carefully documenting his experience with the firmware.

kw4c, 2020-04-29:
I'm having the same issue with the 10A and even 5A firmware. Nozzle and bed are off so it does not appear to be a power issue.

kw4c, 2020-04-29:
Ugh I’m so dumb. My MD is just old. Ran the M92 E48 fix. Life is good again.

aegean-odyssey, 2020-04-29:
Thank you for confirming that the "M92..." fix can work for older machines. I'm sure others will benefit from your comment.

aegean-odyssey, 2021-01-18:
moved to the FAQ

closed 2021-01-18 by aegean-odyssey