aegean-odyssey/mpmd_marlin_1.1.x #11: [ERROR] Error:Probing failed

BoredGitTechie, 2020-05-01:
Firmware: 119r09 and 119r08

When both printing and in pronterface, G29 always fails at the last probe point and then tries to print above the bed, or sometimes even on an angle. Is there any way to use the 3 point probe from the default firmware?

Here is a video of it failing: 

https://photos.google.com/share/AF1QipOpQerdXveVQ59nhlWZ6OPUsvph1YbeF6z00PXFSvpd83a3BwlIPNjzGnq16lESxA?key=alpPRFc3dk02QkxXakNrcHlfT1lGOThtdHIxTUFn

PurpleHullPeas, 2020-05-01:
Not a firmware answer, but it appears that the effector screw is colliding with the back bed clip. On stock firmware, this can happen with G29 P5. Here's a close-up video with a list of possible hardware fixes in the video description, if you want to go that route: 
https://www.youtube.com/watch?v=tC87r8OVIII

BoredGitTechie, 2020-05-01:
I was more hoping that i could change the amount of probe points so it did not collide, but i do not know how to change this to the stock probe points. The 3 probe points from stock always got me level results but stock had other issues so i cant revert back.

TLDR can i do this in software?

BoredGitTechie, 2020-05-01:
I've read the information in the description, but G29 P1 still does the same probe layout. Is this because its only found in stock? if so how do i add this probe layout into this firmware?

aegean-odyssey, 2020-05-01:
I believe @PurpleHullPeas is correct -- the bed level fails because the nozzle never makes contact with the bed at the final probe location.

@tclupper mentioned a similar problem (the nozzle is too high) and a fix:
>    As far as the problem, I found out why the nozzle was high.  It actually started to jam with the orange filament (this may be a problem withe the mini deltas.  This is the second time this has happened to me).  I took apart the extruder head and replaced the brass nozzle with one I had bought (I will probably buy more). When re-assembling it, I noticed the PTFE tube was a bit short ~7mm, which makes the nozzle end up being higher (See this link: [https://www.mpminidelta.com/_media/parts/heat_sink_ptfe_tube.jpg](https://www.mpminidelta.com/_media/parts/heat_sink_ptfe_tube.jpg)).  I ended up making a new one one my lathe: 6.1mm OD x 3.175mm ID x 8.5mm long.  It went together perfectly and set the nozzle at the correct position.  So, I think this problem is officially solved.

As a work-around, you can set the bed level points manually with `M421`. When the bed probe fails on the last point `M503` shows the settings including the "bed level mesh" -- the NaN values need to be set to a numeric value. Though only one location failed, (I3, J6), there may be a quite a few NaN values listed. These are locations in the 7x7 grid that are out of the print area.  The `G29` command usually extrapolates these values at the end of a successful leveling sequence.
```gcode
M503  ; lists settings, including the G29 probe results, NaN values need to be set
M421 I3 J6 Z<value>  ; last probed location
M421 I<index> J<index> Z<value>  ; for each NaN value
```
Also, I think `G29` works as well (home first):
```gcode
G28
M503  ; lists settings, including the G29 probe results, NaN values need to be set
G29 W I3 J6 Z<value>  ; last probed location
G29 W I<index> J<index> Z<value>  ; for each NaN value
```

aegean-odyssey, 2020-05-01:
Actually, there is a simpler work-around. The firmware includes a custom option to the `G29` command, `G29 C1` which computes a least squares fit of a plane to the _existing_ bed mesh values, and it just so happens to ignore NaN values.  So a work-around for you may be:
```gcode
G28
G29 V3  ; this will fail on the last probe location
G29 C1  ; use the existing values to compute a plane
```

>`FYI:` The latest `/setup_gcode/AUTO_CALIBRATE.gcode` includes `G29 C1` -- so though it may look like things went wrong during the calibration, the machine may have successfully calibrated. `/setup_gcode/M500_SAVE.gcode` and "home" and it's ready to go.

BoredGitTechie, 2020-05-01:
Thanks man, its 11 PM here but i'll try in the morning then close the issue.

BoredGitTechie, 2020-05-02:
Yup that did it, thanks for the help guys! :)

aegean-odyssey, 2020-05-07:
Re-titled, re-opened, and tagged for better visibility.

PurpleHullPeas, 2020-06-03:
Someone on Facebook asked, so I'm adding this link for the modified G29 command as a comment so that everything is in one spot: https://github.com/aegean-odyssey/mpmd_marlin_1.1.x/issues/15

aegean-odyssey, 2020-06-03:
Thank you, @PurpleHullPeas.

Yes, it's good to put all of the solutions under one issue. I know this issue is a nusiance for some, but I'm reluctant to change the default radius (in the firmware) from 55 to 50mm:
 
> Specs for the printer claim a build radius of 55mm, but on some printers it doesn't seem to be the case. I think the nozzle sits a bit high on these printers and allows the carriage to collide with the bed retaining pin. Though it is ugly, I prefer that the initial setup of the printer detects this situation so the user is aware that the printer is not quite up to spec.

___If___ the mpmd_marlin_1.1.x firmware manages to gather some users, I imagine we'll have alternate `AUTO_CALIBRATE.gcode` files and simple "How-to's" to provide work-arounds for these types of issues. I still hope to improve the overall calibration process.

PurpleHullPeas, 2020-06-03:
@aegean-odyssey Getting a bit off-topic, but from what I can tell, several people in the Facebook Group have started flashing it. The Marlin4MPMD GitHub now tells people to flash your port, instead.

BoredGitTechie, 2020-06-16:
Been using my printer for a while now, had to disassemble the hotend as it was blocked but its now working fine. Thanks to your help im back up and running much better than stock! Cannot thank you enough!

socialhacker, 2020-07-02:
I have an original Indiegogo version and just ran into this issue.  I can confirm that the alternate G29 command above (G29 L-50 R50 F-50 B50 V3) that reduces the build radius to 50mm worked for me.  Thank you!

aegean-odyssey, 2020-07-13:
Thank you @socialhacker for reporting your experience. All facts are helpful, and interesting that the issue can affect units back to the original Indiegogo run. Perhaps this is a common enough situation to warrant an "alternate" auto-calibration.gcode file; a version of the auto-calbration that uses the alternate `G29` command parameters. I need to ponder a bit about the best way to release new versions of the repository's supporting files when there are no changes to the actual firmware.

aegean-odyssey, 2021-01-18:
move to the FAQ

closed 2021-01-18 by aegean-odyssey

nathanielstenzel, 2021-07-21:
The screw of the effector hits the little black nub for me during G29 too. Not a gcode fix, but I found that using my glass bed fixes the issue.

M665 V is supposed to set the delta calibration radius but it does not seem to be honored by G29. Am I using the wrong parameters? This being a non-standard G29 still gets me a little confused.

PurpleHullPeas, 2021-07-21:
> The screw of the effector hits the little black nub for me during G29 too. Not a gcode fix, but I found that using my glass bed fixes the issue.
>
> M665 V is supposed to set the delta calibration [...]

M665 V controls the calibration radius of G33. For G29, you can follow the example inside CALIBRATE_25MM.gcode. The probing problem is also documented in the [FAQ](https://github.com/aegean-odyssey/mpmd_marlin_1.1.x/wiki/Frequently-Asked-Questions#the-g29-command-reports-errorprobing-failed).