aegean-odyssey/mpmd_marlin_1.1.x #13: Weird layer adhesion issue

TheDukeOfChutney, 2020-05-03:
[image]
I am not sure if this is a bug in the firmware or an operator error. I am more inclined to think operator error. I compiled a recent copy of the source and changed the thermistor, and increased the print radius. Any idea what could be the issue?

TheDukeOfChutney, 2020-05-04:
I am going to close this and make sure its not an extruder issue.

closed 2020-05-04 by TheDukeOfChutney

aegean-odyssey, 2020-05-04:
_(full disclosure: non-expert comment)_
For the project (start gcode, the test models, etc.), I set the hotend temperature to 195&deg;C: kind of on the low side. This temperature was OK for the Solutech PLA filaments I was using at the time, but the (brand name escapes me) PLA filament currently in my machine absolutely requires a higher temperature (I use 205&deg;C) to print successfully.  At the lower temperature, the new filament has "extrusion" problems and clogs the nozzle. Problems disappear at the higher temperature. So, a long explanation to say, "perhaps the nozzle temperature is too low."

Also, out of curiosity, how much larger were you able to increase the print radius?