aegean-odyssey/mpmd_marlin_1.1.x #14: M600 U Parameter doesn't seem to work

whudson, 2020-05-05:
I have been doing filament change tests over the past few days. I have noticed that when I try to specify my own U value to the M600 command that it does nothing at this step. Omitting the U value results in an unload retraction of 100mm as defined in the configuation_adv.h.

I see that the sign of my U shouldn't matter but just to be thorough I tried:
- U300.00
- U-300.00
- U300
- U-300

whudson, 2020-05-05:
I am interested in trying to build the firmware myself. Right now I can't see why this isn't working but I would like to help investigate further.

aegean-odyssey, 2020-05-05:
Hmm, the M600 code _looks_ ok, so it seems like something may be wrong with the function call,
`parser.value_axis_units(E_AXIS)`, in the (unmodified) `parser.h` file from the original Marlin source files. I'll look into the problem.

The good news (for me) is that you're the first to mention using the M600 code! Does it otherwise work? I tried as much as possible to preserve the "spirit" of Marlin's M600 with the work-arounds for the printer's limited UI options, but it's hard to tell if the result is really usable. It will be helpful to get insights from someone who uses the M600 command. 

As far development, I use the minimal linux setup described in the [wiki](https://github.com/aegean-odyssey/mpmd_marlin_1.1.x/wiki/Development). Using, `make all`, I get two firmware .bin files to do some  "testing". I let TravisCI build the myriad of variants, when I create a release.  The tools are all command line (no GUI), so it's not too difficult to setup in a virtual machine or chroot environment. On my linux machine, I use chroot to keep the development tools and an OS completely separate from linux machine's OS.

aegean-odyssey, 2020-05-06:
Well I was way off. I think the problem is with the configuration. The firmware prevents any E move that it considers a "lengthy extrude". In `Configuration.h`, we have `#define EXTRUDE_MAXLENGTH 300`. Could you confirm with a value such `U200`? And, if we've found the trouble, please suggest a better value for `EXTRUDE_MAXLENGTH` -- I'll make the change straight away.

whudson, 2020-05-06:
Yep! The 200mm unload works.  I measured the whole bowden setup, it's almost exactly 500mm on the nose. I'm aiming to have the printer spit out the whole piece of filament, so a good value for the max length would be at least 500.

If I can make another suggestion for M600, the 45 seconds delay before the cooldown starts feels too short. Even being beside the printer when it starts the change I still find myself having to tell it to warm up again while I'm in the middle of trying to jam the filament in. I think 90 seconds until turning off the nozzle would be a better amount of time to allow for the change.

aegean-odyssey, 2020-05-06:
Great feedback. I've made the changes you suggested -- also placed a note in the [wiki](https://github.com/aegean-odyssey/mpmd_marlin_1.1.x/wiki/G-M-code-Support).

Thank you for your help and for giving the firmware a try.

whudson, 2020-05-06:
That's great thank you! I'll try the new version today!

closed 2020-05-06 by whudson