aegean-odyssey/mpmd_marlin_1.1.x #15: G29 crashes into bed and returns error

whudson, 2020-05-07:
[Here is a video showing what happens. The crash happens at 1:50.](https://www.youtube.com/watch?v=vUtY4JDnoLI)

Here is my serial log.
```
09:54:05.665 : N15 G29*38
09:54:07.768 : echo:busy: processing
09:54:09.765 : echo:busy: processing
09:54:11.767 : echo:busy: processing
09:54:13.763 : echo:busy: processing
09:54:15.768 : echo:busy: processing
09:54:17.765 : echo:busy: processing
09:54:19.767 : echo:busy: processing
09:54:21.763 : echo:busy: processing
09:54:23.767 : echo:busy: processing
09:54:25.765 : echo:busy: processing
09:54:27.768 : echo:busy: processing
09:54:29.767 : echo:busy: processing
09:54:31.767 : echo:busy: processing
09:54:33.764 : echo:busy: processing
09:54:35.764 : echo:busy: processing
09:54:37.763 : echo:busy: processing
09:54:39.763 : echo:busy: processing
09:54:41.764 : echo:busy: processing
09:54:43.764 : echo:busy: processing
09:54:45.762 : echo:busy: processing
09:54:47.764 : echo:busy: processing
09:54:49.762 : echo:busy: processing
09:54:51.765 : echo:busy: processing
09:54:53.765 : echo:busy: processing
09:54:55.762 : echo:busy: processing
09:54:57.763 : echo:busy: processing
09:54:59.913 : echo:busy: processing
09:55:01.925 : echo:busy: processing
09:55:03.923 : echo:busy: processing
09:55:06.059 : echo:busy: processing
09:55:08.069 : echo:busy: processing
09:55:10.068 : echo:busy: processing
09:55:12.068 : echo:busy: processing
09:55:14.068 : echo:busy: processing
09:55:16.088 : echo:busy: processing
09:55:18.099 : echo:busy: processing
09:55:20.095 : echo:busy: processing
09:55:22.164 : echo:busy: processing
09:55:24.163 : echo:busy: processing
09:55:26.165 : echo:busy: processing
09:55:28.163 : echo:busy: processing
09:55:30.163 : echo:busy: processing
09:55:32.163 : echo:busy: processing
09:55:34.165 : echo:busy: processing
09:55:36.166 : echo:busy: processing
09:55:38.257 : echo:busy: processing
09:55:40.255 : echo:busy: processing
09:55:42.327 : echo:busy: processing
09:55:43.978 : Error:Probing failed
09:55:43.988 : X:0.00 Y:55.00 Z:1.40 E:0.00 Count X:7566 Y:7529 Z:13938
```

aegean-odyssey, 2020-05-07:
From the data, it looks like the nozzle is getting hung up before it reaches the bed. I think this issue is a duplicate of issue #11.  If not, please follow up and keep this issue open.

Though the `G29` command couldn't sample the last point, it is followed by a `G29 C1` which will ignore missing data when computing a bed level mesh. So the process does produce a valid set of calibration parameters. Just be sure to home the printer afterwards to sync up the stepper motors.

Also, a `G29 L50 R50 F50 B50 V3` should make the test grid a little smaller, though I've not tested using the `G29` command in this way.  

Specs for the printer claim a build radius of 55mm, but on some printers it doesn't seem to be the case.  I think the nozzle sits a bit high on these printers and allows the carriage to collide with the bed retaining pin. Though it is ugly, I prefer that the initial setup of the printer detects this situation so the user is aware that the printer is not quite up to spec.

whudson, 2020-05-07:
Ah, my apologies. I somehow missed that issue. I will try this, it does sound like the same problem.

whudson, 2020-05-07:
Yes that was the issue, and your suggested G29 command worked too. Thank you.

closed 2020-05-07 by whudson