aegean-odyssey/mpmd_marlin_1.1.x #16: Fans will not turn on

legoguy3632, 2020-05-20:
Hey, thanks for working on this port!

When I installed it on my MPMD, the fan for the electronics would not turn on, and when I tried to print something, the fans on the hot end would not turn on either.

If it helps, the printer is pretty old (open box buy in 2018) and came with the V43 firmware.

Thanks!

aegean-odyssey, 2020-05-20:
I believe the fans should work for you. Please bear with me as I get up to speed:

_First,_ do the fans work with the V43 firmware? I think only one fan is actually connected to the mainboard, so if _both_ do not work, I highly suspect that:
  * there is a wiring or connection issue; or
  * there is a problem with the power supply.

Check the ground and power connections, carefully.

If the fans are known to function (that is, they work with your V43 firmware), then here are some things to check:

* The electronics fan should always be on whenever the printer is powered up;

* If you are using an -AC version of the firmware, the hot-fan should automatically turn on when the nozzle temperature reaches 50&deg;C. You can test with the pre-heat buttons on the printer.

* If you are using a -PC version of the firmware, the hot-end fan is controlled via the M106/M107 commands. __OR__ you can test by connecting the USB serial port and sending the command, `M106 S80`. You can also create a file (e.g. FAN_ON.gcode), place it on the micro SD card, and "print" it -- something like the next two lines should do the trick.

```gcode
G28       ; home so we know that the file is "printing"
M106 S80  ; turn on the fan, 80% 
```

I hope this information helps. Please feel free to follow up.
Thank you for giving the firmware a try.

legoguy3632, 2020-05-21:
Thank you!  It looks like the printer was wired wrong in the factory so the part cooler and case fan were swapped.  However, this means that something is wrong with the actual case fan slot...  I'm going to keep testing, but it is strange that it only started having problems with this now.

closed 2020-05-21 by legoguy3632

aegean-odyssey, 2020-05-21:
[image]

Here's an image of the same mainboard I find in my printer (purchased Dec. 2018).

* Connector `F0` goes to the `CASE Fan`.
* Connector `F1` goes to the `HOTEND Fan`.

Yes, it does seem strange... check connections carefully. Look for solder breaks at the connector pins on the printed circuit board, and wiring problems in the connector plugs. Since the case fan is always on,  the connector has 12v on it or not -- if everything else is working, things really point to an issue with the connector or wiring. I'm sure you'll track it down.