aegean-odyssey/mpmd_marlin_1.1.x #17: Linear advance?

Jozer99, 2020-05-31:
Is linear advance enabled in your firmware?  With my other printers, I have noticed a major quality improvement in enabling it.

PurpleHullPeas, 2020-06-02:
Yes, check the latest release.

aegean-odyssey, 2020-06-02:
Yes, the firmware includes Marlin's "Linear Advance 1.5" feature as of release, 119r11.

My printer has logged "tens" of hours printing with the linear advance feature enabled (`M700 Kn`; where n > 0), but I've yet to do the proper [tuning](https://marlinfw.org/tools/lin_advance/k-factor.html) to find a "K" value for the filament currently in the printer.  Motion with linear advance seems to have a more "laid-back" feel -- it's not really slower; but rather more "relaxed".

`FYI:` I cleaned up the planner and stepper code a little bit while getting the linear advance feature to work. One tiny consequence is that a "K" value of less than 0.01 is considered zero and disables linear advance. Another consequence is that extra computational load required for the feature forced me to reduce the default `M665 S500` (segments per second value) to `M665 S200`. So the absurdly high (by Marlin standards) 500 value is now a very high (by Marlin standards) 200 value. A higher value may be possible, I've not done any testing to find the limit.

I'm working now to update the [PrusaSlicer settings](https://github.com/aegean-odyssey/PrusaSlicer-settings) to take advantage of the firmware's linear advance feature. The filament profiles in the current settings do not enable linear advance but can be easily changed to do so.

Thank you for giving the firmware a try. I am certainly interested in your observations or experience with the linear advance feature. I've seen encouraging results here.

Jozer99, 2020-06-03:
I'm running a board modified with the TMC silent stepper drivers.  I've heard these can have problems with linear advance.  Do you have any test experience with this combination?

PurpleHullPeas, 2020-06-03:
Depending on when you bought your silent board, you may or may not have the TMC on the extruder. It should be pretty obvious by the sound whether or not yours has it. The feature was just released 16 days ago, so you will likely be the first person to test that combination if you have the TMC on the extruder.

aegean-odyssey, 2020-06-03:
I don't know for sure, but I _think_ it should work. The main board does not offer a way to configure the TMC driver (no serial connection to the stepper driver), so the TMC stepper driver operates in its default , that is "legacy", mode. As far as I can tell, the issue with the TMC stepper driver is related to __not__ meeting minimum pulse width requirements for the driver in the stepper motor interrupt. The mpmd_marlin_1.1.x firmware is configured to guarantee a minimum pulse width, so hopefully the TMC stepper drivers will function properly.

Please report your findings. I am sure your experience will be helpful to others.

see3d, 2020-07-19:
I just ran my first simple part after enabling linear advance.  The printer certainly behaves different with it (M900 K2.0 for PETG).  The part was very simple and included a 3 layer base and a vase mode cone on top of it.  So there was not a lot of features to zip around (start simple first, right?).  What I noticed is that the printer acceleration was a lot slower.  It did more accelerating and deceleration than the print needed, since it was in vase mode making a circle.  It did not "jerk" the printer around as much, so it will probably last longer.  Prints will take a little longer.  On close inspection, there were some small blobs on the base top surface that were not on the LA print.  I don't plan on turning it off.  I can take a little slowdown for less work in manual cleanup after the fact.  We will see how it goes over time.  PLA would not need such a high K number, so it might not have as big an effect on speed.  PETG is quite viscous so it is not fast to change pressures.

[image]

aegean-odyssey, 2020-07-21:
Great! Thanks for reporting on the Linear Advance 1.5 feature.

According to Marlin documentation, a K value of 2.0 is at the high end for a K value. Did you go through the [calibration procedure](https://marlinfw.org/docs/features/lin_advance.html) suggested by the Marlin folk to find this value?

I've been printing PLA with K values between 0.6 and 1.2, though I've not done a "true" LA calibration for the filament. With LA, I've noticed sharper corners and more uniform top layers particularly with long extruding runs. So, so far, linear advance looks like a good thing.

see3d, 2020-07-21:
I did go through the cal process with the generated gcode test print.  Don't forget the PETG is a lot more viscous than PLA, so It will need more pressure and will lag more.

aegean-odyssey, 2020-07-21:
Thanks. Good information.

see3d, 2020-07-25:
In playing with the K parameter on some prints, it seemed like my print speed commands were ignored and the print speed was determined solely by the K value.  Higher K = lower print speed.  I expect this is because the extruder could not keep up with a faster X,Y print speed, so it slows everything down.  It did seem erratic though.  When printing a circular vase like print as above, some of the circle was very fast and some would slow down to a crawl each time around.  The extruder was advancing and reversing in a way that did not seem to match to a rather continuous filament flow part.  It seemed that it should all have worked at the same speed.  Perhaps the problem is lost in the details of how the gcode came out of Cura.  However, the print looked good, so the speed of the extruder was matching the speed of the motion.

aegean-odyssey, 2020-07-31:
Interesting. I think Linear Advance is a good thing, and worth getting it to work properly in the firmware. With the latest release, `119r12`, out today, I think Linear Advance is working all around.

aegean-odyssey, 2020-07-31:
Tagging this with `research`. Hopefully any "Linear Advance" info, know-how, tips, and discoveries wil find their way into this issue.

see3d, 2020-07-31:
@aegean-odyssey did anything change in the Linear Advance code from 119r11 to 119r12?

aegean-odyssey, 2020-07-31:
Other than not crashing :) there is no change to the Linear Advance 1.5 feature

There are two changes that involve the stepper motor interrupt:

* Tiny change to reduce the pulses on the E direction output, nothing that would be noticeable to the end user. The E direction output is now only updated when there is a transition on the output. This is a "backport" patch from recent Marlin 2.0 changes. Very simple, very small change.

* I've never be happy with the endstop interrupt. In Marlin, it has the ability to "re-enter" itself, and I don't think the endstop code, specifically endstop.update(), can safely operate in that fashion. I changed the code to prevent this from happening by never calling endstop.update() directly -- it is now called by triggering the endstop interrupt. There is one place in the stepper interrupt where I disable the endstop interrupt briefly while it directly calls endstop.update(). These are small changes that give guaranteed behavior to the endstop interrupt.

mcheah, 2020-08-13:
@aegean-odyssey, just curious how well your implementation of LIN_ADV works.  Last year, I spent the better part of 2 months trying to adapt it for my firmware and found the Marlin 2.0 implementation of LIN_ADV to be particurly lacking for deltas.  I think the issue at the time was how they treated the individual segments of the delta move as individual block instructions, which would cause the LIN_ADV calculation to constantly be spamming crazy fast accelerations and decelerations in the extruder motor.  There were also some conditions with high K_factor (or whatever the equivalent setting for 1.5) where after certain moves, the extruder would actually _retract_ at the end of the move because it had accumulated too many ADV_STEPS.

I never got a great consensus on how well it was supposed to work for deltas partially because I never had it working correctly on a cartesian printer.  Just curious if you had seen these types of issues and if so any thoughts on how you'd implement a fix.

aegean-odyssey, 2020-08-13:
You are way ahead of me on the topic. As far as I can tell, linear advance _seems_ to make a difference. I notice it in corners where they appear to be a bit sharper, and I notice the effect on top layers where the extrusion seems more uniform. The overall motion has kind of a laid-back feel instead of the more bang-bang motion with LIN_ADVANCED disabled. As for the implementation, with the tiny backport-type changes I made to Marlin's 1.1.x's planner and stepper code, the mpmd_marlin_1.1.x firmware's linear advance is pretty much the Marlin 2.0 code. I made a few changes to remove some unnecessary code in the planner and stepper files, but I doubt the changes would improve performance -- my motivation was to reduce the compiled size. Not my doing, but the linear advance code does seem to check and cap accelerations -- perhaps this is a change since you worked on it.

@see3d has been using K values successfully that according to Marlin are on the high end of the range.  Honestly I don't know if linear advance is a good thing or bad thing for the MP Mini Delta. I started to do a little testing, but got distracted with bed leveling and calibration ideas for the firmware.

If you are using the linear advance feature, please report your findings and opinion -- you are uniquely qualified, knowing the machine and its firmware inside and out.

see3d, 2020-08-13:
@mcheah I have been using the 1.1.x LA on my prints for a month or so now and 15 hours a day for the last week.  The printer is much better behaved and the print quality is vastly improved.  Instead of the extruder motor being the slave to the axis motors, they become the slave to the extruder capability.  I can play with print speeds or M220 all I want and they are basically ignored.  Jerk is irrelevant too.  The higher the M900 K value the longer the print time.  So, the extruder speed needs to be set to something that it can actually maintain.  I am running my PETG at a higher temp (245C) so that my stock extruder can push it out faster.  I can see the value in switching to a better extruder with LA.  It is after all the limiting factor.  I started with a K2.5 and a lower temperature.  I noticed some fine strings at random places.  Iplayes with temps and K values using the fine strings as my guide.  I have settled on the 245C and K1.2 as the values that result in the least strings and fastest print time.  

While running a part that I am printing right now in production mode (I have also run it in M4MPMD), The printer motion no longer jerks around the sharp corners.  It has (as the Marlin LA creator says), a more relaxed movement.  The printer sounds like it will last a long time, rather than beating itself to death.  The tradeoff is it takes longer to print because the corners are taken slower, but the print quality goes up and manual post processing is reduced.

Watching the extruder motor run does not seem to make sense to me compared to the motion.  That is because it leads the motion by a different amount, depending on the print particulars.  

I am still trying to understand the nobs that I can turn and what they will affect.  The Cura slicer settings are another mystery.  I have to play with some settings like equalize filament flow and other to get what I want.  Of course, coasting must be off.  Perhaps someone has compiles all the proper settings already and explained them somewhere.  I have not gone looking for them yet.

My Part:
[image]

mcheah, 2020-08-13:
So i have no doubt that LA is doing something, I'm just not sure it does what it's supposed to for deltas.  In my case i was running a much bigger Delta with a 1:5 geared extruder so my K values were higher and the penalty for jerky extruder motions was much more likely to result in skipped steps.  

The test i came up with was to run a long slow movement at a Z height of 50mm across the xy plane.  Set segments per second to 1 and watch the extruder wheel.  What you'd typically observe is the extruder would max jam real quick to prime the nozzle at the start of the move then it will effectively coast towards the end of the move when it starts to decelerate.  When you increase the segments per second and repeat you'd get different behavior because the planner is effectively going through a bunch of different trapezoids and accelerating / decelerating, which throws off the LA planner.  At the very end the LA step count would be off and you'd actually get a retraction on the extruder.

Was just curious if anyone else observed this or if what I'm saying even makes sense.

aegean-odyssey, 2020-08-14:
I see your point. With linear advance on a delta printer, it does seem like the extruder would be tracking the wrong coordinate system, that is the tower positions, rather than the cartesian XYZ coordinates. Looking at the code, though, I don't think this is the case. It appears that the planner computes extruder corrections based on the move's distance (or the distance of the segments of the move) in the cartesian coordinate space. Segments are computed by uniformly "chopping up" the move in XYZ space as well. So as far as I can tell, the linear advance calculation would be the same for a segmented move on a delta printer as for a segmented move on an XYZ printer. If one works, then the other should too.

The chief difference between the two is that a segmented line on a cartesian printer is equivalent to the original line; on a delta printer it is only an approximation. As to the ramifications for linear advance, I haven't a clue. Another thing to look into.

Thank you for sharing your experience. I think your "slow, straight line" test is the test to demonstrate whether or not the linear advance feature functions as advertised. With mpmd_marlin_1.1.x firmware, such a test could be run with no filament installed and all heaters turned off. It's something I'll probably try.

mcheah, 2020-08-14:
To make it easier to observe I drastically reduced the stepper interrupt frequency by altering the hardware timer.  You want the move to be running at normal feedrates so that the jerk/acceleration behavior is the same, and you don't want to disturb the step counts, but you want it to move slower than real time so you can actually watch the extruder behavior.

PurpleHullPeas, 2020-10-22:
> al process with the generated gcode test print. Don't forget the PETG is a lot more viscous than PLA, so It will need more pressure and will lag more.

@see3d Did you have to do anything special with the generated gcode from [this page](https://marlinfw.org/tools/lin_advance/k-factor.html)? I tried it after adding a P0 after G29 and it never starts printing for me. 

I'm about to either try some Cura plugins or just do trial-and-error. From a tutorial standpoint, that might be better so that the test can be done on a raft for people with imperfect bed leveling or really small nozzles.

see3d, 2020-10-22:
@PurpleHullPeas I don't recall having to do anything special other than supplying my startup gcode for it.  I found that the trial and error method for K was better to determine the quality vs print speed tradeoffs.

PurpleHullPeas, 2020-10-22:
@see3d False alarm. My partial colorblindness just made it too difficult to distinguish between the filament color vs the build plate color with so little filament being extruded. Thanks.

aegean-odyssey, 2020-10-23:
FYI for those using PrusaSlicer, linear advance settings can easily be kept with the filament parameters by using the filament start and end g-code settings:

* _filament start g-code_ `M900 K<k value here>  ; turn on LA`
* _filament end g-code_  `M900 K0 ; turn off LA`

@PurpleHullPeas, you bring up a very good point:
>I'm about to either try some Cura plugins or just do trial-and-error. From a tutorial standpoint, that might be better so that the test can be done on a raft for people with imperfect bed leveling or really small nozzles.

Here, my machine's very rough bed leveling makes it difficult to arrive at a suitable linear advance K value using the Marlin-style test pattern. Printing a tower (similar to a temperature tower) would take longer, but could be a work-around the imperfect bed level. This [(linear advance) calibration tower](http://projects.ttlexceeded.com/_images/3dprinting_techniques_linear_advance_calibration.jpg) from [this article](http://projects.ttlexceeded.com/3dprinting_techniques_calibrating_filament.html) seems to show the linear advance effects nicely. I like the raft idea as well.

aegean-odyssey, 2021-01-27:
Closing and moving this issue into the Discussions area.

closed 2021-01-27 by aegean-odyssey