aegean-odyssey/mpmd_marlin_1.1.x #18: Easiest way to customize firmware?

Ompapamowmow, 2020-06-03:
Hello,
I would like to add some of my configuration data into the firmware so that I can take it out of my start gcode. I've cloned the project in VScode and it looks like the files in the "marlin" folder are default marlin files and all the changes you have made are located in the "marlin changes" folder. I was tempted to just replace the default files with the files from the marlin changes folder, but I am not sure what changes need to be made to account for stepper setup, PS amperage, and preferred fan operation. Any advice would be appreciated.

aegean-odyssey, 2020-06-03:
> A little note: most "customizing-type" settings can be saved with the M500 command, so it may not be necessary to compile the firmware. 

I think you've got the gist of how the firmware is built. I can't speak to VScode, but look at the __`Makefile`__; it documents the entire build process. Here are the highlights:

###### preparation
* original Marlin source files and the STM32 files are unchanged;
* Marlin files replaced or altered can be found in `marlin_changes`;
* altered, replaced, or supporting (STM32) SDK files can be found in `stm32f0xx`;
* _make_ creates a directory, `Marlin`, and copies the Marlin source to the directory;
* it then copies the files in `marlin_changes` to `Marlin`, so `Marlin` is the source directory;
* _make_ creates `build` as the place to actually compile code.  

###### building
* _make_ builds variants by passing -DMAKE_... symbols to the compiler (see the section, `BUILD VARIANTS` in the `Makefile`);
* _make_ creates a list of object files to build by collecting a list of sources and then ___excluding___ files (e.g. template files, replaced files, etc.) that should not be part of the build (see the section, `SOURCES` in the `Makefile`); 
* the search path for _include_ files is defined in the `INCLUDE FILE SEARCH PATH` section;
* linker and compiler build flags, along with paths to the build tools are defined in `BUILD FLAGS` and `BUILD TOOLS` sections, respectively;
* _make_ uses fairly typical build rules to compile and link the program.

Poring over the intricacies of `make` isn't necessary, but understanding what source and include files are part of the build, the MAKE_... symbols to control which features are compiled, and the compiler/ linker flags (more on flags below) will help you get set-up in VScode. 

The compiler and linker flags in the `Makefile` instruct the compiler and linker to aggressively optimize the code. Most of this occurs in the linker and reduces the code size significantly. It is the only way I found to fit the firmware in the 118k (128k - 8k for the bootloader, and -2k to "fake" an eeprom) of available flash program space.

There's a few notes concerning the linux build system used on this end at the wiki, [Development](https://github.com/aegean-odyssey/mpmd_marlin_1.1.x/wiki/Development). This information may be helpful as well.

aegean-odyssey, 2020-06-03:
___ALSO___,  for configuration and small changes to the source, it may be simpler to fork this repository and set-up your own travis-ci build. The travis-ci setup for this project is quite straightforward -- it would not be difficult to change `.travis.yml` to "deploy" on any tagged commit. Sorry I missed this suggestion in my first comment.

Jaxtheripper, 2020-06-14:
Sorry to post here, completely off topic but couldnt find an easy way to let you know, just flashed the firmware and my machine required the SM1111 version. Not sure if its of any use to you, just noticed the (we've not seen this, yet) and thought i'd let you know they exist.

aegean-odyssey, 2020-06-15:
Thank you for reporting your experience! I've updated the wiki with your findings.

TVRepairman, 2020-07-22:
Just in case it's helpful, turns out my machine is an SM1111 version too (I've installed SM1110, just went to print and the filament is coming out of the machine rather than into it!). Purchased from Monoprice Europe end of Nov '19.

aegean-odyssey, 2020-07-22:
Yes, facts are useful. Thank you for reporting, and for giving the firmware a try.

It's odd to me that Monoprice would wire the printers differently. It just seems simpler on so many levels to build the printers in a consistent manner.

aegean-odyssey, 2021-01-18:
More of an FYI here:

As part of an experiment to investigate issue #35, I've created a "github action", `Build`, that can compile the firmware in a one-off fashion. You should be able to use it directly from a forked repository to build `mpmd_marlin_1.1.x`. In this way, git and the text editor of your choice should be all that is required for development. For small changes to the source code, a web browser alone may be sufficient.

aegean-odyssey, 2021-01-27:
Closing the issue -- converting and moving it to the Discussions area

closed 2021-01-27 by aegean-odyssey