aegean-odyssey/mpmd_marlin_1.1.x #19: Hot end carriage colliding with rear bed retainer during levelling

bakedpanda, 2020-06-04:
I'm having some trouble with this firmware (mpmd_marlin_1.1.x-119r11-SM1110-PCfan-10Alimit) where the carriage is colliding with the rear bed clip during levelling. Would you be able to compile a release with a smaller probing grid?

My printer is using the the stock parts. 

Thanks

PurpleHullPeas, 2020-06-04:
This is a repeat of https://github.com/aegean-odyssey/mpmd_marlin_1.1.x/issues/11 .
There are fixes and workaround options in the thread, above. The collision is due to false advertising. Monoprice/Malyan advertises a 110 mm build plate diameter, but designed the stock printer in such a way that it is impossible without modifications (such as a glass bed or alternate bed clip design).

aegean-odyssey, 2020-06-05:
Yes, it does seem that some printers do not meet the manufacturer's own spec. Here, we've seen the issue on one of four printers _(so far)_.  Probably the easiest and quickest solution for you is to change the `AUTO_CALIBRATE.gcode` file in the `setup_gcode` directory of the micro SD card contents. It's a simple change that will direct the `G29` command to shrink the grid that it uses to generate a bed level mesh.

With a text editor, change the line
`G29 V3`
to
`G29 L-50 R50 F-50 B50 V3` 

Thank you for giving the firmware a try.

I'll transfer this duplicate issue to issue #11.

closed 2020-06-05 by aegean-odyssey