aegean-odyssey/mpmd_marlin_1.1.x #21: G29 Parabolic Post-Processing

PurpleHullPeas, 2020-07-14:
Feature Request: It was discussed briefly on Reddit, but I would like an option to perform a parabolic fit to the G29 mesh values via G29 C2 or something like that. 

I have found that this works well on my WhamBam surface. I suspect it will also work well on stock surfaces. I hypothesize that this works well because the bed heater being located in the center causes the metal to warp a bit upwards in that location. For the same reason, it may not be the optimal choice for a glass bed. 

The parabolic fit might also be a good choice for an alternate AUTO_CALIBRATE_SMALLER.GCODE script that uses the 50mm radius calibration discussed in another issue. 

Parabolic Fit Post-Processing Script: 
https://github.com/aegean-odyssey/mpmd-calibration-experimental

My Calibration Experiments/Results: 
https://github.com/PurpleHullPeas/MPMD-AutoBedLevel-Cal/blob/PurpleHullPeas-advanced_readme/advanced/Advanced_Readme.md

aegean-odyssey, 2020-07-15:
Extending the calibration via `G29 C1`, `G29 C2`, ... was my thought early on. Unfortunately, there is not enough program space (flash memory) to include code to calculate the parabolic fit. And the parabolic fit code is quite a bit larger than the linear fit code, so even the option to replace the `G29 C1` code with `G29 C2` code is not available to us.

I am working on a simple web-based solution to process an uploaded `CALIBRAT.TXT` log file (taken from the printer's micro SD card) and produce a custom `Bed_Level.gcode` command file. This file would contain g-code commands to adjust the bed level mesh. The parabolic fit would be one of the initial (post-)processing options for the web page.

PurpleHullPeas, 2020-07-17:
Completely understandable. Thanks for your work!

closed 2020-07-17 by PurpleHullPeas

aegean-odyssey, 2020-07-21:
Early pass at some online tools to do post-processing:
[MPMD Calibration (experimental)](https://aegean-odyssey.github.io/mpmd-calibration-experimental/)

It's a first attempt at using github's pages ("static pages only") web hosting and jekyll.  Since a github pages site only serves up static pages, the original I ported the original python scripts to javascript, so they can execute "client-side" in a browser. I hope to expand the [mpmd-calibration-experimental](https://github.com/aegean-odyssey/mpmd-calibration-experimental) project to include all things "mpmd_marlin_1.1.x calibration" related.