aegean-odyssey/mpmd_marlin_1.1.x #22: Support for M205 B gcode

see3d, 2020-07-15:
I noticed that in my previous M4MPMD FW the M205 had a B20000 parameter.  The Marlin doc says this is the min segment time.  When I do a M530 with 1.1.x, it shows M205 Q20000 parameter.  Marlin doc does not mention this parameter.  Is this a new code or a typo?

aegean-odyssey, 2020-07-15:
Sigh, not a new code and not a typo. It turns out that Marlin 1.1.x indeed uses the `Q` parameter for the minimum segment time instead of the `B` parameter as the Marlin documentation states. Marlin 2.0, however, uses the `B` parameter. Go figure. 

I'll patch up the mpmd_marlin_1.1.x source to bring it in line with Marlin 2.0's version of the `M205` code. I'll mention the difference in the wiki, as well.

Thank you for using the firmware and taking the time to report the issue.

aegean-odyssey, 2020-07-31:
fixed in `119r1`

`M205 B<n>` (instead of `M205 Q<n>`) now matches Marlin 2.0

closed 2020-07-31 by aegean-odyssey