aegean-odyssey/mpmd_marlin_1.1.x #25: G33 auto bed level adjustment for M665 R,L

see3d, 2020-07-18:
Adjusting M665 R or M665 L individually will change both the X,Y dimension scaling and the bowl/cup amount.  However, dimensional scaling can be decoupled from the bowl/cup adjustments by adjusting both R and L at the same time.  For a stock MPMD, the formula are (where K = 1.5):
L(new) = L(old) + K(R(new) - R(old)) ;for adjusting R

I have only measured K for a stock MPMD.  Other delta geometries may have a different K value.

If K were a parameter, then default of K=0 would not change current L values

FYI: R(new) = R(old) + (L(new) - L(old))/K ;for adjusting L, but not used by me to date

I believe that the L (length of the arm), is the easiest to measure, so the average arm length of 6 arms is the number I would start with.  Then determine the R number that brings the dimensional accuracy across X,Y (or tower X,Y,Z), to 100mm.  Once that is done, you don't want to lose the dimensional accuracy while doing the G33 to get the bowl/cup flattened out across the bed. 

I would expect that M665 A,B,C should be preloaded with the difference in arm lengths per tower.  M665 D,E,F could start out with zero.  If a bowl/cup adjustment needed to be made on a per tower basis, then the above formula can be also used with the D,E,F values.  This probably needs to be done externally to the FW, because it is too involved and the program space is too small.

In theory, the L,A,B,C values would be measured and never change.  The R,D,E,F values could be measured or found by trial and error iteration.  The bed should come out flat and dimensionally accurate.  However, in practice this does not seem to be the case due to imperfect geometries in the printer construction.  Both L and R values need to be adjusted a little from their measured values to get the best dimensional and flatness results in practice.  There is only one best answer for L,R that gives a flat, dimensionally accurate bed level.

aegean-odyssey, 2020-07-21:
Interesting, you mention:

> Both L and R values need to be adjusted a little from their measured values to get the best dimensional and flatness results in practice. There is only one best answer for L,R that gives a flat, dimensionally accurate bed level.

Just so it's clear to me, `M665` parameters `A, B, C, D, E, F` are _trim_ values so I'd expect `M665 L100 A0 B1 C-2` to be equivalent to `M665 L98 A2 B3 C0`. Is this not true?

It seems that there should be a way to _derive_ the __K__ value from the kinematics of the printer. I hope to spend a bit more time exploring the kinematics equations after the next firmware update.

see3d, 2020-07-21:
Yes the trim values should work per your example.

The K value should be able to be mathematically derived if you had accurate mechanical measurements of a printer.  It does not help that the printer frame, arms, effector plate, rods, carriages, and ball joints have a degree of wide manufacturing tolerances and even some backlash.  It is much easier to find the value based on the easier measurement to take, which is the actual distance moved vs the distance commanded when the dome/cup is flat.  

My calibration values are L122.42, R63.04.  The ratio is 1.942.  You might expect the the almost 2:1 value and the K=1.5 might be related.  I thought that, but when I tried to make a small adjustment to bring these into alignment, my results did not track.  It might be that they should match up, but with so many manufacturing variables, and the fact that I have never achieved a truly flat contour from end to end across the plate, it makes it hard to know if my experimental results were dead on.  The experimental value which was right on K=1.5 gave the flattest bed overall.  It is easy to take it experimentally, since it is linear.  It just takes two points.  I did those tests before I discovered that the bed is pretty easy to make flat out to 25mm radius.  A new round of experimental results may look a bit different if all these things were taken into consideration.  Perhaps it is actually 1.515.  It was not very satisfying to me that my experimental result was exactly 1.5 instead of 1.515. 

However, since we start out close to the right dimensions a 1% error in K will only make a small incremental error in the dimensionality.

aegean-odyssey, 2020-07-21:
Interesting, interesting, interesting. 

It seems we're looking for L and R values in the kinematics that don't change X and Y positions; something along the lines:

For a given L<sub>0</sub>, R<sub>0</sub> find L<sub>1</sub>, R<sub>1</sub> such that:

 Y(L<sub>0</sub>, R<sub>0</sub>) = Y(L<sub>1</sub>, R<sub>1</sub>)
 X(L<sub>0</sub>, R<sub>0</sub>) = X(L<sub>1</sub>, R<sub>1</sub>)

where X(l,r) and Y(l,r) are forward kinematic equations.

Very na&iuml;ve and overly simplistic, but I think this is the general idea.

see3d, 2020-07-21:
I think it can be even simpler.  Take a single tower only.  Now you are only dealing with X.  X(L0, R0) = X(L1, R1).  Now it makes sense for individual adjustments on a per tower basis.  In my case I am actually doing this:

For a given L0, R0 find L1, R1 such that:

X1-X2(L0, R0) = X1-X2(L1, R1)

PurpleHullPeas, 2020-08-13:
I know there is a lot of ongoing discussion eleswhere regarding calibration improvements, but I would really like to see this improvement from a documentation standpoint. I.E., adding this to G33 would remove the need for my Python script and make dimensional accuracy calibration much more straightforward.

aegean-odyssey, 2021-01-27:
Closing this issue to move it into the Discussions area under the category of "A Better Calibration". There's too much good information here to leave it buried in a closed git issue.

closed 2021-01-27 by aegean-odyssey