aegean-odyssey/mpmd_marlin_1.1.x #30: G30, G29 probe method improvement request

see3d, 2020-08-11:
I have studied the repeatability of the MPMD bed probing for a few years.  I believe that there is a better way to get a more repeatable result in less time than averaging 3 probes.   My spreadsheet probes twice (t probe average each).  I average those results, then if the two probes differ, I color code the result by the magnitude of the difference.  That way the user is alerted that something was not right and can opt to rerun the whole probing pattern gcode.  

Since FW has access to all probing results and can change the probe sequence on the fly, a different approach could be used.  I have noticed that most multiple probes at the same location come up with the same result or within one count.  However, on occasion there is an outlier that messes up the average.  The two probe method which throws away the first probe and takes the second tries to address that.  However, both of these approaches are blind sequences.  An intelligent approach would give a better result than either of these.

My suggestion is this algorithm:
Take two probes.  If they match within one count, return the average result.  If not save the last probe and repeat another probe... etc.

It combines the goodness of having one probe to unstick the mechanics, getting rid of outliers and having a little averaging of close numbers.  And it will usually only take 2 probes to get a better result than 3.  

This is very easy to test to see how much it improves the variability of the probing results, just like my spreadsheet does now.

aegean-odyssey, 2020-08-11:
You "beat me to the punch" here. I'm thinking, too, that my configuration choice to average 3 samples during probing maybe adversely affecting the readings. The "two-tap, use the second tap" method is probably the better choice in Marlin. I'm hesistant to do anymore just yet than what Marlin offers, as the (1.1.x) Marlin code seems to be a minefield of side-effects. I'll configure the firmware for the "two-tap" approach, but I'll have to do a little experimenting to adust the probing speeds as well to tone down the impact of the first tap -- it looks like Marlin uses the configured probing speeds differently for each method.

see3d, 2020-08-11:
The hardware needs are the important consideration here.  MPMD stock FW uses two equal speed probes and sends back two readings as if two G30s were commanded.  I think our little machine has a little more unique bed switch arrangement than what is typically found in the industry or even DIY machines.  The Marlin FW is meant for the majority of previous designs.  

We don't want to hit the bed material fast with a hot probe.  It could damage the surface right at the point we are trying to measure.  We also don't want to hit it two slow, burning the surface.  I think both probe speeds need to be in that middle ground that causes the least damage to the bed surface.  What I proposed would give no more probes than needed to get a good result (one of the reasons for probing the G29 with a cold or only warm (100C) nozzle and a warm bed, and saving the result).  

It would be interesting to note that there would be good real time feedback of the probe consistency when more than two probes are needed to get a good reading.  That could be an early warning that the bed is getting sticky and needs cleaning or a switch is failing.  Otherwise, there is not really any way to know what the quality of the probes are.

I am assuming that the probe sequence is a call that does its thing and returns a number.  If that is true, then there should not be any side effect of making a better probe, but who knows.  I know that on occasion I have my hands someplace that affects the bed when probing (like removing errant strings).  It would be nice if the bogus results from that got rejected and redone on the fly.  Otherwise, I have to cancel and restart the print (PITA).  

I know I get more variation that I would like from the stock FW two probes or the 3 probe average.  The only way I can fix it externally is to write a stand alone program that does what I just proposed for probing (and in that case, it should just be one probe, not two).  If there were no built in G33 alignments, then it has to be done externally anyway, but with the G33, attention should be paid to the probing quality.

aegean-odyssey, 2020-08-14:
Turns out that somewhere along the line, Marlin changed the behavior of its "double-probing" feature. Instead of discarding the 1st tap and using the 2nd tap, it computes a weighted average of the two taps -- not at all what we were going for, and not as documented. 

I've rewritten the probing code to restore the documented "double-probing" behavior and reworked the conflicting `V` (verbose level)  implementations in `G29` and `G30` codes. There are still similar issues with the `G33` code.

```gcode
G29 V3 ; print "Bed X: ..."
G29 V4 ;; also print each probe

G30 V0 ; print current position when done
G30 V1 ;; also print "Bed X: ..."
G30 V2 ;; also print each probe
```

aegean-odyssey, 2021-01-27:
`G30` has been improved (at least changed), so I'll close this issue. But I'm sure it's not the final rendition -- I'll also convert the issue to a topic in the "Discussions" area so we don't lose track of the worthwhile discussion here.

closed 2021-01-27 by aegean-odyssey