aegean-odyssey/mpmd_marlin_1.1.x #31: Bed temp never reached, stays 1.2-1.8 degrees C below target

howels, 2020-08-17:
Moved to mpmd_marlin recently and have the 12V 10A PSU.  All is working great except for some odd bed-heating issues.  If I set a bed temp of 70C then my readout quickly approaches 70 but *never arrives*.  The reading will stay as 68.3 - 68.7 indefinitely.  This isn't linked to the power of my bed heater as the same thing happens if I set temp to 75, 80 or other values - the reported bed temp is always sitting 1.2-1.8 degrees below the target.  Only way to do a print is to set the bed to a high temp, wait until the temp is way above desired then start the print and wait for the bed to cool until it hits the desired range. Then it'll go back to sitting 1.5 degrees under the desired temp but the print will run.

Is there a way to configure this to not back-off the heating before hitting the desired bed temp?

Slicing with Cura, printing with Octopi.

Running latest `mpmd_marlin_1.1.x-119r13-SM0001-ACfan-10Alimit.bin`

aegean-odyssey, 2020-08-17:
I'm thinking that maybe the default PID parameters are not working so well in your situation. The machine can "auto-tune" the temperature controller; the codes of interest are `M303`, `M304`, and `M503`. The `M303 U` option is necessary to actually use (set) the values, otherwise the new auto-tune parameters are only displayed. The parameters can be set manually with the `M304` code, as well. 

The procedure is something like:
```gcode
M303 E-1 S80 U1  ;  PID auto-tune for the bed (U1 use the values, not just display)
M500             ;  save the values to flash (if it's an improvement)
```

If you find better values for the PID parameters, please report back. The default values were chosen early on after a few trials with the stock power supply, and with lower target temperatures -- not really a fit to your situation. I'm sure your findings can be useful to others.

Thank you for giving the firmware a try.

howels, 2020-08-18:
FYI I have done the auto-tune and it worked brilliantly, now temps are reached quickly.  I read up on the M303 command and increased the repetitions to 7 as I was seeing a lot of variance between each test:

`Send: M303 E-1 S80 U1 C7
Recv: PID Autotune start
[...]
Recv:  bias: 201 d: 53 min: 74.94 max: 80.18 T:27.73 /0.00 B:75.53 /0.00 @:0 B@:127
[...]
Recv:  bias: 227 d: 27 min: 75.38 max: 80.38 T:27.61 /0.00 B:79.04 /0.00 @:0 B@:127
[...]
Recv:  bias: 217 d: 37 min: 78.98 max: 80.55 Ku: 59.77 Tu: 20.81
Recv:  Classic PID
Recv:  Kp: 35.86 Ki: 3.45 Kd: 93.28
[...]
Recv:  bias: 198 d: 56 min: 79.99 max: 81.48 Ku: 95.60 Tu: 21.30
Recv:  Classic PID
Recv:  Kp: 57.36 Ki: 5.39 Kd: 152.72
[...]
Recv:  bias: 185 d: 69 min: 79.97 max: 81.57 Ku: 109.83 Tu: 13.27
Recv:  Classic PID
Recv:  Kp: 65.90 Ki: 9.93 Kd: 109.31
[...]
Recv:  bias: 180 d: 74 min: 79.84 max: 81.32 Ku: 127.12 Tu: 10.98
Recv:  Classic PID
Recv:  Kp: 76.27 Ki: 13.90 Kd: 104.65
[...]
Recv:  bias: 179 d: 75 min: 79.70 max: 81.29 Ku: 120.21 Tu: 10.32
Recv:  Classic PID
Recv:  Kp: 72.13 Ki: 13.98 Kd: 93.06
Recv: PID Autotune finished! Put the last Kp, Ki and Kd constants from below into Configuration.h
Recv: #define DEFAULT_bedKp 72.13
Recv: #define DEFAULT_bedKi 13.98
Recv: #define DEFAULT_bedKd 93.06
Recv: ok P63 B7`

Those are the values from my MPMD for an 80 degree calibration.  I've been experimenting with PETG so the higher bed temps have been needed.

closed 2020-08-18 by howels

aegean-odyssey, 2020-08-18:
Good to hear. And thanks for posting your results -- good information for others to see.