aegean-odyssey/mpmd_marlin_1.1.x #32: Improper boot without an SD card inserted

BackwardEcho, 2020-08-26:
I guess it could be called an 'improper boot'... 

119r13
I've noticed that when my SD card isn't inserted the firmware version does not flash on the screen and the LED button is a solid red when I power the printer on. There is also no temperature readout either from the bed or hotend. This is repeatable for me.

I've witnessed it reset itself after reinserting the SD card without a power cycle, though it's not immediate.

aegean-odyssey, 2020-08-27:
Not the expected behavior. 

Here the printer works fine with or without a micro SD card. Trying to list files (Print) without an SD card shows "No Files". What you describe sounds like the firmware may not have "flashed" properly. Please check the SD card for disk errors and the `firmware.bin` file and give it another try.

Once the firmware is installed there is no reason to keep the `firmware.bin` file on the SD card, and it's probably better to remove it from the card to prevent repeatedly flashing the firmware on a power cycle.

Also, it's odd that inserting the SD card has any effect -- the firmware cannot detect it, and does nothing with the card until the "Print" button is pressed. The firmware, though, does not do a whole lot of error checking with the SD card. It's possible that disk (card) corruption could cause issues; but not the "improper" boot when no card is actually in the machine.

Again, please try re-installing the firmware. And please report anything you discover; I'm sure your experience will be helpful to others.

BackwardEcho, 2020-08-27:
So I've actually used the stock SD card to flash the firmware to the printer and haven't touched it since. To print I've been using one of the recommended kingston cards that I got from amazon. It's a 16 gig card so I'm not too sure the hardware will accept it as a "firmware flasher". Though, for the heck of it I just tried a sd card extender to see if just having the slot occupied triggered something (nope) as well as a previous card that did not work on the stock firmware (SanDisk Ultra A1 HC1 Class 10 16 gigabytes). The empty extender did not force a correct boot but the previously unaccepted card did, a card that has not had the `firmware.bin` ever on it. So it seems like the hardware/firmware does some sort of check on the sd card?

I can still flash the firmware again if you'd want. Should I do anything prior (except copy my current `M503` down somewhere) to maybe the card or use one of my other 2?

aegean-odyssey, 2020-08-27:
It does look like Marlin tries to run (print) an "auto start" file from the SD card on power-up. If there a file in the root directory with the name `autoN.g` where N is a number, Marlin will open and try to print the file. You are correct, Marlin reads through the root directory on power-up. This process, though, only happens on power-up; inserting the card after power-up is not noticed by the firmware, and the card is not accessed unless explicitly instructed to do so (e.g. via sd card M codes, the print button, etc.).

So, if there is a problem with the card such a disk corruption, it _could_ cause an issue on power-up just by being in the machine. Utilities like CHKDSK (windows) or fsck (linux) will likely fix such issues. On the other hand, the machine should work without a card, regardless. If you still have an issue when the card is not in the machine, please try to re-flash the firmware, followed by `M502`, `M500` to completely initialize the machine to default settings.

BackwardEcho, 2020-08-29:
So, I'm not too sure what to think of anymore? I just rebooted my delta without a sd card installed and it booted fine. I didn't reflash yet as it hasn't broken the printer in anyway yet.

aegean-odyssey, 2020-08-29:
I'm thinking, if not a hardware problem (I don't think it is), then either a problem with the stored settings or SD card corruption (particulary in the directory structure). The firmware does not do extensive checking when _using_ the settings (pretty much all checks are done when changing a setting) or when reading from the SD card. These problems can cause the machine to behave in bizarre ways, though once corrected, everything works. Hopefully, your "ghosts in the machine" have gone.

aegean-odyssey, 2021-01-18:
Btw, does the `mpmd_marlin_1.1.x-119r15` firmware resolve this issue? Can the issue be closed?

BackwardEcho, 2021-01-18:
I shall check when I get home, but I'll just close it anyway. Sorry for leaving it open.

closed 2021-01-18 by BackwardEcho