aegean-odyssey/mpmd_marlin_1.1.x #33: Skipping custom gcode commands

BackwardEcho, 2020-08-26:
119r13

I saw there were placeholder "load / unload filament" files in the setup_gcode folder and it prompted me to make my own. It seemed to have worked at first: It would do things slowly as I didn't dial in the feed rate or not move enough of the filament. But it still worked. Now it doesn't even move the extruder before it 'finishes'.

Here is what I put:
"FILAMENT_LOAD"
`G21`
`M109 S200`
`G28`
`G1 Z10 F2500`
`M83`
`G92 E0`
`G1 E450 F2000`
`G1 E80 F250`
`G1 E-3 F250`
`G28`
`M118 {TQ\:100}{SYS\:STARTED}#`
`M118 {E\:Filament Loaded!}#`

"FILAMENT_UNLOAD"
`G21`
`M109 S200`
`G28`
`G1 Z10 F2500`
`M83`
`G92 E0`
`G1 E-560 F2000`
`G28`
`M118 {TQ\:100}{SYS\:STARTED}#`
`M118 {E\:Filament Unloaded!}#`

I liked how the `M118`s flashed text on the screen so I kept it in.

aegean-odyssey, 2020-08-27:
Your commands look like they should work. I'm not sure exactly what's going on, but here's some info:
* `G21` is not implemented -- it will generate an error message, but it shouldn't prevent anything from working properly;
* the first `M118 ...` instructs the display to show 100% done and move on to the top level screen, while the second `M118 ...` displays a message -- the sequence seemed to be necessary to get the flaky lcd display to work;
* `M400 #` will wait for all movement to complete _and_ the command buffer to empty -- it might be useful before the first `M118` code;
* the firmware does not do a whole lot of checking for SD card errors -- I suppose a read error could cause the problem (but not likely on _both_ command files).

If you come up with a solution, please report back. I'll try to reproduce the behavior here. Thanks for implementing the commands -- they look like the right idea.

BackwardEcho, 2020-08-27:
I think it might have bee the `M118`s that were messing it up. While the `M400` seemed to have worked on the "loading" file, it didn't fix the "unloading" one. For the heck of it I removed the `M118`s to see what would happen, and even though I didn't try both of them back to back numerous times (only load, unload, load) it seems to have worked.

Here is what I have:
Load
`M109 S200`        ; preheat to pla temps
`G28`                    ;home
`G1 Z10 F2500`    ; move hotend downward to straighten the bowden tube as much as possible
`M83`
`G92 E0`
`G1 E400 F2000`  ; feed most of the way
`G1 E80 F250`      ; feed slowly so filament doesn't get stuck (ideally), purge some filament
`G1 E-3 F250`       ; retract so oozing stops
`M400`
`G28`

Unload
`M109 S200`
`G28`
`G1 Z10 F2500`
`M83`
`G92 E0`
`G1 E-80 3000`      ; yoink the molten plastic out so it hopefully doesn't string itself in bowden
`G1 E-450 F2000`
`M400`
`G28`

aegean-odyssey, 2020-08-27:
Btw, the `#` character at the end of the line causes Marlin to wait until the entire command buffer is empty before reading and processing more input. The `M400` code causes Marlin to wait until "moves" have finished. I think in your case, both are needed; `M400 #`.

The `M118`s do kind of "force" an end, so if they get read and processed before the actual motions are complete, it can look like the program is "done" when it really isn't.

BackwardEcho, 2020-08-27:
That seemed to have worked. I re-added the end messages. Here are the files as copy/pasting is getting a tad annoying when trying to show both. For the unloading file, I'm not sure if going slower or faster at the start is better to prevent stringing filament back up the bowden. I might try slower at some later time or even a lower temp, but for now it's fine. 

[FILAMENT_LOAD.txt](https://github.com/aegean-odyssey/mpmd_marlin_1.1.x/files/5138329/FILAMENT_LOAD.txt)
[FILAMENT_UNLOAD.txt](https://github.com/aegean-odyssey/mpmd_marlin_1.1.x/files/5138333/FILAMENT_UNLOAD.txt)

aegean-odyssey, 2020-08-27:
Good. Please report back on your "fine-tuning" of the commands. If you don't mind, I'll work them into a future release.

BackwardEcho, 2020-08-29:
Hey... Is there a way to request user input via gcode? Such as for "do this, wait for button press, then do this other thing".

I ask because I'm interested in a "Filament Reload", combining both the Load/Unload files into one. I get that there is a way to make the printer wait a specific amount of time before continuing, which might be the better option if such a feature was desired (outside of the load/unload files).

aegean-odyssey, 2020-08-29:
Take a look at the `M0, M1` codes. These should pause a program and wait for user input. The printer's user interface is _extremely_ limited so I was forced to use the LED push button as the means for the user to signal "continue". Sending `M108` from a serial host will also signal "continue". When the printing is waiting, the LED flashes (green, I believe).

If you're looking at a process to "change the filament", take a look at the `M600` code. It is specifically designed for the purpose. At least one user has had success with it.

BackwardEcho, 2020-08-30:
While I appreciate the info about `M600`, I do not think I have any idea how to go about a filament change during a print. I was thinking of just combining the load / unload files into one so someone doesn't have to go through the menu a few more times just to change filament before a print.

aegean-odyssey, 2020-09-04:
Got it. Btw, some G/M code specifics for the printer can be found at [G/M code support](https://github.com/aegean-odyssey/mpmd_marlin_1.1.x/wiki/G-M-code-Support).

aegean-odyssey, 2021-01-18:
Looks like the main issue is resolved here, so I'll close it out. If this is not the case, please do not hesitate to re-open the issue.

closed 2021-01-18 by aegean-odyssey