aegean-odyssey/mpmd_marlin_1.1.x #34: Strange extruder results and unload/load limitations

see3d, 2020-09-03:
I just upgraded to the popular BMG extruder.  It is a double toothed wheel extruder and a 4:1 gear reduction.  My G92 is E430 now (was E100).  The extruder direction direction was correct.  It works fine from the LCD switches all the time.  Obviously the extruder motor runs over 4x the step rate as direct drive extruders, but has much higher torque.  It can print a part from the SD.

First problem is that the extruder has no way to manually release the filament.  It must be loaded and unloaded from Gcode or front panel.  My Bowden tube path is about 600mm.  There is an M603 load/unload length setting that will not allow that long a number.  The other Gcode motion commands will also not allow that size move.

Second problem is that sending extrusion G-codes via USB sometimes works, sometimes is ignored, and sometimes runs in the opposite direction.  I am still at a loss on when it decides to ignore or reverse the direction.  When reversed, both extrude and retract are reversed.  It seems that the first time I send the command from USB it works then the next ones after that are ignored.  If I run the LCD manual extrusion, it works and I can get only the next G-code extrusion command to work.  I have cold extrusion enabled and pressure Advance is enabled.

The extrusion sequence sent by the console program is: G91, G1 Exxx.xx, G90

aegean-odyssey, 2020-09-04:
Yes, the maximum length for extrude is set at 500mm -- a Marlin safety feature. I increased the value as a fix for @mhudson's issue #14. Though the limit is not entirely arbitrary, if you've a better value, I'll make the change.

Your extrusion sequence is identical to what the firmware does to support the printer's front panel, so the confusing behavior via the usb connection is puzzling. Looking through the code, things look like they should work, BUT I do notice that the `G90`/`G91` codes can/do execute while motion is queued in the printer. It doesn't _look_ like it would cause a problem, but I'm not sure. Placing `M400` (wait for motion to complete) before the `G90` and `G91` would rule out the scenario. Another possibility is intervening commands coming from somewhere -- the firmware can do this on occassion responding to events, and I noticed that Octoprint could do the same. Possible, but I think not likely as this would be noticed and more of a problem in general.

@BackwardEcho in issue #33 has put together some G-code commands to unload and load the filament. Information in the issue may provide insight into getting a load and unload process working. I plan to include @BackwardEcho's load and unload commands in the next release.

see3d, 2020-09-06:
I ran my Bowden tube through a simpler routing that took out one 90 deg bend, but added a couple of inches to the length.  In my case having a safety limit on extrusion length of 600mm would cover any Bowden arrangement that I can imagine for the MPMD.

aegean-odyssey, 2020-10-06:
No problem. Changed for the next release. Thinking of posting an "intermediate" release (with this and a few other small changes) as I've not made much headway on the larger issue of improving the autocalibration. A release should be up in a day or two.

aegean-odyssey, 2021-01-27:
Coming back around. Is this issue resolved?  In particular:

> Second problem is that sending extrusion G-codes via USB sometimes works, sometimes is ignored, and sometimes runs in the opposite direction. I am still at a loss on when it decides to ignore or [...]

Please if you think that there is something amiss in firmware, I'll check into it.