aegean-odyssey/mpmd_marlin_1.1.x #36: 2004LCD support

mcexperts, 2020-10-11:
Hi.

I looking for way to connect and support for classic LCD like 2004 or bigger, to replace standard LCD from stock and add classic marlin menu with all functions.

It is possible?

aegean-odyssey, 2020-10-12:
No, supporting a 2004-style LCD is not really possible. The 2004-style of LCD requires a very different interface than the printer's Malyan (Marlin's designation) display. The Malyan display uses a simple 3-wire tty serial interface, so a display/input device that can act like a simple terminal _could_ work with a bit of rework to the printer's firmware -- rework to remove the Marlyan code and to use Marlin's menu system. 

With some good technical info on the Malyan display board itself, I think an interesting solution is to re-design/re-program the firmware in the existing display to function as a Marlin compatible (generic serial) user interface. It's a shame that the display wasn't overhauled in this fashion in the early days of the printer's design; though I imagine there must have been a good reason or two to stick with the current clumsy implementation.

Effort/reward/cost -wise, I think adding a host on such as OctoPrint via the usb port offers more opportunities to improve the overall "MP Mini Delta" printer system. Truly working WIFI, cameras, lights, better user interfaces, ... are all possible by adopting this route to expanding and improving the printer.

aegean-odyssey, 2021-01-27:
moving to Discussions

closed 2021-01-27 by aegean-odyssey