aegean-odyssey/mpmd_marlin_1.1.x #37: two Problems : 1.Printer freezes still at the end of a print 2. Calibration out of the bed...

BjoernBjoernson, 2020-11-06:
So my MPDM has the Gigidigit Silent board and the Bed Clips and a 10 Amp power supply .

i used the : mpmd_marlin_1.1.x-119r14-SM0001-ACfan-10Alimit Firmware what works on my printer .

I jumped from your Marlin 12 to 14 and have a problem that right when the print ends the printer freezes with nozzle and bed heat still on .
Its the same end G Code like i used on your Marlin 12 Firmware : 
; mpmd_marlin_1.1.x firmware
; heaters off, home, motors off
G1 E-3 F100
M104 S0 T0
M140 S0
G1 E-3 F100
G28
M84

is there a problem with the code ?

second problem : 
during the calibration process one test point in the front left corner is out of the printing bed ... 
and over all where are the probe offset commands in the Settings folder ? 🧐
with best regards Björn

BjoernBjoernson, 2020-11-06:
so new test new gcode file same problem ... printer freezes when finish ... 
[image]
thats what printrun says when the MPDM freezes , when i try then to Cancel from the printer menu it does nothing .. the nozzle stays on the last position on the print and the heat stays on :-(

[image]

before i go back to 113 or 112 what can i also test ?

mulcmu, 2020-11-07:
Did you run the `M502_FACTORY.gcode` after installing r14?  If you still have your SETTINGS.TXT file from r12 `M503_REPORT.gcode` you can copy all the settings into a new .gcode file with M502 M500 added to beginning and M500 at the end.  

Do you get same behavior printing from SD card with usb cable removed from printer?  I've encountered some problems with erratic motion when using Pronterface.

BjoernBjoernson, 2020-11-07:
Hello 
So i have installed r14 and done a auto Calibration , nothing else , nothing from old firmware ... 

And i normally only print from SD Card ... i had done only the connection to the computer to show you whats going on in Pronterface ... but doesnt effect the freezing when the print ends , it freezes so or so ... 
ive done now a K Factor test with the ready Gcode from the K Factor test site and there the printer doesnt freeze at the end 🧐 curious 
i also tested to delete the complete end code in cura , but the printer freezes also at the end of the print 😩
so when i understand what you mean with all the copy report code stuff i will do this 🤷🏻‍♂️ but for now i dont understand ... must have a look in a quiet minute ...

BjoernBjoernson, 2020-11-07:
so i checked i have a settings file from 1.1.2020 it should be from r12 firmware . √
so how is the going ? i copy this setting file in the root directory and run a M502 ..? then it overwrites the memory with the old settings file ? 
and the save all with M500 ?
is this th correct going ?
(by the way sorry for my english i´m a german Guy from Saxony .. that makes it double hard for correct english  ;-). )

mulcmu, 2020-11-07:
Your English is way better than my German.

The release notes recommend running the M502 command which loads the factory default settings followed by M500 to save them.  This makes sure the settings stored in flash memory are all valid.  From here you could rerun the auto calibration routine like after installing r12.  My suggestion is using the calibration data previously saved using the M503 report.gcode which saves the current settings to SETTINGS.TXT.  This file has some extraneous contents that can be deleted and the "echo:" needs to be removed from the start of each line.
 
Here was a copy of the file I use to reload settings:  [cal.gcode](https://github.com/aegean-odyssey/mpmd_marlin_1.1.x/files/5505651/cal.gcode.txt).   I added some notes to the file, most of it should be copied from your settings.txt file with the settings that you want to restore.  And, yes, it will restore the old bed leveling mesh data from r12.  If this isn't desired, remove all the rows with G29.

Also, the probe offset commands were moved into a sub-folder: /setup_gcode/M851/

BjoernBjoernson, 2020-11-08:
OK i´m not sure that i checked the over all work around ... thats a little to complicated 😩
when i use a different Settiings .txt also the mesh will be copy...
but when i then make the calibration new and make a new settings.txt all the other data will be saved again from r14 version ... so isnt it senseless to do this ?🧐
i have now created a new Settings.txt from my r14 version what is installed and makes this freez at the end problem . can you see there something wrong ? what causes this failure ?
[SETTINGS.TXT](https://github.com/aegean-odyssey/mpmd_marlin_1.1.x/files/5506604/SETTINGS.TXT)

BjoernBjoernson, 2020-11-08:
sooo i flashed now the r13 and the same problem ... printer freezes after finishingprint ... still the same problem ...
here is the Settings.TXT from the r13 :
[SETTINGS.TXT](https://github.com/aegean-odyssey/mpmd_marlin_1.1.x/files/5507060/SETTINGS.TXT)

BjoernBjoernson, 2020-11-08:
And again .. i flashed now r12 and i still have the same problem 🧐 now i think its not a Firmware problem .. but what else ? 🧐

BjoernBjoernson, 2020-11-08:
Sooo News again : 
i test printed now an old Gcode file and voila it prints until the end without freezing the printer ... 

so how can i figure out whats wrong now with my new files ? wheres the failure 😩 you have any idea ? 
because a deleted the complete end code in cura and the printer freezes also at the end ... can this effect come from a cura adjustment ? 🧐

aegean-odyssey, 2020-11-09:
@BjoernBjoernson, I think we should focus on your second problem; the issue where the calibration goes askew. A few things come to mind:

a) there is a known issue #11 (with work-arounds) where the rear bed clamp can interfere with the print head during the calibration procedure. You mention the left tower (not the rear tower), but since you changed the controller board, is it possible that you've swapped the connections to those two stepper motors? I doubt it, but just something to check.

b) I suspect, as @mulcmu suggested, that the stored settings in the machine are somehow corrupt. These settings are kept in the flash memory (not on the micro SD card), and for reasons that are not clear to me, these settings are not always properly initiaized after flashing new firmware in the printer. To manually reset these settings use the M502 command (factory reset) followed by the M500 command (save settings). You can find these commands in the setup_gcode folder on the micro SD card.

c) On this end, we test (I use the term loosely) on a few printers, but not the 10A firmware. It's possible that something is wrong with the firmware, but I think I would have heard about such an issue from some of the "power users" using the firmware. Still, it could be the firmware.

Looking at your `cal.gcode` and the `SETTINGS.TXT`, though,  I can see that they are not consistent. In the cal.gcode, you use:
```gcode
M92 X57.14 Y57.14 Z57.14 E69.54
M665 L120.53 R62.35 H121.63 S100.00 V50.00 X0.00 Y0.00 Z0.00
```
so I'm guessing that you've an older 1/8 stepping printer. If this is the case, please see issue #8 for a simple extra step to reset the factory settings.

I notice, too, that your settings are quite a bit different than the default settings. Is your printer hardware not the stock hardware? If it is more or less stock hardware, please try the auto-calibration procedure with the default settings.  It may shed some light on the problem. 

I do think that your issue is a settings problem. Please try using the default settings and report back so we can sort out your issue.  Thank you for your patience. We'll figure it out.

BjoernBjoernson, 2020-11-09:
hello aegean
so ok 
first about the 1/8 micro stepping correction ...
The printer runs before this freezing issue with Marlin r12 Firmware without problems , i have never used this 1/8 stepping correction command (M92_XYZ57_E48.gcode )
and the prints were correct in measurement 👍🏻
then i swapped the board to the Gigidigit silent board and then i had also no problems with r12 Firmware , first time it freezes after print was now directly the first print after flashing the r14 firmware .

i will flash the r14 again with following process:
1. flash firmware
2. remove SD Card - delete FCUPDATE.FLG  and FIRMWARE.BIN
3. run M502 
4. run M500 
5. run M503 to save current setting s in Settings.txt file to load it up here 

will see if the first print after this has the freezing problem again ... 👍🏻

to your a) no i dont think that i have swapped the cables because the direction of the printed part on the printbed is correct like before ,
maybe i have this test probe touch out of the printbed also in the back corner but its hard to see there ... i noticed it devinately in the front left corner 

"I notice, too, that your settings are quite a bit different than the default settings. Is your printer hardware not the stock hardware? If it is more or less stock hardware, please try the auto-calibration procedure with the default settings. It may shed some light on the problem."
what settings do you mean ?
the motors are stock , only the extruder himself is a China Alloy double gear extruder , why i have to use different E Steps value ... tested out with the 100mm extrusion test 👍🏻 i´m arround 140 E steps Value to get correct 100mm Extrusion .

BjoernBjoernson, 2020-11-09:
so ive flashed the r14 again and have done it strickt like the upper post ! 
1.
2.
3.
4.
5.
 and i did a test print and it freezes again 😩
here are the settings.txt and Calibrat.txt 

[CALIBRAT.TXT](https://github.com/aegean-odyssey/mpmd_marlin_1.1.x/files/5512104/CALIBRAT.TXT)
[SETTINGS.TXT](https://github.com/aegean-odyssey/mpmd_marlin_1.1.x/files/5512105/SETTINGS.TXT)

and now ? what should i test next ?

BjoernBjoernson, 2020-11-09:
how is the effekt when i have a older 1/8 Stepping machine and use 1/16 normal firmware without Stepping adjustment ? are the measurings wrong from printed parts ?
or what is happened🧐

BjoernBjoernson, 2020-11-09:
Sooo printed now a part sliced with Simplify and used exact the same start and End G Code and voila no freezing ... the print ends normal and the Printer homes and then cooles Down .. so its a Cura Problem ... but which option can cause a printer to freeze ? 🧐

aegean-odyssey, 2020-11-09:
Thanks for testing, @BjoernBjoernson. I took a look at the files you put up and the information looks OK. I understand now about the funny extruder settings, so it makes sense.

I'm still a little troubled by the calibration glitch you describe, but I agree with you -- your machine should be working (and seems to now).

Could you please try the following ending G-code:

```gcode
G1 E-3 F100
M104 S0 T0
M140 S0
G28 
M400 #  ; wait for motion to complete and command buffer to empty
M84 
```

This change (the `M400 #`) will ensure that all commands are executed before disabling the stepper motors. It shouldn't be necessary, but there was similar quirkiness in issue #33.

Also, if you can, please send the G-code file that _does not_ work -- I would really like to figure out what the trouble is.

Thank you for all of your efforts.

BjoernBjoernson, 2020-11-09:
HA Something new ! 
After the part is printed the MPDM stops right at the moment with the Nozzle touching the part . 
so the last few prints i tried to quit the print , that causes to crash the printer completely and it does nothing ... so i switched it off and on again that the nozzle/heatbreak gets cooled down .

NOW i tested your End G Code , but i was not there when the print ends . It stops right after the last command right on the printed part like before ... BUT  after 1-2 minutes the printhead drives up and homes like normal ... so there should be any time delay after the last print command 🧐
what the hell can cause this in Cura ? 😳

BjoernBjoernson, 2020-11-09:
so now i video taped the Pronterface during Print end ... and only command came right when the printer stops was M31 (print time) , the printer halted with the nozzle right on the print for about two minutes 8during this two minutes came a lot of echo ; Busy; processing commands  and then continued end scene with homing and stoping and the last thing what pronterface noticed was print time .... 
[image]
[image]

BjoernBjoernson, 2020-11-10:
HAHA HEUREKA What the Hell i know what he is doing !!!! 
He is pulling out the filament all the time from stopping the Nozzle moving when the print is finished until he goes for homing and really ending ... i mean he pulls the Filament backwards out of the extruder :-( 

Why the Hell Cura is doing this ???? 
[image]

aegean-odyssey, 2020-11-10:
Good job. The culprit is probably in the last 50 lines or so of the G-code file. I don't use Cura, but possibly this line in the ending g-code is not working as expected:

```gcode
G1 E-3 F100
```

It looks innocent enough; move the filament a little bit. ___BUT___, I think what actually happens depends on "relative" vs "absolute" positioning mode (codes `M82`, `G90`, `G91`). Perhaps instead of a _relative_ move of -3mm, the command is attempting to return the extruder to the _absolute_ position of -3mm.

The difference between the slicing programs may be a missing `M82` or something similar.

BjoernBjoernson, 2020-11-10:
OK thats sounds possible ... 
BUT why is doing cura this also when i complete delete the end code ? 🧐

So relative  is from the actual point -.- 
and absolute is from the over all where the Extruder position was first noticed . maybe during switched on ... ?
so i should write a extra M83 right over my G1 E-3 F100 command .. will try this ..

BjoernBjoernson, 2020-11-10:
Hello together :-)
with a happy signal i can say WE ! have solved the problem Big No1 !
It was really a cura function : use relative or absolute extrusion ... its under : special functions ...
i dont know when i pushed this button 🤷🏻‍♂️ and this caused the MPDM to wait so long after finish the print that i thought it was freezed and switched it off and on again ...
So MPDM Users DONT use this Button ;-) 

Problem no.2 some  Calibration points are not on the print bed ...
So during often re flashing and calibrating i could specifie this , the r14 Calibration process has two types of meshes , the first type is ok . But at the second type nearly at the end some points are out of the bed ... then the nozzle touches the aluminium and not the print surface ... 
Maybe you can simple make the test circle 5-6mm smaller ? then we should also solve the problem that at some printers with bed clip the whole nozzle construction plate hits the bed clip holder bevore the nozzle reaches the bed ... ? 😉

i will say a big Thank YOU to this community 👍🏻👍🏻👍🏻👍🏻 great work !

BjoernBjoernson, 2020-11-10:
Hmm would it be possible to adjust the Extruder Flow rate during printing ? 🧐 like the temp and speed on the screen ...

aegean-odyssey, 2020-11-10:
@BjoernBjoernson, glad you tracked down the problem!

I believe the calibration trouble you mention is discussed in issue #11. One work-around is to use the calibration command, `CALIBRATE_25MM.gcode`, rather than the `AUTO_CALIBRATE.gcode` command. The `CALIBRATE_25MM.gcode` command uses a smaller pattern and may solve your problem. Both command files should be in the `setup_gcode` folder of the micro sd card contents.

As for making adjustments to the extruder flowrate during printing -- the firmware does allow this, but alas, the printer's user interface ("malyan ui" in marlin speak) does not. So while it is possible to control such parameters during printing via the serial port (usb), it not possible change the parameters directly at the printer. Issue #36 contains the lament.

Seems the original issue is solved, so I will close this issue. Please feel free to re-open it if the issue is not resolved. Thank you for giving the firmware a try.

closed 2020-11-10 by aegean-odyssey

BjoernBjoernson, 2020-11-10:
Ahh ok thanks for the Explanation 👍🏻