aegean-odyssey/mpmd_marlin_1.1.x #39: Flashing red light

skylersell, 2020-11-25:
I am trying to heat up my printer and it stopped at 109 and had a red flashing light on the led by the DC jack. Could it a issue with the sensor?

skylersell, 2020-11-25:
I traced the wire back up to the connector and one of the pins on the jst connector came loose and I just put in back in and that solved my issue.

closed 2020-11-25 by skylersell

aegean-odyssey, 2020-11-25:
It _could_ be the sensor (or another component) but first...

The red flashing LED indicates that the Marlin firmware entered its "emergency stop" state; that during the normal course of operation, the firmware detected a condition that made it unwise or impossible to continue. Broadly, detected anomolies in the printer's motion control or heating control cause this issue.

Most commonly, the error stems from Marlin's "thermal run-away detection" safety feature. A temperature reading that does not advance as expected is one of its triggers. I believe this is the issue you describe.

Things to consider:

* Has the printer ever worked with the firmware? In yes, then something in your setup has perhaps changed (e.g. faulty connection, faulty component, incompatible component, ...). If no, 
then maybe changing some of the heating (PID control) settings can solve the problem;

* Is it the proper firmware variant installed? Using the `-10Alimit` firmware with the stock 60 watt (5 amp) power supply will properly not work when trying to heat both the hotend and the bed together; 

* Does the printer display a message on the screen? It should give some indication as to why the firmware stopped the printer;

* Your issue could be one of common issues related to power and thermal run-away with Marlin firmware on the Mini Delta printer. More compete discussions of the issues with thermal run-away can be found in #1 and #31. In essence, adjustments are necessary to compensate for the printer's limited power supply;

* Is this a new printer (or thermistor)? If so, then it's possible that the printer is (now) using an incompatible thermistor and the firmware is not reading it properly. The thermistor type is selected at compile-time, so short of replacing the thermistor or changing and re-compiling the firmware, there is not a simple fix; and

* Is the printer connected to host software? If so, then the host software may be deliberately stopping the machine for any number of reasons (this is an issue I ran into personally).

Hopefully, this list will point to some things to try.

Please report back. I'm sure your findings can be helpful to others. I am particularly interested (concerned) that you've a new (incompatible) printer variation from the manufacturer. We're all ready building numerous variants of  `mpmd_marlin_1.1.x` firmware to accommodate the manufacturer's _same_ printer that seems to have four different stepper motor wirings.

aegean-odyssey, 2020-11-25:
@skylersell, I see you solved the problem as I was posting. Good job!

Thank you for giving the firmware a try.