aegean-odyssey/mpmd_marlin_1.1.x #41: Lower EXTRUDER_AUTO_FAN_SPEED

PurpleHullPeas, 2020-12-07:
In Configuration_adv.h, I'd like to suggest setting EXTRUDER_AUTO_FAN_SPEED to 192, like it is in Marlin4MPMD 1.3.3. 

This should help prevent overcooling on the first layer and maybe help prevent thermal runaway early in the print. My assumption is that you could still raise the fan speed to 100% via slicer settings.

I had issues with this after both upgrading my part cooling fan and trying to print PETG. I can reach 240 and run PID Autotune while homed, but when the nozzle gets close to the build surface, my temperature rapidly drops, resulting in thermal runaway. I didn't have this problem when printing PLA at 215 with the same hardware setup. I just went back to using a weaker fan for the time being. 

I may try compiling the firmware this way myself later this month, but I think I just expended the rest of my free time for the week.

aegean-odyssey, 2020-12-07:
Would setting the fan (e.g. `M106 S192`) in the start-up G-code work?. I'm not very familiar with Cura slicer, but with PrusaSlicer this works, since the slicer treats the fan as off by default. The option "no fan for first N layers" only generates G-code to turn the fan ON after N layers (it doesn't explicity turn the fan OFF), so setting a fan speed value in the start-up G-code acts as the fan's initial speed. Also, PrusaSlicer allows "per filament" G-code as well, so another option would be to keep the filament-specific fan speed G-code with the filament settings.

With the `-ACfan` variant of `mpmd_marlin_1.x.x ` firmware, only the zero value or "OFF" setting of the fan speed enables the automatic fan operation, so placing `M106 S1` in the start-up G-code essentially starts a print with the fan off, and an `M106 S0` in the end G-code returns the fan to its automatic operation when the print completes. I think this scheme can be made to work with most slicers.

aegean-odyssey, 2020-12-08:
Since the one fan also cools the hotend and serves as a part cooling fan, I've been leery about disabling or slowing the fan for the first layers as is normally done with a separate part cooling fan, concerned that "heat creep" will lead to filament jams. You have much more experience experimenting with the fan speed -- you're thoughts?

PurpleHullPeas, 2020-12-08:
You are correct that heat creep is a very common and recurring problem with many peoples' MPMD. If you want to err on the side of caution, that is a completely understandable position. I have had heat creep mitigating upgrades installed for a while now, so I may not be the best person to ask "why" 75% fan speed was deemed acceptable @mcheah's Marlin4MPMD 1.3.3 and (I assume) Malyan stock firmware.

I do think this problem is fairly specific to printing at high temperatures with part cooling near the beginning layers. Fan speed/strength and maybe even nozzle type may play a role, because I did not have the problem when using an MK8 nozzle on my MPMD.

I plan on trying your gcode suggestions sometime this week and reporting back. If I am able to get my desired results just with Start Gcode and slicer settings, I'll add a more-detailed report. I actually feel bit silly that I didn't think to try gcode before searching the source code.

Also, your continued support on this project is greatly appreciated. Lots of people in the Facebook Group are using it, even if you don't see it here.

PurpleHullPeas, 2020-12-16:
Adjusting the gcode seems to work for my situation. I feel silly for just assuming gcode fan control wasn't an option on the AC firmware. 

Just a recap, as I might link to this for troubleshooting and/or future tutorials.

### Problem Summary:
Thermal Runaway error, print halts/resets when trying to print PETG at 240 C after recently upgrading to a [higher airflow hot-end fan](https://www.thingiverse.com/thing:3019033).
PLA printed fine, and (off-topic) showed good print quality improvements.
Temperature struggles started for me around 220 C.
Usually happened within the first few layers (fan air reflecting off of the print bed back onto the nozzle).

### Side Note:
I am using Capricorn tubing inside my heat break/throat, which has a higher heat tolerance than standard PTFE tubing - i.e., I can safely print at high temperatures without damaging the heat break liner.

### Previous Mitigation Efforts:
-[Silicone Sock](https://www.thingiverse.com/thing:3501929) (stock heat block insulation is probably okay)
-[10A 12V Power Supply](https://www.mpminidelta.com/parts/power_supply)
-[MPMD Marlin 1.1.X](https://github.com/aegean-odyssey/mpmd_marlin_1.1.x/wiki/Quick-Start), 10A, ACFan (auto-controlled fan)
-[PID Autotune](https://marlinfw.org/docs/gcode/M303.html) with the fan at desired speed.

### New Mitigation Effort (Gcode):
Adjust fan speed via **both** Start Gcode ([M106](https://marlinfw.org/docs/gcode/M106.html)) and Slicer Settings. I.E., lower the fan speed for preheating and the first few layers. For this example, I will use a "safe" speed of 75%, corresponding to stock firmware and the older Marlin4MPMD 1.3.3's defaults, which I assume was built around the stock hot-end fan. 

_If your fan is faster than stock, you may need to go lower than 75% (like me), but proceed with caution because of heat creep issues._

**Cura Settings:**
Enable Print Cooling
Initial Fan Speed - 75%
Regular fan speed at height - 3mm (corresponding to common tabletop miniatures base heights)

**Start Gcode:** 
I modified [Loc-Deu's start gcode from the Wiki](https://github.com/aegean-odyssey/mpmd_marlin_1.1.x/wiki/Configuring-a-Print#startend-g-code)

> G28
> **; set and wait on hot end temperature (just below auto-fan enable)
> M104 S40 T0
> M109 S40 T0
> ; pre-set fan speed so that auto-fan is at 75% for the first layer
> M106 S192 ; fan speed between 0 to 255**
> ; set and wait on hot end temperature (Set nozzle to just below PLA melting point)
> M104 S155 T0
> M109 S155 T0
> ; set and wait on bed temperature
> M140 S55
> M190 S55
> ; set and wait on hot end temperature (Finish raising the hot-end temp to set value)
> M104 S{material_print_temperature}    ;Start heating extruder
> M140 S{material_bed_temperature}      ;Start heating bed
> M109 S{material_print_temperature}    ;Wait for extruder to reach temp before proceeding
> M190 S{material_bed_temperature}      ;Wait for bed to reach temp before proceeding
> ; home axes, probe/adjust z-offset, and pause 4s
> G29 P0
> G0 X0 Y0 Z60 F3600 ; [DV: added feedrate]
> G4 S4
> ; extrude a strip outside of the perimeter
> G92 E0
> G1 X-54 Y0 Z0.32 F2700
> G3 X0 Y-54 I54 E20 F900
> G92 E0

closed 2020-12-16 by PurpleHullPeas

aegean-odyssey, 2020-12-17:
Thanks for taking the time to write it up. It's a big help.