aegean-odyssey/mpmd_marlin_1.1.x #43: Cannot calibrate, G33 "Error:Probing failed"

e-lambert, 2020-12-14:
When performing G29 or G33, the nozzle just stops as it is about to touch the bed, not even 1mm above. The nozzle will never touch the bed, only over few mm above it. If I cheat and manually bring the nozzle closer to bed and perform G29 without homing, it starts leveling properly.

-I'm running firmware 119r14.
-I made sure to check the whole wiki and other opened issues but I don't see anything to help, I tried factory reset, correction for 1/8 stepper. 
-I checked if there was any mechanical/electrical issue but everything seems to be ok. 
-The printer has upgraded stepper drivers to TMC. Everything was running ok with stock firmware and M4MDM, no extra steps had to be done. 
-Checking at CALIBRAT.TXT, the log has ".Height:130.00 ". Maybe the step per mm is off?

aegean-odyssey, 2020-12-14:
I _think_ the machine stops because as far as it's concerned,, the probe (the nozzle) traveled to Z<0 without tripping a bed switch. The height from the `CALIBRAT.TXT` file is the default height -- it's set to a value (130) that is larger than actual height so that the probe will make contact with the bed before Z reaches zero.

As you mention, the steps/mm may be off __OR__ the default height value I chose is not quite large enough for your machine. You can try the g-code, `M665 H135`, to bump the height value up to 135mm before issuing the `G33` g-code. The `G33` command will change the height value to the actual height value.

Please report back. If the configuration (steps/mm ...) is correct for your machine and the 135 value solves the problem, I'll update the default value in the firmware. Thank you for giving the firmware a try.

e-lambert, 2020-12-15:
@aegean-odyssey yep that was it. For some reason, my step/mm is off and the printer thinks the height is 133mm. I will calibrate and try to adjust the steps per mm.

aegean-odyssey, 2020-12-15:
Good to hear that things are working. Perhaps the 133mm value is correct. If the endstops are a few millimeters on the high side (not a problem) then maybe your machine does indeed home to 133mm above the print bed. 

If you have a stock printer with the 1/8 stepper motors,  please share your `CALIBRAT.TXT` when you've found satisfactory settings. We've none of these printers here and it's entirely possible that the firmware defaults and the corrections we provide in the G-code command file are simply wrong and need to be updated.

Again, thank you for giving the firmware a try.

e-lambert, 2020-12-17:
[CALIBRAT.TXT](https://github.com/aegean-odyssey/mpmd_marlin_1.1.x/files/5711402/CALIBRAT.TXT)

I just had to change the extruder steps per mm to `M92 E102.25` but after printing MDM_Hex_Calibration_v3, the dimensional accuracy seems to be ok. I think the default steps per mm is good (sorry I said I tried 1/8 printer correction but my printer still is 1/16). I still haven't properly complete the "Dimensional Accuracy Calibration" guide in the wiki.

The ruler does show that my printer really has 133mm between bed and the nozzle when homing. I even have a 2mm thick mirror laying on the bed, that means distance would be 135mm if I had the stock bed surface. I have no clue why, I never moved the homing sensors. Anyway, `M665 H135` solved the issue, you can close the bug.

I think this firmware is a major improvement compared to stock firmware, I will would like to thank you very much for all your work and support. This is a must for every Delta Mini owner!

aegean-odyssey, 2020-12-18:
Thanks for the update. I need to think about changing the `G29` command to handle situations like yours rather than merely setting the default height to some arbitrarily large value. After the machine homes (`G28`), the height value is taken as the height of the nozzle above the bed -- the machine technically does not need to see a `G29` code to print. A high default height value and no `G29` to adjust it will likely defeat the software endstops safety feature (`M211`) which prevents the nozzle from moving outside of the print volume. The bed switches aren't used as a safety stop since they are too "twitchy" during printing. Not an urgent problem, though, since you've confirmed we have a simple enough work-around. 

Btw, I attached a pdf file of your `CALIBRAT.txt` file as viewed with the [calibrat_txt.html](https://aegean-odyssey.github.io/mpmd-calibration-experimental/calibrat_txt.html) tool. It looks ok, and shows the limitations of the automatic calibration -- another something for me to look into.

[CALIBRAT.TXT.pdf](https://github.com/aegean-odyssey/mpmd_marlin_1.1.x/files/5714313/CALIBRAT.TXT.pdf)

closed 2020-12-18 by aegean-odyssey