aegean-odyssey/mpmd_marlin_1.1.x #47: Crash when trying the 25mm calibration gcode

BackwardEcho, 2020-12-29:
I uh... don't really know what else to put aside from I'm using the 119r14-SM0001-ACfan-10Alimit variant. It crashed when I started the 25mm calibration gcode twice but worked when I started the same gcode through Pronterface via "SD" print w/e it's called.

I don't know what can be done about this but figured I'd at least mention it.

aegean-odyssey, 2020-12-29:
Thanks. It's good to keep track of these things.

Ok, not all crashes are equal.

In the order of severity, most severe to least severe:
* Did the printer crash and reboot on its own (LED never showed red)?
* Did the printer crash and reboot on its own (and the LED was solid red for a bit)?
* Did the printer crash showing a solid (not flashing) red LED?
* Did the printer crash showing a  flashing red LED?
* Did the printer simply not print, immediately showed done, etc?

The two crash and reboot on its own are subtly different, and often it's hard to "catch it in the act" to determined which one (red LED or no) occurred. A short pause and red LED before reboot means that the computer actually trapped the error. In either case, something seriously wrong occurred.

The other types of crashes result mostly from conditions the firmware can detect, but cannot do much about.

Recently, my machine stopped dead in the middle of printing the second layer on a print. The display flashed, but motion halted and the bed and hotend heaters stayed on. I tried several times, I thought the printer was broken. Turns out, I was sorting the files on the SD card with fatsort, a program that alters the directory structure on the card. It's a simple way to have the printer show the files in a order to my liking.  Anyway, the file I wanted to print was corrupted in the process -- truncated. The printer did everything asked of it, but the G-code abruptly stopped, so no commands to home or shutdown the heaters, etc. -- it really did look like the printer failed.

Any details, the CALIBRAT.txt file (if it was created), something captured by Pronterface, ... anything may be helpful in tracking down what's going on.

The printer here has logged many hours with very few problems, so I'm fairly hopeful that we'll find the cause(s) for your printer's troubles.

BackwardEcho, 2020-12-30:
It crashed and rebooted without the LED flashing red. I checked the gcode file and didn't see any corruption that I could notice, everything was legible and not a bunch of gibberish special characters. It actually **needs** pronterface to be connected or it will crash my printer. Doesn't matter if I started the gcode from the printer itself instead of through pronterface, it would crash if not connected. 

[CALIBRAT.TXT](https://github.com/aegean-odyssey/mpmd_marlin_1.1.x/files/5753306/CALIBRAT.TXT)

BackwardEcho, 2020-12-30:
[Pronterface capture.txt](https://github.com/aegean-odyssey/mpmd_marlin_1.1.x/files/5753322/Pronterface.capture.txt)

aegean-odyssey, 2020-12-30:
@BackwardEcho, thanks for the captures.  Info is helpful. Please take a look at the video in issue #48. Do you think that your issue could be the same issue? I hope so... but at the same time, it's a serious bug that I think will be difficult to find. On the up side, @mulcmu seems to have reduced it to a repeatable, specific set of circumstances. 

Thanks for your patience. We may be zeroing on the problem.

BackwardEcho, 2020-12-30:
I did see the hesitation before, but the printer doesn't even move before rebooting when I start the gcode and I saw the hesitation once with David Keeton's bed leveling script. Might be related though.

However, I can confirm that the printer works fine with either the usb unplugged or connected with a port open. I can try and take a video if you want.

aegean-odyssey, 2020-12-31:
No need for the video, but I may need some testing in the future. For now, I'll look over the code and see if I can work out what's happening. Thanks for your help.

BackwardEcho, 2020-12-31:
No problem, I hope you are able to find the issue. I guess it's a good thing I had the idea try and streamline the recalibration process for me for when I need to do a M502 (by that I mean put the important values in a gcode to then start the whole process with the carbon paper and script by David)

aegean-odyssey, 2020-12-31:
Yes, reset and reload. A backup of the `CALIBRAT.txt` file also captures the settings, just in case.

aegean-odyssey, 2021-01-18:
Btw, does the `mpmd_marlin_1.1.x-119r15` firmware resolve this issue? Can the issue be closed?

BackwardEcho, 2021-01-18:
Oh, sorry. I've been preoccupied with a move I have not gotten time to check. Once I am off work I will check.

BackwardEcho, 2021-01-19:
It appears to no longer crash after just applying the update. I tried both the regular and 25mm auto calibrate files just to be sure while having the usb cable in, out, connected, and disconnected. Sorry my bug "report" wasn't like mulcmu's (woah, did he dig into it) and I also apologize for not trying it sooner. I recently moved.

and time will tell for that weird boot error from months back. After a number of times flipping the switch on it, I didn't see the led go red at all. I closed the issue though after realizing you probably wanted it gone. sorry >_>

closed 2021-01-19 by BackwardEcho

aegean-odyssey, 2021-01-19:
Good to hear it worked out. I appreciate your perseverance. Some problems can be particularly frustrating.

And @mulcmu's bug reports are exceptional -- a high bar for anyone to emulate, but fine examples none the less.  They've certainly encouraged me to "up my game" in that regard.