aegean-odyssey/mpmd_marlin_1.1.x #51: Z height

skylersell, 2021-01-17:
Hi I change my hot end over to a e3dv6 heartbreak and heatsink but with the original heater block and I believe my Z height at 450 is no longer adequate. How do I go about finding the correct Z height. Thanks

aegean-odyssey, 2021-01-18:
Someone else may have more insight and a better answer, but here's what comes to mind. From a USB connection and terminal (or host):

* first, home with `G28`
* and set the height with `G29 P0`
* then probe the center (0,0) with `G30 V3`

During the tapping process, the `G30` command should spit out information about the Z positions and offsets along the way. I think with this information, it should be possible to determine a reasonable offset to use with the `M851` command. Then:

* set the probe (nozzle) z-offset with `M851 Z<your offset here>`

skylersell, 2021-01-19:
thanks I think it worked

closed 2021-01-19 by skylersell