aegean-odyssey/mpmd_marlin_1.1.x #52: Firmware Flashed Successfully, SD Card No Longer Reads

TaylorDeeves, 2021-01-22:
I have the Indiegogo model and installed the SM0001 AC 5ALimit firmware using the original SD card that came with the printer. Seems that the firmware installed fine, I can home and preheat but after the upgrade I am unable to get any SD card to read. “sd card is empty” pulls up every time I select Print. If I reload the card with the bin and flg files it will reflash the firmware, but once rebooted it says the card is empty.

aegean-odyssey, 2021-01-22:
I _think_ it should work. Here, when the card isn't read from time to time, it turns out to be mild disk corruption (here, likely caused by me not properly ejecting the card from the laptop). Your issue seems a little odd to me because usually it's the "firmware flashing" that is finicky about the micro SD card, not the other way around.

Is it possible to try another card?

Also, the printer only shows files with a .g?? extension (e.g. .G, .GC, .GCO). It will display "no files" if it can't find a file with a "gcode" extension.

Maybe there's a simple cause. Please report any findings. Hoping that it's not a bug, but if so, we'll track it down.

TaylorDeeves, 2021-01-22:
I edited my post, but just to clarify, when I go to stock v41 firmware the SD reads fine. 

With marlin installed (it shows v11915.113.3 on boot) it does not detect gcode files in root or in the setup_gcode folder and says sd card empty. 

These gcode files display fine on stock firmware.

aegean-odyssey, 2021-01-22:
Got it. The `setup_gcode` folder should definitely be displayed. I will try to reproduce the issue here.

aegean-odyssey, 2021-01-22:
Ok, here's what I found.

On the printer here (a model from 2018):

I tried a micro SD card that came with an MP Mini Delta printer (from an open box buy) -- never used. I worked fine, displaying the gcode files that were on the card.

I tried the micro SD card that came with my printer -- with setup_gcode, firmware.bin, and gcode files on the card from a while back. It showed "No files", just as you described. Popped the micro SD card in the laptop, and performed a disk check (fsck.vfat on linux) -- it said the disk needed repairs; it performed the repairs. Subsequent checks showed no errors. When I tried the card again in the printer, all was ok -- files and folders displayed as expected.

The machine here is not from the Indiegogo campaign, so I can't set up an exact test environment. I do think, though, that there is something wrong with the micro SD card. Here even with the corrupted card, the laptop read and displayed the files and folders on the card (appearing like nothing was wrong). But the laptop mounted the drive as "read-only" -- it's a tell-tale sign that something was amiss with the card. The fsck disk check confirmed (and repaired) the trouble.

If the micro SD card is error free, and reformatting the card doesn't fix the problem, then we'll have to figure out how to recreate the conditions here to figure out what's going on. I appreciate your patience. It's much easier to tackle a problem if I can reproduce the issue here.

TaylorDeeves, 2021-01-22:
Do you have a recommended tool to verify and repair on OS X? I will try that next.

aegean-odyssey, 2021-01-22:
I believe on the Mac, it's "Disk Utility" and at one time, "Disk FIrst Aid" (it's been some since I've used a Mac). I think "Disk Utility" should be able to repair an SD card -- or at the very least identify that there is an issue with the card.

aegean-odyssey, 2021-01-22:
Oops, sorry. Just re-read your comment. "Disk Utility" is out. Maybe under the hood, since it's unix, there is fsck (fsck.vfat) -- but I really don't know. Apple has their own way of doing things.

mulcmu, 2021-01-22:
My machine was a pretty early one (2017 black Friday sale).  It came with a micro sd card without any markings, plain black.  File system reported it was 1000 MB but it was either damaged or counterfeit as only first 256 MB were valid.  Might be worth trying a counterfeit card scanner to verify the size.  That card was the source of much frustration, it would not give any errors when saving or reading a file back.  However about half way thought normal G-code would be replaced by random gibberish.

aegean-odyssey, 2021-01-22:
Doing a little reading -- getting under the hood of OSX to try to fsck seems a bit daunting, involved, and error prone, so I do not recommend it.  If you're up for it, I suggest two things:

* try another micro sd card (it will be good to know that something works). The experience here is that most cards can be used to print, but some just will not work to update (flash) the firmware.

* reformat the existing card rather than try to repair it. Since the Mac is OK with the card, copy the contents of the card to a folder, reformat the SD card, and then copy the folder contents back onto the newly formatted card.

TaylorDeeves, 2021-01-22:
Okay I found a spare 8gb card that came with my Ender 3 I popped it in and it reads fine. 

Very strange. I will attempt to fix the default SD with deck later I will update this thread if I am successful.

TaylorDeeves, 2021-01-22:
Repairing file system.
Volume was successfully unmounted.
Performing fsck_msdos -y /dev/rdisk4s1
** /dev/rdisk4s1

** Phase 1 - Preparing FAT

** Phase 2 - Checking Directories

** Phase 3 - Checking for Orphan Clusters

315 files, 880280 KiB free (1760560 clusters)

File system check exit code is 0.
Restoring the original state found as mounted.
Operation successful.

aegean-odyssey, 2021-01-22:
Thank you for following up. I'm very glad it's working for you and not the mystery it was shaping up to be. And the silver lining: `fsck_msdos -y /dev/rdisk...` known to work on OSX.

I'll close this issue out. Thank you for giving the firmware a try.

aegean-odyssey, 2021-01-23:
I'm sorry, I'm re-reading your comment -- not sure I understood. Did `fsck_msdos` actually work? Is the printer able to read the SD card?

aegean-odyssey, 2021-08-28:
Closing. If you feel the issue should remain open, please comment and re-open it.

closed 2021-08-28 by aegean-odyssey