aegean-odyssey/mpmd_marlin_1.1.x #53: autolevel and bed thickness

drji3m, 2021-01-24:
Aut.o level works fine and bed spacing is fine with zoffset  0,
however if i increase the bed thickness and rerun g28,g29 the new thckness (glass) is not compensated. 

From all the reviews of the stock MPMD autoleveling takes care of thickness changes. Noot so with this firmware  v15.

am i missing something?

PurpleHullPeas, 2021-01-24:
Glass beds work fine on this firmware. A z-offset of 0 is never going to be correct because of how the bed switch system works  

What is your Start Gcode? Send M503 over EEPROM and Copy-Paste the results. I imagine the problem is in there somewhere, because as mentioned in the Quick-Start Guide, certain things like G29 and z-offset are a different in this firmware. Otherwise, it could be a hardware issue, but let's rule out gcode, first.

PurpleHullPeas, 2021-01-24:
The stock firmware threads you read are from a standpoint of having no additional G33/M665/M666 calibration. Glass beds can work fine with this firmware, but everyone who uses them probably goes through the whole calibration process with them, because it will affect final calibration results. 

A z-offset of 0.0 makes no sense on the MPMD and makes me think that there is some confusion here. Stock firmware keeps M851 at Z0 and adjusts z-offset via G29 Z. The recommendation for MPMD Marlin 1.1.X (as mentioned in the Quick-Start Guide) is to set your z-offset in M851 and to just use G29 P0 in your Start Gcode. I do not know what happens in this firmware if you also define G29 ZX.xx, but it did some weird stuff in Marlin4MPMD 1.3.3. 

There is also the possibility of a hardware obstruction. Most of the time, glass retainer clips do not play nicely with the MPMD's ABL system. I use a thermal pad. 

You could also post your Start Gcode and your M503 data. That might shed some light on the situation.

drji3m, 2021-01-24:
Actually I am using piezo discs at the three corners which trigger the probe well before the mechanical switches. The probe z offset is very close to zero since the bed doesn't really move when probed. As much as I know the z offset setting should be the difference between probe and nozzle.  Which is why I set it to zero. With the micro switches the zoffset is fairly large since the bed travels down visibly.
I have confirmed the value of zoffset (m851) is simply added to the table produced by g29. And I thought the g29 znnn was a fine tuning and also added to the compensation. That's why I thought bed thickness would wash out.

drji3m, 2021-01-24:
I also used a thermal pad for the glass test. No clips.
My point is that the g29 table produces results larger by the thickness if the glass as I would expect. But I thought the table values were used to correct the z value but this does not happen. The nozzle just crashes into the glass as if no adjustment to z was made

PurpleHullPeas, 2021-01-24:
Thanks, that makes more sense. I had some nozzle crashing issues a while back, and it was due to EEPROM corruption. A fresh flash, followed by M502->M500->Power_Cycle ended up fixing it. That suggestion was in the release notes at some point, but I ended up adding it to the main Quick-Start later because of how often that screwed people up. I would still only adjust z-offset inside M851, just to simplify things. 

For my setup, I routinely run G33-like calibration with a glass bed, remove it, put my flex plate/WhamBam back on, run G29, and then print with M851 Z0.28 and G29 P0. Lots of other people in the community (including the developer) print with a glass plate on this firmware.

aegean-odyssey, 2021-01-25:
Any confusion about the `G29` command is entirely justified:
* there are many options that aren't always consistent;
* the stock firmware kind of considers "bed level" and "machine calibration" the same thing;
* Marlin has it's way of doing things; and
* I've added a few non-(Marlin)-standard things, as well.

Simply put, `G29`... it's complicated.

But for your situation, @drji3m, all should work the same way. Starting with the least complicated:

* `M851` is indeed the probe offset that is added to the `Z` value of a probed (tapped) position. In the case of the stock machine, the probe is about 0.5mm _below_ the _surface_ of the bed (`ΔZ = -0.5mm`). Since we're talking about a delta, the glass plate thickness doesn't factor into things here (it does affect the height calculation in `G33`, though). `M851` adjusts where the printer defines `Z=0`.
* The `G33` command tries to automatically adjust parameters associated with the printer's geometry (height, endstops, radius, arm length, tower angles). It's important to note, that the `M851` value is _not_ used during this calibration process -- so with `M851 Z0`, after the `G33` calibration, a move to `Z=0` would send the nozzle below the bed (to where the bed switches triggered during calibration).
* In Marlin (in this firmware), the `G29` command is all about the the bed level mesh and technically nothing to do with the machine's geometry. ___BUT___ there is an exception: `G29 P0` -- it is actually an alias for the `G33` command that determines the height parameter, placed in the `G29` g-code to be backward compatible with Marlin4MPMD and I guess to look a little like the stock firmware. All of the other `G29` parameters relate to the 7x7 bilinear bed mesh. The mesh compensates for errors associated with the print bed, but it can be used to compensate for errors in the printer's geometry settings, as well.

So to your original questions about `G29 Z...`, the `Z` parameter in the `G29` command is the z value (actual or offset) applied to the _mesh_:
* When creating the mesh automatically (via probing) the z value is added to the probed value.
* When writing directly to a mesh point with the `W` option, the z value is the value stored for the particular grid point.
*  When the z value is used with the `C0` option (reset the mesh), it is the z value set for every grid point in the mesh.

One important side-effect of the way things work is that the probe offset, the value set with the `M851` g-code, is ___included___ when the `G29` command does a bed tap to determine the "measured" z value. And to be honest, I get confused trying to understand the implications, so I set `M851 Z0` before using a `G29` command that probes for the bed level mesh. In the g-code setup files, I set `M851 Z0.6` after the `G29` as a "safe" value to move `Z=0` above the build plate.

So in your situation, I think calibration would look something like:
```
; perform an auto-calibration
G33
; create a bed mesh
M851 Z0
G29
; if the piezo triggers at or just below the bed, use only a little offset
M851 Z0.05
; don't forget to save
M500 
```
and then printing (part of the start-up g-code):
```
G28
G29 P0 ; re-calibrate just the height
```

I think your set-up should work just fine. Please report back your findings. And perhaps start a full-blown discussion (in Discussions) about the bed level switch modifications -- it sounds interesting.

aegean-odyssey, 2021-01-25:
@drji3m, there's more:

I neglected to mention `M420` which enables/disables the bed level mesh. It is disabled by default. It is generally not an issue because `G29 P0` automatically enables the bed level mesh.
```
G29 P0
; is equivalent to
G33 P1 V1
M420 S1
```
If you're not using `G29 P0` in your startup g-code, then there should probably be an `M420 S1` in its place to enable the bed level mesh.

drji3m, 2021-01-25:
Wow. Thanks for the explanations.

My brain had been full of massive confusion about this.

When I put the glass on and did a g29 v4 the table showed bed points
hovering around z4 mm.
No adjustments to zoffset would make the nozzle not think z0 was still
beneath the glass.

Then I ran your calibration script an like magic g29 then produced a table
with small deviations around the top of the glass (the correct zero point).

Good.

But when I tried to recalibrate without the glass the probing would stop in
midair about where the glass was. Resetting the printer did not help. So I
restored factory settings and now am back to normal after a full
recalibration.

So I guess the rule is to

1. Measure zoffset and set and forget.
2. do a full cal with any change in surface or nozzle.
3. Just use g29 p0 znnn to fine tune.

Saves time on each print.

BTW do all the docs for marlin g29 not apply?

Also g30 produces a huge value of z and what is q? It does add in to
changes in zoffset but I have no idea what its use is.

drji3m, 2021-01-25:
BTW 'Discussions' appears to be locked to any new posts.

mulcmu, 2021-01-25:
There might be an issue going on with the `G29 P0` correction.  I started a print with a 3mm thick piece of acrylic placed on the bed.  `G29 P0` did the double tap on the acrylic which was then removed before the printing started.  I expected first layer to start getting deposited 3mm above the bed, however print started on the bed surface.

drji3m, 2021-01-25:
So maybe g29 p0 does nothing.
My gcode start code does a g29 on every print. But now my paradox is that
even that won't account for bed leveling since it doesn't account for bed
thickness (even the small amount do to level shift) and a g33 might be
required.

At this point it seems like its either a waste of time or an error to do
any probing on each print- just like manual leveling.

drji3m, 2021-01-25:
Actually I don't understand the point of probing a single point anyhow.
What would it correct for?
A bed that somehow moved up or down without tilting? Seems not even
possible.

I thought the mesh building was to correct for be shape as well as tilt and
therefore thickness to some degree.

When I look at marlin ubl stuff what would be really cool is to fully
calibrate for a given surface and build the mesh. But then on each print do
the 3 point probe to correct for physical tilt by transforming ( tilting)
the mesh data as they describe. But even then physical tilt is really only
likely with spring loaded beds .

(Below is a related but different discussion)

The mini delta really doesn't tilt once the slop is removed. In Dennis's
case his binder addons put downward pressure on the bed to stop the rattle.
Unfortunately this preloads to micro-switches to higher sensitivity but
adds trigger instability.

In my case my pizeo mounts put upward pressure on the bed against the oem
stops and preloads the pizeo discs (very much allowed) This removes wobble
and gives high but stable sensitivity

So even if g29 p0 worked as you expected I'm not sure what its value would
be for the delta.

aegean-odyssey, 2021-01-25:
@mulcmu, your test should have produced the result you expected; that the nozzle should have stopped above the build plate. Something is not right. I'm thinking that after the `G29 P0` probe, the machine is not taking up the new height value. Does homing (`G28`) after the `G29 P0` change the outcome?

@drji3m, the `G29 P0` command probes to find a new "height" value (`M665 H`). The height figures prominently with a delta configuration because the endstops for the machine are not at the zero mark for the axes, but rather at the opposite end, the maximum limits for the axes. When the printer homes, it sets the position to X:0, Y:0, Z:_height_ (+ probe offset) and then goes from there. 

The 3-point calibration of the stock firmware seem to require tinkering with every print to get a workable first layer -- with mostly failed prints, it was very frustrating (I think the probing process was more variable than the actual bed position). The calibration then 1-point probe of the Marlin4MPMD firmware didn't fix the first layer completely, but it did produce consistent results and allowed for much more frustration-free printing although, here, still with a lousy first layer. 

You are correct, though, if the printer's bed doesn't move, its geometry isn't changing, and the endstops are repeatable, then there's no need to probe at all; just home and enable the bed mesh (`G29` followed by `M420 S1`).

The `Q` in the output of the `G30` (probe point) is the z value with the bed level mesh correction applied. It should only show up if the bed level mesh is enabled. The `G30` g-code can spit out quite a bit of information. The goal was to make the `G30` command useful when creating external programs to calibrate or exercise the printer.

The `G29` options largely follow the Marlin docs (for AUTO_BED_LEVELING_BILINEAR) except as noted in the wiki, https://github.com/aegean-odyssey/mpmd_marlin_1.1.x/wiki/G-M-code-Support. It produces a different output and more output, again with the idea that the output may be parsed by an external program for some kind of added utility for the printer.

And finally, I apologize for the confusion with "Discussions". It's a new Github feature, and apparently I don't how it all works yet. So, I've got some kinks to work out.

drji3m, 2021-01-25:
OK. G29 P0 doesn't seem to do anything whether followed by g28 or not.

Changing m851 zoff does have an effect IF followed by g28.

I successfully made the glass bed recognized by using g29 followed by g28.
Then g0 z0 moves to the top of the glass. No need for a full calibration.

So I remain confused but way less confused.

drji3m, 2021-01-25:
BTW G28 keeps bed level in on or leaves it on.

M420 s0 ;BL -> off
G28
M420 ; reports BL on

M420 s1 ;BL -> on
G28
M420 ; reports BL on

aegean-odyssey, 2021-01-25:
Yes, thanks. A couple of things:

* `G29 P0` does not work as advertised. Actually, it goes through all the motions and then sets the height to the value it had when it started -- it's a bug. I'll track it down. As if `G29` wasn't confusing enough!

*  The firmware is compiled with `RESTORE_LEVELING_AFTER_G28`, so the current state of the bed leveling _is_ preserved after a homing command (`G28`). Plus, the bed leveling state is stored in "eeprom" with the `M500` g-code, so the user can choose to have the bed level mesh enabled (`M420 S1`) or disabled (`M420 S0`) when the printer starts up. (Of course, the mesh is enabled only _if_ there is a mesh stored in "eeprom" as well).

drji3m, 2021-01-25:
I just had a weird thing happen.
I started a print which has a plain g29 in its startup code. After some
temo stuff it started the probe at the first point then just stopped and
reported an error.

I stopped the job, reset the printer and then manually sebnt g28 g29.
Same thing. Just the first point then error.

The only fix was a factory reset and full calibration. Now working normal.
Something get written for flash that breaks things. Oh well.

aegean-odyssey, 2021-01-26:
@drji3m, I think you bumped into another little "quirk" with how the bed probe works:

As a safety feature or sanity check, Marlin does not allow the nozzle/probe to travel too far (a few millimeters, maybe) below the print bed. The error is generated when probing the bed and the probe reaches zero or so without tripping a bed switch. The condition can easily be created when the height parameter (`M665 H`) is small enough to keep the nozzle from reaching the bed before the motion reaches `Z=0` (slighly less than zero). There is more discussion about this in issue #43. 

The factory settings use a fairly large height value to ensure that a bed switch will trigger before the probe reaches `Z=0`.

aegean-odyssey, 2021-08-28:
Closing. Multiple issues, but underlying bug is fixed. If you feel that this issue should remain open, please feel free to comment and re-open it.

closed 2021-08-28 by aegean-odyssey