aegean-odyssey/mpmd_marlin_1.1.x #63: Corruppt display

john-clark, 2021-02-09:
I followed the instructions and ended up with a non functioning printer.

The display looks like this: https://i.imgur.com/TSITvip.jpg

The web page gcode box does not respond, the serial port also will not connect.

I renamed mpmd_marlin_1.1.x-119r15-SM0001-ACfan-10Alimit.bin to firmware.bin and put it in the root of the sdcard.

aegean-odyssey, 2021-02-09:
A corrupt display is  a sign of an issue with the printer's _UI controller_  circuit board; the `mpmd_marlin_1.1.x` firmware is installed on the printer's _mainboard_ circuit board. And in the case of the MP Mini Delta printer, the Marlin firmware has very little ability to control the _UI controller_ board let alone alter its programming -- I am completely at a loss as to how updating the firmware on one circuit board has seemingly corrupted the firmware on the other circuit board. 

Stil, there are a couple of things:
* The web page/ server is part of the _UI controller_. It collects the user input from the web page and passes it on to _mainboard_. The `mpmd_marlin_1.1.x` firmware in the mainboard does __NOT__ support this communication. So using the WIFI connection to control the printer will __NOT__ work with the `mpmd_marlin_1.1.x` firmware.

*  On the other hand, the USB serial port should very much work. The port is physically on the mainboard, and is entirely under the control of the `mpmd_marlin_1.1.x` firmware. The UI controller is not involved at all. Are you using a Windows computer with the USB? If so, issue #3 discusses the need for a "comport" driver update so that Windows will identify the printer's USB port.

As for fixing the UI controller firmware, I have __zero__ experience with this issue. There is a quick mention on the [mpminidelta.com](https://mpminidelta.com/firmware/ui_assets_update) site, but the required `ui.bin` file does not seem to be available. Perhaps someone will "chime in" on this issue. OR, perhaps contacting the very knowlegable Matthew Upp will help -- [Matthew Upp (facebook)](https://www.facebook.com/matthewupp) and [Matthew Upp (reddit)](https://www.reddit.com/user/tiny_tim_).

Please report any findings. Yours is the first report of "updating the firmware affecting the UI controller". Also the firmware and mainboard should work even _without_ the UI controller board, so we should try to figure out why the USB port is not working for you.

PurpleHullPeas, 2021-02-09:
I have seen at least a couple of others mentioning that "flashing the firmware" (in the most general sense, not necessarily MPMD Marlin 1.1.X), "bricked their printer." At least one person came back to report that they fixed it with a successful firmware flash. 

I do not know what they did differently. I just know that the MPMD can be very finicky about SD Cards and the stock SD Card is prone to failure. If you lived in my area (North AL), I would offer to let you try my known working card. 

Chris from the Facebook Group also found a way to flash the bootloader and firmware using an ST Link, instead. I [copy-pasted it here](https://drive.google.com/file/d/1qK-VuOvtXk0OAKperCJYZGfuytqHtzNo/view?usp=sharing) to make his information accessible outside of Facebook.

rgrobe, 2021-02-09:
While I have flashed the various MPMD firmwares multiple times, I also tried my luck updating the UI software more than once (mostly following the mpminidelta.com suggestions), and somehow managed to damage the UI assets slightly (not as bad as here, but with some erroneous bits in the extruder motion submenu).
I finally managed to download a UI.bin file from another Monoprice-related forum (don´t remember exactly from where)  and could reload the UI assets properly using the MiniDelta´s own webpage (following the upload explanation on the mpminidelta.com site). I have attached it here - please use at your own risk. Good luck!

[m200_-_ui_assets_for_v100.2_-_v120.2_-_monoprice_logo.zip](https://github.com/aegean-odyssey/mpmd_marlin_1.1.x/files/5951995/m200_-_ui_assets_for_v100.2_-_v120.2_-_monoprice_logo.zip)

john-clark, 2021-02-10:
I tried the firmware @rgrobe suggested and I had a different mangled display. https://i.imgur.com/S0ncLbb.jpg 

I did find various lcd.bin/update.bin, ui.bin, firmware.bin, combinations available although none with any real documentation. I was able to get my display back by flashing various firm ware found online.  To get back to something semi functional I flashed:

mpmd_-_motion_v41_-_20170815_-_firmware.zip
m200_-_ui_assets_for_v150.2_-_v160.2_-_malyan_logo.zip

Although home is grinding the top of the device. To be honest I am not sure if it was the combination or the result of  the accumulation. Now, I will now start over with mpmd_marlin_1.1.x to try and get a functional flash as I may have had a corrupted download or something wrong with the card. 

My guess is that when flashing the firmware all these different bin files flash different parts of the controller, and there is no protection for large files over writing other parts. It would be nice if they released code and documentation for the firmware, since seems it to be abandoned by the OEM.

john-clark, 2021-02-13:
With the above flashing the web ui was corrupt so I clicked upload to get back the default webui.

Then I flashed mpmd_marlin_1.1.x-119r15-SM0001-ACfan-10Alimit.bin and it seems to be working. Except, now have weird flashing on the lcd when I select menus, but the menus are not corrupt.

I got to the /setup_gcode/AUTO_CALIBRATE.gcode set and it ran about 5 times back to back without stopping, I was able to cancel. I am also getting this error as well

`Error:MINTEMP triggered, system stopped! Heater_ID: bed
[ERROR] Error:MINTEMP triggered, system stopped! Heater_ID: bed

Error:Printer halted. kill() called!
[ERROR] Error:Printer halted. kill() called!`

The serial connection is working somewhat,

aegean-odyssey, 2021-02-15:
First, @PurpleHullPeas and @rgrobe, thank you for helping out. I've no experience with this display issue, and I'm still trying to come up with a plausible set of circumstances that lead to flash memory on the UI display board somehow being altered while programming the flash memory on the motion controller board -- especially since there is only the single serial port connection between the two boards _AND_ there is no code at all in the Marlin firmware that tries to "configure" the UI board other than an initial "reset" sequence. So it seems that it would be one helluva fluke for the controller board to happen to write something to the UI board that corrupts the UI board's firmware. On the other hand, we have evidence that it can happen.

I recall seeing a comment somewhere that perhaps in some of the early Monoprice printers the display board could be flashed by placing `ui.bin` on the micro SD card along with all endstops tripped and some other hoops to jump through. At the time I thought it was some extra function the Monoprice firmware performed to be able to program the UI board. But maybe it's something that is built into the bootloader. As @john-clark mentions, it's hard to know without documentation or source code for the bootloader, etc.

@john-clark, the `G33` code can make quite a few passes before it decides that another calibration pass won't improve things and that it's time to wrap up the calibration procedure.  As for the "MINTEMP" error, I believe this occurs when a temperature set with the `M104 S...` g-code cannot be established or maintained. This behavior is part of Marlin's safety features regarding the printer's hotend.

_BUT_, the `/setup_gcode/AUTO_CALIBRATE.gcode` command doesn't mess with the hotend or bed temperatures, so if your scenario didn't include setting the hotend or bed temperatures, it sounds like a bug. Either way, I'd like to better understand how the error manifests itself, so any additional information you can provide will be helpful.

Glad to hear that your printer's display is again working and I appreciate your perserverance with the `mpmd_marlin_1.1.x` firmware.

aegean-odyssey, 2021-08-28:
Closing. If you feel the issue should remain open, please comment and re-open it.

closed 2021-08-28 by aegean-odyssey