aegean-odyssey/mpmd_marlin_1.1.x #64: E1"Heating" Error

zapata7515, 2021-02-19:
I recently replaced the motherboard with a silent board from gigdigit and after swapping I keep getting this E1 error. I checked that all my cables were set properly but it didn't help. I just swapped back to the old motherboard which also was running this firmware and I have no issues with the old one so I can rule out a bad thermistor. I have messaged the board maker to see if he can help but I wanted to check and ask here if anyone has had this problem and maybe I'm just doing something wrong.

mulcmu, 2021-02-20:
I've encountered this twice so far on`119r15` as well.  This release had some changes made to the thermistor/temperature so rolling back to `119r14`  if you were running `119r15` on the silent board would help troubleshoot.

[image]

[image]

PurpleHullPeas, 2021-02-20:
I am still on release 13 with a Gigdigit silent board and upgraded hot-end fan. I have to slow my fan speed down during initial heatup to avoid this error, but it is otherwise working fine. You could also flash the older Marlin4MPMD 1.3.3 to see if it has the problem. I don't have any proof, but I feel like, in terms of heating error sensitivity MPMD Marlin 1.1.X > Marlin4MPMD 1.3.3 > Stock Firmware.

zapata7515, 2021-02-20:
Thank you for the responses I am running the latest 15 version I will try to roll back down or try Marlin4MPMD

zapata7515, 2021-02-20:
Welp I'm at my wits end I swapped board but I must have mucked something up cause my printer is now giving me homing errors even with my old board.

aegean-odyssey, 2021-02-20:
I'll look into the thermal run-away code.
A few questions:
* Are you using the `-05Alimit` or `-10Alimit` firmware?
* Does the fault occur right away, or after the nozzle has been heating for a few minutes?
* Does the fault occur before the nozzle and bed reach their target temperatures?

Also, when swapping out firmware, be sure to restore the settings to the factory defaults. Sometimes the existing settings are corrupted when installing (flashing) the `mpmd_marlin_1.1.x` firmware (or most certainly wrong if installing over other firmware), and `mpmd_marlin_1.1.x` firmware is not very good about detecting this corruption. Erratic motion or behavior is a tell-tale sign of corrupted settings.

zapata7515, 2021-02-21:
I was using 10Alimit firmware as I have a 10amp psu. The error only happens when I preheat or try to print so it doesn't happen right away. The error pops up before the nozzle and bed reach target temperature. I made sure to run the gcode for factory defaults and ran through the calibration but once I tried to make a test print it failed and gave me the E1 error. I swapped back to my old motherboard for now but I am going to try to swap back with an older firmware as others suggested.

PurpleHullPeas, 2021-02-21:
Shot in the dark here, but what is your Start Gcode? Some people have gotten screwed up by using Brian's old start Gcode from the wiki which has some really terrible (for Marlin) PID values hard-coded at the top. That used to be a super common cause of heating errors back when his Start Gcode was more prominent. 

Regarding homing problems, I had it happen recently when my belts were too tight. It can also happen if wires/tubing are blocking the carriage from reaching the top.

zapata7515, 2021-02-21:
I was just using the presliced files included here from the repository. As for the homing issue I fixed it I just messed up the wiring. I'm waiting to get a new nozzle so once I fit it I'll swap boards again and see if it works with the lower firmware.

aegean-odyssey, 2021-02-22:
Thanks for the info. There's a handful a places where the error is generated. I'll look into the conditions that trigger the error to see if we can figure out what's going on.

mulcmu, 2021-02-22:
My two occurrences were on a open box printer I purchased recently while running `mpmd_marlin_1.1.x-119r15-SM1110-ACfan-10Alimit`.  Each occurred on the 2nd layer in different locations while printing a spiral test pattern.  Nozzle temp was 215°C.  After the first occurrence I ran the PID autotune.  I was able to print some other small test items without the runaway triggering.  I ended up loading `r14` and was able to print the spiral twice without the runaway triggering.

mulcmu, 2021-02-27:
After a few more tests of the spiral file that was giving me problems, runaway was encountered on `r14` and `r15` sometime during the 2nd layer.  I realized initial layer temp was set to 215°C in the file but then dropped to 200°C.

aegean-odyssey, 2021-02-28:
@mulcmu, thanks for reporting. I've been trying to figure out how the problem could be in `r15` but not `r14` -- the fact that it occurs in both makes more sense. I need to look at the thermal run-away code again, but I think that in the situation that you describe, the nozzle temperature is not dropping from 215&deg;C and settling in to 200&deg;C within the 40 seconds configured for the thermal run-away code. Is this a possibility? The fault should occur approximately at the same _time_ into the 2nd layer.

There seems to be a couple of error checking loops for the hotend and again for the heated bed (each with different configuration parameters), plus it appears that the "loop times" (error checking intervals) vary with the CPU load. I'm not sure how it all interacts, or how to fix or improve it just yet.

zapata7515, 2021-03-06:
I dropped down to version 13 and it doesn't seem like I am getting any problems anymore but at the cost of not being at the latest firmware.

aegean-odyssey, 2021-03-07:
@zapata7515, thanks for the report. There have been a few fixes and some restructuring in the firmware between `r13` and `r15`. The firmware is now configured to display the A/D values of the thermistor measurements and to use less (code) space, I rewrote the code that periodically displays the temperatures. Otherwise, the temperature control's function should be unchanged.

It's a bit of a mystery (though likely something I did) why the temperature control performs differently from one firmware version to another, so any observations about the thermal runaway are helpful.

mulcmu, 2021-03-13:
The below gcode was an attempt force a bunch of extraneous processing to try to trigger the runaway.  It opens file for output on sdcard, spams text to LCD message, `M503` to output lots of text, and `G0` moves to run the carriage steppers.  I was not able to trigger runaway with this test on my MPMDs running `119r15`.

There were some pauses between the `G0` commands when the USB was connected.  When USB serial was not connected motion didn't have any pauses.

[Stress Test.gcode.txt](https://github.com/aegean-odyssey/mpmd_marlin_1.1.x/files/6134828/Stress.Test.gcode.txt)

aegean-odyssey, 2021-03-15:
It appears that the thermal run-away code is constantly monitoring (as it should) the target temperature and the actual temperature. A fault occurs if the actual temperature does not find its way to the target temperature in the configured time frame (40s) or if actual temperature strays too far away from the target temperature. At the same time, the PID controller is driving the actual temperature toward the target temperature. If all goes well, the dynamic response of the PID control loop does not trigger a thermal run-away fault.

I suspect that in some cases the PID control does not eliminate the temperature error (that is, drive the actual temperature to the target temperature) quickly enough and a fault occurs. Another possibility is that a bogus temperature reading somehow triggers a fault. It's not clear to me why the behavior seems to vary by firmware version, or why the fault is not more repeatable.

I think a test that drops the nozzle temperature during printing might trigger the fault. Something along the lines of: 

```gcode
M109 S215 ; wait for nozzle temperature
M190 S60  ; wait bed temperature
M104 S205 ; set a new target temperature, but continue
; add print command, etc. here to run for a minute or so to see if a run-away fault occurs
M104 S0
M140 S0
```

I not sure how the "load" of printing during a large temperature excursion affects the thermal run-away code or the PID control code -- both pieces of code execute in a loop that may be sensitive to the overall activity of processing g-codes and controlling the stepper motors. Also, since the PID controller only controls a heating element, probably dropping to a desired temperature takes longer than rising to a desired temperature, and is a bit more variable -- determined by factors such as the ambient temperature, fan speed, etc. 

Once we get a better handle on what's going on, I think we'll be able to adjust the thermal run-away's configuration parameters appropriately, or fix code if necessary.

BackwardEcho, 2021-03-23:
[thermal test.txt](https://github.com/aegean-odyssey/mpmd_marlin_1.1.x/files/6186999/thermal.test.txt)

I tried this on `r15` with no problems, though it doesn't actually print anything. It just moves the print head around for a couple of minutes at 215/50c. If anyone wants to try it and or add to it, feel free and please tell me. I'd like to try whatever changes are made to it.

BackwardEcho, 2021-03-23:
...Well, that gcode I made works but an actual print doesn't. Not for either `r15` or `r13`.

BackwardEcho, 2021-04-01:
Should I consider replacing my thermister if `r13` isn't working?

PurpleHullPeas, 2021-04-01:
> Should I consider replacing my thermister if `r13` isn't working?

It depends. From my observations, the thermistor is the easiest to accidentally damage while working on the hot-end (seen many times in the Facebook Group); however, I'd bet the heater cartridge is more likely to fail on its own. I've had to replace the heater cartridges on both of my MPMDs, but never a thermistor. I upgraded to a 40W heater block this last time, and it has been great. 

If your printer either has trouble with initial heat-up or only has trouble with higher temperatures, then I'd guess heater cartridge.

BackwardEcho, 2021-04-01:
I knew of the one from gigdigit, I'm just looking to guarantee quality atm with any others. I'll wait for a `r16` before I make my purchase, maybe even look around now that I know equivalent specs. Thank you.

PurpleHullPeas, 2021-04-01:
12V 30W is stock. I upgraded to 12V 40W because I deduced that the mosfet on the board can handle it, since the same mosfet is used for the 40W bed heater. The higher wattage is also why I mentioned my upgraded PSU.

aegean-odyssey, 2021-04-02:
@BackwardEcho, from your test and description, I think the issue is on the heating side of things, as @PurpleHullPeas suggests. If the heating element and its connections are ok, then maybe somehow power is diminished during the printing process.  

The thermal runaway code will shutdown the printer if the temperature drifts too far from its set point, so the PID controller can come into play here as well. 

One way that the thermistor _may_ be the issue is if it causes the machine to report a temperature that is much lower than the actual temperature. In this way, the printer could be at its limit power-wise and unable to maintain temperature.

BackwardEcho, 2021-04-03:
Any idea what could cause the loss of power during the process? I really can't think of anything aside from a hardware malfunction as `r13`, which works for other people, does not work for me. I do have a 10A psu.

PurpleHullPeas, 2021-04-03:
> Any idea what could cause the loss of power during the process? I really can't think of anything aside from a hardware malfunction as `r13`, which works for other people, does not work for me. I do have a 10A psu.

The heater has to work a bit harder on the first couple of layers, because the air from the fan reflects off of the build plate back onto the nozzle.

BackwardEcho, 2021-04-03:
> The heater has to work a bit harder on the first couple of layers, because the air from the fan reflects off of the build plate back onto the nozzle.

Just tried a `M303` with the nozzle close to the bed on `r13` and `r15`, both failed to heat to temp and caused the error.

aegean-odyssey, 2021-04-08:
I'm not sure if it's a "loss of power", but more a matter of insufficient power (heat). If the heater and thermistor components and their wiring check out, then there's a possibility that the power supply isn't really supplying sufficient power or that some other component (e.g. stepper, fan, ...) is drawing more power than it should. And something mis-wired may be drawing more power even with the printer just idling. I'm fairly certain that the firmware fault you've encountered results from the nozzle not reaching the set-point temperature fast enough.

For comparison, the stock printer here (60W power supply) has no trouble maintaining a 215&deg;C nozzle temperature even with the stock fan running at 100%.

So:

* you can try to heat the nozzle with the original power supply to rule out a power supply issue;

* if you can measure the nozzle temperature independently to confirm that the temperature isn't running ~20&deg;C hotter than the printer displays -- this would be at the limit of the printer I'd say and a fault would be expected -- making the thermistor suspect;

* you can temporarily disconnect fans and steppers and test to rule out these components as drawing more power than they should.

> It's possible, too, that a mosfet (switching transistor) is damaged on the controller board. I'm not up on the likely failure modes for this type of transistor or the power switching circuit -- particulary if it can _seem_ to be working when it isn't, but it is the way power is delivered to the heating element.

BackwardEcho, 2021-04-08:
- Should I install the `-5A` variant while I do or just leave it?
- I cannot, is there some thermometer you'd recommend that I can get from a Home Depot?
- Will try along with everything else in order.

... now I REALLY want gigdigit to get back in stock soon.

PurpleHullPeas, 2021-04-08:
> ... now I REALLY want gigdigit to get back in stock soon.
Tons of stuff has been listed as "out of stock" for quite a long time on Gigdigit. Certain stock parts (original motors) are not available at all anymore. Reddit/Facebook activity is nowhere near what it used to be. The point is, the MPMD is near the end of its support cycle. It is probably just time to look at installing non-stock aftermarket parts. Compatible thermistors, heating elements, and power supplies are all easily found. The MPMiniDelta wiki (ran by the same person who runs Gigdigit) usually lists the specs for the parts, as well.

BackwardEcho, 2021-04-08:
> Tons of stuff has been listed as "out of stock" for quite a long time on Gigdigit. Certain stock parts (original motors) are not available at all anymore. Reddit/Facebook activity is nowhere near [...]

I just want a main board ;~;

But I guess I might have to look into Teaching Tech's upgrade then...

aegean-odyssey, 2021-04-08:
@BackwardEcho, I've no recommendations for a temp probe, and my experience is from a long, long time ago.

Yes, @PurpleHullPeas, I think maybe Monoprice is losing interest its Mini Delta, and there's not really a good low-end delta printer alternative. And the RAMPs board while way more configurable, is probably slower than the stock controller board -- a step backwards in some ways, I think.

It's a shame -- the little delta printer is a lot of fun and (in my mind) still a stand-out in a sea of cheap cartesian printers. Despite its limitations, I think better part cooling and bed leveling would be the incremental improvements I'd shoot for in a new design -- quieter would be nice, too. 

On the firmware side of a new design, porting to another STM32 based board would not be too difficult (most of the work would be related to the UI, I think), and I'd make some improvement as well: replace Marlin's SD card code and file system with something that truly supports long file names; rework the much faster "simple upload" so that the host printer server (e.g. Octoprint) can pre-heat the nozzle and bed _while_ the gcode file is being transfered; and support special features of the (new) controller board. 

I'd hate to see delta printers disappear. Maybe another

aegean-odyssey, 2021-04-09:
Apologies, closed inadvertently during a keyboard flub writing my last entry. Re-opened.

BackwardEcho, 2021-04-10:
So I guess it just can't hold temp when it's close to the bed?

- The original power supply couldn't heat up all the way when the nozzle was close to the bed, same as my 10A
- I unplugged everything but the hotend, the cooling fan, and the switches. Did not help it.
- I barrowed a Milwaukee IR temp gun from a friend at work. It might be off, my aim could suck, or the delta does not get up to requested temp but says it is: the nozzle is roughly -15 degrees than requested. 

@PurpleHullPeas I purchased a thermistor and stock heater cartridge from Gigdigit, but also that 40W version you linked me. Did you have to crimp something on or was there somehow enough wire to reach a plug? I might have to do something extra aside from switching out the black plug (as reviews on Amazon say that black plug can't handle the voltage.

PurpleHullPeas, 2021-04-10:
I replaced the entire heater cable except for a small section near the main board so that I could steal the plug that goes into the board. I soldered it there. I am not experiencing issues printing at 240.

BackwardEcho, 2021-04-12:
So Marlin4MPMD 1.3.3 worked for me in that the nozzle can keep temp when it's close to the bed. So my hardware is fine, may need to upgrade to the 40w if I cant hit abs temp's later, but so far it works.

PurpleHullPeas, 2021-04-12:
> edit: Could it have been the fan? the M4MPMD doesn't seem to spin the fan at insane speeds compared to 1.1.x

Most definitely possible. Try setting M106 S255 and see what happens. 

Also, possibly relevant: https://github.com/aegean-odyssey/mpmd_marlin_1.1.x/issues/41

BackwardEcho, 2021-04-12:
> 
> 
> > edit: Could it have been the fan? the M4MPMD doesn't seem to spin the fan at insane speeds compared to 1.1.x
> 
> Most definitely possible. Try setting M106 S255 and see what happens.
> 
> Also, possibly relevant: #41

The temp dropped a few degrees before catching itself and returning to target temp of 215.

aegean-odyssey, 2021-04-12:
@BackwardEcho, from your tests and observations, it seems you've ruled out all but the heating element or the circuitry that drives it -- I concur with @PurpleHullPeas' take that heating element is a strong suspect. The printer should not struggle to heat the nozzle to 215&deg;C.

On the (stock) machine here with the stock power supply, the nozzle heats fairly quickly to temperatures up to 250&deg;C. The `-05Amp` firmware gives priority to heating the nozzle, so the nozzle temperature stays fairly stable at the expense of the bed temperature.  On the rare occasion, the printer generates a fault when it cannot maintain the bed temperature. With this scheme, it can take a long time (~10min) for the bed temperature to reach its set point, but a thermal run-away fault is avoided as long as temperatures are making progress toward the set points.

BackwardEcho, 2021-04-12:
I just tried the test again but with a temp of 250: it could not regain its target temp and stayed around 239. Should I stick with the stock heater or go with the 40w? I have both replacements already

PurpleHullPeas, 2021-04-12:
I guess I am having a hard time understanding why you are so hesitant to just change out the heater and be happy, like I suggested quite some time ago. Here are some more data points: 

1. I was previously (prior to failure) able to print PETG at 245 C with the stock heater and 10A mpmd_marlin_1.1.x firmware. 
2. My other MPMD's heater failed while still on Marlin4MPMD 1.3.3. 
3. Another trusted member of the community (muffinhead) also had their heater fail while still on Marlin4MPMD 1.3.3. 

So, yes, I have stated before (somewhere here), that I suspect heating errors trigger at a lower threshold on mpmd_marlin_1.1.x, but I do not think that firmware is the "cause" of heater cartridge failure. Based on my past experience, if you continue to print, as is, it is just a matter of time before you will not even be able to reach 200 C. 

I chose 40W for several reasons: I needed to change the heater on a completely different 3D printer around the same time. 40W is a much more common heater cartridge across 3D printing than 30W. I also wanted to reach my target temperatures more quickly.

BackwardEcho, 2021-04-12:
Sorry for my hesitation, it is just that I don't know the exact connetor
name of the heater plug that goes into the mainboard, so if something where
to go wrong with my soldering job, I am screwed until I can find a
replacement. I was hoping the stock heater from gigdigit would have the
section of wire that plugs into the mainboard, but it only goes to the
inline break that I completely forgot existed.

BackwardEcho, 2021-04-20:
Well, my 'soldering hands'(or w/e they are actually called) arrived and I believe I made some decent to nice soldering on the wires and reinstalled the heater along with the new thermister I bought from gigdigit. It's now printing fine. Though if anyone can't figure out why they might have an underextruding problem even though they might have gone through every step to fix it: **suggest they delete their print settings profile and try again**. I've had an underextrude problem for a while outside of this heater problem, even with slow print speeds and high temps for something like PLA. I switched back to a .4 nozzle from a .2 and was about to chalk them both up as partially clogged until I just deleted my settings profile. Now I wonder if I can go back to a .2, but thats for later. For now, I need to get printing.

PurpleHullPeas, 2021-04-20:
I have never had a problem swapping between nozzle sizes. I mostly just change the nozzle size under the machine->extruder settings and the Line Width in Cura. I think your extrusion issue is a good reminder that the common "have you tried printing the Lucky Cat" advice also applies to extrusion issues. It eliminates the slicer from the equation. Lucky for us, aeagean-odyssey includes a set of pre-sliced files for this firmware, as well.

BackwardEcho, 2021-04-20:
I've gotten it to print before with the .2, but for some reason at some point when messing with the profiles something changed and gunked it up. I think I'll just delete the other mpmd profiles I downloaded and just keep two: one for .4 and the other for .2 but both of them would be a copy of a working .4 profile.

BackwardEcho, 2021-05-11:
If it helps anything, I've gotten the 'Thermal Runway' error a couple of times since installing the new heater ONLY after it placed the nozzle near the bed once heated. The temp readout would have it go down to like 204 and then go back up, but sometimes not fast enough.

This does not happen if I have the nozzle pre-heat near the bed before printing.

aegean-odyssey, 2021-05-15:
Interesting. A couple of questions:
* Do you think that the fan somehow is able to cool the nozzle when the nozzle is placed near the bed?
* Does auto-PID tuning for the nozzle (at the target temperature) help the nozzle temperature recover any faster?

BackwardEcho, 2021-05-15:
- I'm thinking that is the case. I just tried to recreate it without a print: it dropped to 210 and then recovered. I let it sit at 215 for a few minutes prior to lowering the nozzle.
- I tuned it after installing the new heater cartridge although I don't think I tuned it with the nozzle close to the bed. I will retune it once I install my silicone sock that is currently curing.

gordsellar, 2021-10-11:
I'm struggling to deal with the E1 error. However, I also can't seem to connect to my printer with wifi (not sure why but it's never worked well on this printer), so I'm dependent on using the calibration files, which have been a godsend. 

I tried to write some setup_gcode files to do this, logging to a file titled AUTOTUNE.TXT, but it's not working. Can anyone tell me what I'm doing wrong?

I figured it would be 2 files:

Here's the first, "PID-Autotune.gcode":

```
M988 /AUTOTUNE.TXT
M106 S200
M303 S210 C10
M989
M118 {E\:Done! (see /AUTOTUNE.TXT)}#
```

And here's the second, which you'd need to edit (at the spots marked with x) to input the values. "PID-Autotune-Save.gcode"

```
M301 Pxx Ix.xx Dxxx   
M500 

```

That is, unless the firmware is set up to automatically store those values and I can just get the same done by adding the M301 and/or M500 commands at the end of the original file? 

Any advice would be very helpful! Replacing the thermistor on my unit would be a huge PITA and I'm trying to get something printed up for my kid this week. 

Alternately, if I've seen some people suggest that they avoided this problem by rolling back to r13, but when I checked it out I noticed it doesn't contain the (invaluable) setup Gcode files. Would I be able to use the setup files from r15 with the firmware from r13, if need be?

Thanks!

gordsellar, 2021-10-11:
Thank you so much! 

1. Ah, ok!

2. I couldn't get USB to work. Not sure if the fact I'm using a Mac is the issue. 

3. That's helpful. Thanks!

4. Weird! The output file ("AUTOTUNE.TXT") is completely blank when I run the first code. 

If I set it to the following, would it work?

```
M106 S200
M303 S210 C10 U
M500 

```

Or maybe if I added that to my startup code? I don't mind the startup taking a little longer. 

5.  I'm using one of the start codes that were posted here:

```
G28
; set and wait on hot end temperature (Set nozzle to just below PLA melting point)
M104 S155 T0
M109 S155 T0
; set and wait on bed temperature
M140 S55
M190 S55
; set and wait on hot end temperature (Finish raising the hot-end temp to set value)
M104 S{material_print_temperature}    ;Start heating extruder
M140 S{material_bed_temperature}      ;Start heating bed
M109 S{material_print_temperature}    ;Wait for extruder to reach temp before proceeding
M190 S{material_bed_temperature}      ;Wait for bed to reach temp before proceeding
; home axes, probe/adjust z-offset, and pause 4s
G29 P0
G0 X0 Y0 Z60 F3600 ; [DV: added feedrate]
G4 S4
; extrude a strip outside of the perimeter
G92 E0
G1 X-54 Y0 Z0.32 F2700
G3 X0 Y-54 I54 E20 F900
G92 E0
```

That said... now I'm looking at it and thinking maybe I'm supposed to edit those last few lines?

6–8. Yeah, I probably need to upgrade to a better power supply soon. That should be doable, at least. Getting a cartridge would be a PITA where I live (either slogging through local webpages in a foreign language, or ordering from abroad), but maybe I'll just have to bite the bullet.

9. No worries, and thank you so much! It's not a life-or-death deadline, at least!

Weberowsky, 2021-11-28:
I have same error.
My GCODE:
`;This is a long one. The purpose of this script is to eliminate the tiny blobs and strings at each probing point. If you a G29 command that probes 3 or 4 points then don't bother with this script. The 25mm purge is so when the G29 command moves the effector plate down the cooled string of filament will eject out the side (doesn't always work).
G90 ; switch to absolute positioning - this line was added in case Cura doesn't include it by default
M82 ; set extruder to absolute mode - this line was added in case Cura doesn't include it by default (may not be needed)
G92 E0 ; reset extrusion distance
G1 E25 F300 ; purge 25mm of filament to prime nozzle.
G92 E0 ; reset extrusion distance
G4 S5 ; Pause for 5 seconds to allow time for removing extruded filament
G28 ; start from home position
G1 E-6 F900 ; retract 6mm of filament before starting the bed leveling process
G92 E0 ; reset extrusion distance
G4 S5 ; pause for 5 seconds to allow time for removing extruded filament
G29 P4 Z0.38 ; 4x4 mesh (12-points) with offset - YOU MAY NEED TO ADJUST THE Z OFFSET EXAMPLE: change "Z0.38" to Z0.45 or Z0.2
G1 Z60 ; raise Z 60mm to prepare for priming the nozzle
G1 E4 F300 ; extrude 4mm of filament to help prime the nozzle just prior to the start of the print
G92 E0 ; reset extrusion distance
G4 S5 ; pause for 5 seconds to allow time for cleaniing the nozzle and build plate if needed`