aegean-odyssey/mpmd_marlin_1.1.x #76: FAN needs support

BackwardEcho, 2021-05-24:
@aegean-odyssey , Matthew Upp, whom I'm sure you know runs the Gigdigit store, has populated the empty FAN header on this recent batch of quiet boards. I just got my board in the mail and was pleasantly surprised by it. Did you see the post, did he contact you at all on it, or is this the first time you are hearing about this?

[the post in question](https://www.facebook.com/photo.php?fbid=10215846042264335&set=p.10215846042264335&type=3)

aegean-odyssey, 2021-05-27:
No, hadn't heard. I'm curious as to what I/O pin might control the FAN (if any). If it can be controlled by the CPU, then I think it's worthwhile to incorporate it into the firmware somehow.

BackwardEcho, 2021-05-27:
I am willing to help out as much as I can to help you find out which I/O it is. If anything, components in spaces R41, R42, and Q3 were installed. So maybe if you were to find what I/O uses R1, R2, and Q2, you might find the missing link nearby.

aegean-odyssey, 2021-06-02:
Glancing at a photo of the circuit board, I _think_ the circuit is a typical mosfet switch. One of the resistor traces should make its way back to the cpu where the dot on the CPU marks pin# 1. Pretty much all of the I/O pins are in use, though, I believe.

#### Digital I/O
[image]

#### CPU Pinout
[image]

BackwardEcho, 2021-06-02:
"In use" as in every pin is attached to something or as in you know it
activates something? Because we can try all 48 pins the hard way if you
want just to find it.

mulcmu, 2021-06-02:
Looks to me like PA3 is setup to control Q3 as low-side switch.  PA2 & PA3 are connected to the unused serial pads by the MCU, so PA3 should be "free" to repurpose without impacting anything else. 

[image]

Here was another picture of the PCB area in question. 
[image]

@BackwardEcho Could you post a close up picture of your installed Q3?  Is it installed upside down?

BackwardEcho, 2021-06-03:
@mulcmu [here](https://scontent.fphx1-1.fna.fbcdn.net/v/t1.6435-9/186553502_10215846044224384_6404020170173820467_n.jpg?_nc_cat=108&ccb=1-3&_nc_sid=dbeb18&_nc_ohc=g7qf8uh8Gc8AX9JgNBq&tn=KsJ_8L7QieblDa5g&_nc_ht=scontent.fphx1-1.fna&oh=c84c31da3256e3963357ecd5befb9346&oe=60DC969C) is a photo Matthew took when he was talking about it. It is leagues better than what my phone can take.

BackwardEcho, 2021-06-07:
[image]

I guess here is the photo for Github to have for itself.

aegean-odyssey, 2021-06-17:
I _think_ this can work: `PA3` allows for a PWM output (still checking on this) as an alternate function for the pin, so it shouldn't be too difficult or too much code to add another fan (as `P1`) to g-codes, `M106` and `M107`. _[see `M106`](http://marlinfw.org/docs/gcode/M106.html)_  

Does this sound like a reasonable way to use the output? Is there another approach that is prefered?

BackwardEcho, 2021-06-17:
I don't see why not. Is `P1` the normal "part cooling" index for `M106`?

BackwardEcho, 2021-07-02:
Installed a y splitter connected to that new port in anticipation.
[image]

aegean-odyssey, 2021-08-10:
Yes, Marlin G/M-code docs for `M106` and `M107` specify that the default fan is always the part cooling fan, but the documentation is vague about any particular numbering scheme for the fans.  So, for the `mpmd_marlin_1.1.x` firmware, here's the plan for `M106`, `M107`:

*  option `P0` will always select the orginal fan output;
*  option `P1` will always select the additional fan output;
*  in the `-ACfan` variant of the firmware, `P0` will be selected when no `P` option is supplied;
*  in the `-PCfan` variant of the firmware, `P1` will be selected when no `P` option is supplied; 
*  in both variants, the original fan output (P0) will use the `-ACfan's` "automatic" mode.

I think this works. The original behavior of the `-ACfan` firmware doesn't change, but `P1` will be supported. And in the `-PCfan` firmware, the behavior will match a typical two-fan setup. 

I'll get to work on it. Please point out any glaring omissions.

BackwardEcho, 2021-08-10:
What about those that don't have the new fan output? I feel like the `-ACfan` variant should be the only version to get this new code as it wont affect anyone's current set up including mods. It was only the recent batch of boards that actually got this new header installed so I don't think many people will have it. If I understand your plan, changing how the `-PCfan` variant works to use the new output rather than the original output would force the user to have the board with the new output installed.

aegean-odyssey, 2021-08-16:
I see your point. My approach does seem overly complicated. Rather than try to accommodate some future theoretical 2-fan configurations, perhaps it's better to just add the additional fan output as explicitly controlled `P1` with `M106`, `M107`. The additional fan ouput would work the same way in both `-ACfan` and `-PCfan` variants of the firmware and original behavior is preserved for both. I'll get started on the groundwork code changes while we ponder this some more.

BackwardEcho, 2021-08-16:
So the original fan output will be labeled explicitly as `P0` and the new output will be explicitly `P1`. The only difference will be that the `-ACfan` will have  `P0` controll itself while the `-PCfan` will have `P0` be controlled by the user. So the choice would be "1 automatic fan and 1 controllable fan" or "2 controllable fans".

Correct?

aegean-odyssey, 2021-08-17:
Yes, correct. Recall, though, that in the current `-ACfan` firmware, the single fan output (`P0`) can be controlled or automatic (setting the speed to zero enables the automatic mode).  The added `P1` fan output would only be controllable.

I'm hung up on what `M106` and `M107` use for the default fan output when no `P` option is supplied. The spec calls for selecting the __part cooling__ fan when the `-P` option is omitted.  The original fan output (`P0`) in both the `-ACfan` and `-PCfan` firmware would now be considered the part cooling fan.

For the `-PCfan` firmware, I'm thinking that it might be nice to have the added (`P1`) fan output function as a combination controlled and automatic fan control. With this flexibility, if a g-code file doesn't include an `M106 P1 S...` command, the `P1` fan output would be automatically controlled; the typical approach to cooling a hotend. Using `M106 P1 S...`, the `P1` fan output is PWM controlled; and perhaps suited for some extra type of part cooling application.  

Anyway, first step is to get the new fan output working.

BackwardEcho, 2021-08-18:
I feel like I'm letting you down by not being able to follow your train of though on this. Sorry >_>

In an attempt to: are you hung up on how to designate the original header for a "hotend" or "part cooling" function since the original fan is both? Why not have `P1` copy what the `P0` does until explicitly told to do otherwise? Are you able to differentiate between "automatic control" and "user control"? (Like have `P0` do it's thing for the hotend alone, but when prompted with a `M106 / M107` they both act the same without the `P` function) Would this do something funny to the boards that don't have this new header by activating an incomplete circuit? 

>...function as a combination controlled and automatic fan control.
I can't wrap my head around this, though I feel like I might have touched on it in my questioning above.

>Anyway, first step is to get the new fan output working.
Let me know when you have a test build ready, since I might be the only active one here with a applicable board.

PurpleHullPeas, 2021-08-18:
I have not been following this thread intently, but I think it is important to look at the end goals. 

a) Stock Setup (AC Firmware). The fan will automatically turn on after a temperature threshold, but you can [definitely override this with gcode commands](https://github.com/aegean-odyssey/mpmd_marlin_1.1.x/issues/41). So, it is still controllable, but it is not advised to turn it off due to heat creep.

b) Part Cooling Mod (PC Firmware): Hardwire the heatsink fan into the "always on" 12V to avoid heat creep, wire part cooling fans into the PWM connector for control. 

c) New PWM Fan Header (PC Firmware). Same as b, except you plug the heatsink fan into the new PWM header so that you can turn it off when not printing. You could also probably wire up some lights so they turn on/off with the print. I think this one applies to @BackwardEcho 's setup.

d) New PWM Fan Header (AC Firmware). You have a stock-ish effector setup, perhaps with an upgraded fan, but you have also added external fan(s). See "tube cooling" or the "external fan" photos in [my part cooling tutorial](https://docs.google.com/presentation/d/1woW02wV4u91A42hoTMOYW4DB-7hkvlRo/edit?usp=sharing&ouid=115707042492049565044&rtpof=true&sd=true). So for one of my setups, I would no longer need a potentiometer to control my external fans. 

So, to me, it makes sense to make the new fan controlled via gcode, like mainline Marlin [M106](https://marlinfw.org/docs/gcode/M106.html) with the fan indices.

BackwardEcho, 2021-08-18:
I agree with Purplehue in that it's a good idea to know what you want before you really go into the changes. I'm thinking it'll make things easier as the changes are made. Not to mention those listed options are more or less what I was hoping for, even if I wasn't able to properly convey it.

mulcmu, 2021-08-18:
> I'm hung up on what `M106` and `M107` use for the default fan output when no `P` option is supplied. The spec calls for selecting the **part cooling** fan when the `-P` option is omitted. The [...]

I wouldn't get too hung up on this.  I think `P0` as the default works well.  Anyone implementing these changes could swap the fan connections to match the firmware.  I assume modern slicers would always output the fan index and not rely on defaults.

aegean-odyssey, 2021-08-19:
Thank you all for the comments. I agree that my original draft is too involved for little gain -- the "default" fan should always be `P0`.  I was also trying to accommodate @PurpleHullPeas' nicely  laid out use cases, in particular case d (part cooling fan + automatic fan). And since this new fan I/O is currently configured as part of an auxillary serial port, I'd like to code this in a way that preserves the serial port and the myriad of fan options (likely in the form of compile-time switches).

I wasn't too concerned about this type of issue earlier because there was little program space remaining for firmware changes. Building the `mpmd_marlin_1.1.x` firmware with the latest compiler and libraries though freed up enough space to make it more feasible to customize the  firmware for specific applications. 

So I think it makes sense to "code in" a flexible configuration to help those who want to customize, but as far as what to include in the `-ACfan` and `-PCfan` variants, I'm leaning toward a simple controllable fan designated as `P1` for the `M106`, `M107` codes. I'm open to suggestion, though. My only concerns are not to adversely affect the firmware for the stock printer, and not to create something that becomes a support problem.

Thoughts and comments welcomed.

PurpleHullPeas, 2021-08-20:
@aegean-odyssey I completely agree. I think the fan header mod is (and will be) so rare that those who do it can just setup their slicer to use P1.

BackwardEcho, 2021-08-20:
Is there a way to set a given slicer to use a specific designation for a fan? I thought it would be set to some marlin default. Not like the "if omitted, use this" type of default, but like the "spec marlin has the hotend fan as `P0` and part cooling fan as `P1`, so specify in gcode as such" default.

PurpleHullPeas, 2021-08-20:
@BackwardEcho In Cura, you can set "Machine Settings"->"Extruder 1"->"Cooling Fan Number". 
Alternatively, you can plug the part cooling fans into the stock cooling fan port and then control the heatsink fan in your Start/End Gcode.

BackwardEcho, 2021-08-21:
> @BackwardEcho In Cura, you can set "Machine Settings"->"Extruder 1"->"Cooling Fan Number".
> Alternatively, you can plug the part cooling fans into the stock P0 cooling fan port and then control the heatsink fan via the new P1 port in your Start/End Gcode.

Oh sweet, I didn't know that. Thank you. Personally I'll stick with the automated approach.

BackwardEcho, 2021-08-22:
For the sake of checking what cura generates with part cooling, I got `M106 S255` when that "Cooling Fan Number" was set to the default `0`. I got `M106 S255 P1` when it was set to `1`.

aegean-odyssey, 2021-08-22:
I think I've got all of the changes in. It looks though that the Travis-CI (automatic build process) is somehow buggered. Waiting on support to try to sort it out. And I need to do a little testing here before asking for testers that have hardware for a working extra fan output. Thanks for your patience.

BackwardEcho, 2021-08-23:
Are you gonna be making all 16 variants in this one run? If it cuts down on how many you need to make, I believe the one I use is the `0001` version.

aegean-odyssey, 2021-08-23:
I'm still futzing with the travis-ci builds. 

In the meantime: Completely untested, but compiles -- you can find two build runs under the ___Actions___ tab of this repository. Run \#9 is `-PCfan`, \#8 is `-ACfan`. If you are logged in to Github, then in each run you should be able to peruse the "build" log (only the "identify" section is useful) and download the "artifact" -- a zip file that contains the firmware.

If you give the firmware a try and nothing works, don't spend too much time with it -- I really did not test at all. `M106 P0|P1 S<speed>` and `M106 S<speed>` should now work, and is a good starting point. Also the PWM control for FAN1 (`P1`) is separate code (timer15 of the cpu), so even if `M106` recognizes `P1`,  I may have not have speed control working yet.

BackwardEcho, 2021-08-23:
I was not able to get the fan connected to the new port working. Granted, I didn't open up the printer to plug it directly in (but I did switch the contacts on one of them just to be sure the current wasn't reversed for some reason), instead relying on my Y splitter that I made and installed. I tested the splitter before and it worked, so if it somehow broke during a print, I will find out later when you have a build you want tested.

aegean-odyssey, 2021-08-23:
@BackwardEcho, thanks for giving the 1st round a try. 
* Did `M106` recognize `P0`, `P1`?
* Did the original fan still work (with `P` missing and `P0`)?

And to cover all bases, do we know that the hardware works (and that `PA3` is the control)? In the photo of the main board, the __round__ pad of `J20` should be `PA3`, the io controlling the extra fan output. With firmware `119r15` (not the test versions of firmware), 5v (through a resistor, ~150&Omega;) on the round pad of J20  _should_ turn on the extra fan output. It's a test, but not all that important as we'll discover the answer as I produce better tested firmware.

[image]

Again thanks for testing, and don't spend too much time with it. It's unrealistic to think that I got all things correct on the first, completely untested attempt. I'll try to find the equipment to test the firmware to see if it is generating the expected PWM signal on `PA3`, and actually shake-out the firmware a bit.

mulcmu, 2021-08-23:
I opened up my machine and double checked the schematic I posted above.  `PA3` goes to the round pin on `J20` and to the screw side pad of `R41`.

I had previously hooked a scope up to `PA3` with the `119r15` release firmware installed.  There was a very brief pulse at startup.  Might be long enough for @BackwardEcho to see a fan shift or LED light up when MPMD is first powered.

BackwardEcho, 2021-08-24:
Here are 2 photos of my board with my new phone's macro camera.

[image]
[image]

Gimme me a minute to edit this on my computer after uploading those photos

aegean-odyssey, 2021-08-24:
Thank you all for the testing assist! It real helps narrow things down, and I _think_ I found the problem: I'd removed a line of (initialization) code that actually enables timer15-ch2's pwm output signal. For some reason yesterday's read of the tech specs led me to _deliberately_ remove the line as not required for timer15 -- but  it is and it is now back in the code.

@mulcmu, the firmware initializes all of the io at startup (after it gets control from the boot loader), so the state-change on the signal lines is probably related to the delay before the firmware gets to set the io to a known state. Also, both `-5Alimit` vs `-10limit` should work without a problem. 

I've built two more test versions, runs &sharp;10 and &sharp;11, under the __Action__ tab. Please feel free to test. Hopefully the `FAN1` output will be working now.

BackwardEcho, 2021-08-24:
Same result :(

mulcmu, 2021-08-25:
I soldered wires to the R41 and R42 resistor pads to easily hook up to the scope.  I tried builds ♯8 and ♯10 with a scope hooked up.  Getting a constant 3.3V.  No change in the waveform for `M106 P1 S##`

mulcmu, 2021-08-27:
Here were my best guesses:

FAN:  B2B-XH-A through-hole type shrouded header

Resistors look to me like 0603 package size
R42:  10kΩ
R41:  24Ω would match R1 but it is an odd ball value.  100Ω should work fine.

Q3: AO3402 (A29T) (SOT-23 package) **mounted upside down!**

BackwardEcho, 2021-08-28:
I went ahead and @'d Matthew on FB, hopefully we'll get an answer later. I also sent an email via gigdigit and a PM telling him about the upcoming release and suggesting he make some new skus on the store page for an installation service (send your board in) because I think people are gonna want it. Before he told people to contact him via gigdigit to work it out, but adding a sku will make it obvious.

BackwardEcho, 2021-08-28:
@aegean-odyssey were you able to get a confirmation regarding the components? I see you added them to the wiki already.

MatthewUpp, 2021-08-28:
Apologies for not chiming in sooner. I didn't know this thread existed since I had only been searching the discussions tab to see if a thread for the fan addition was created. After reading the Facebook post and the email from @BackwardEcho, I did some searching and finally found this thread in the issues section.

Anyways, the components that are in the table on the [wiki page](https://github.com/aegean-odyssey/mpmd_marlin_1.1.x/wiki/Extra-Fan-Output#parts-list) are pretty much correct. The two resistor values I chose to use are based on the values that the printer manufacturer uses on the M300 and M200 mainboards. 
For the MOSFET, if AO3402's are available then that will work. A few boards have AO3042 installed on the new header but others have either AO3434A's or a couple other equivalent MOSFET's. At the time there was very few available at Digi-Key and the few other larger component warehouses that I purchase through. They are all rated between 28-30V and 3.8-4.0A. A little lower or higher current values would likely also be fine to use but I wanted to keep the values the same or close to stock values.
Also, just confirming that the MOSFET does indeed need to be installed upside-down.

Since the PCB design for the additional fan is for whatever reason not correct, I created this image is help me visualize what needs to go where. Not really needed for anything here but figured I'd share anyway.
[image]

BackwardEcho, 2021-08-28:
This actually might be very helpful to those who wish to add the components themselves instead of contacting you at Gigdigit. Obviously it might be very few who are comfortable doing it, but the documentation on it is extremely valuable nonetheless.

aegean-odyssey, 2021-08-28:
Thanks @MatthewUpp for confirming the parts list and layout. Is GigDigit going to sell a parts kit, or offer a service to upgrade boards, or sell upgraded controller boards? If so, I'll mention it in the wiki page.

BackwardEcho, 2021-08-28:
@aegean-odyssey He added the parts during his last batch which prompted this whole thing. Since it works I can see it being added going forward for new boards. The other options (parts kit or service) is what I'm thinking people would be interested in; to know it is available. But for the service some things will probably need to be shipped to the customer beforehand in order to get the board safely to him for modification (ie, an anti-static bag, because who is going to readily have one).

BackwardEcho, 2021-08-31:
Is anyone able to add that Gigdigit schematic to the [`FAN1` wiki](https://github.com/aegean-odyssey/mpmd_marlin_1.1.x/wiki/Extra-Fan-Output)? I think it's a good representation on how to orient the components on the board and thus should be added. I'd do it but I am not familiar with anything related to GitHub outside of making simple posts and I don't want to break anything in an attempt to make it match the format.

closed 2022-03-31 by BackwardEcho