aegean-odyssey/mpmd_marlin_1.1.x #77: Control FAN

Electromente, 2021-06-09:
Hi guys, 

Im using the 119r15 version and works excellent, the only issue is that Im trying with ABS and in many webpages recommended to put the heater fan off or set it at min value.

I'm using Cura slicer and set the Fan to 10% but looks like dont do anything.

Could you let me know if this setting really works or this printer cannot control it? 

as far as I remember, the Stock firmware dont took this configuration either.

Thanks for your great work.

PurpleHullPeas, 2021-06-09:
You do not want to do that with the stock MPMD hot-end. Rotate the fan shroud upside down, so that the blower is point upwards into the air instead of on the print. 

The articles you have been reading are referring specifically to 3D printers that have [more than one fan on the hot-end](https://all3dp.com/2/3d-printer-fans-do-i-really-need-one/). One fan is typically used to cool the heatsink and heatbreak - in order to prevent clogs due to plastic melting into the throat (heat creep). The other fan, called a _part-cooling fan_, is used to cool the plastic as it comes out of the nozzle. The ABS recommendations are for turning off the _part-cooling_ fan, not the heat sink fan. 

The stock Monoprice Mini Delta has only one fan that serves both of those purposes. For ABS, you can either block the bottom of the fan duct from blowing on the nozzle or turn it upside down.

Electromente, 2021-06-09:
Thanks to clarify it, I'm printing now in ABS with the normal position. Then I'll turn it upside down and print the same example to compare and let you know.

Electromente, 2021-06-11:
I've made the change to the heater fan, the final piece is very hard and solid, but now I'm trying different temperatures to don't have small parts with spilled ABS hehe. Anyway, is a very good approach may be the critical parts printed without Fan and the smaller or so detailed pieces printed with Fan enables. (or just move on to PLA hehe)

[image]
[image]
[image]

Electromente, 2021-06-11:
Speaking about the details have no major differences between both Fan positions. just you must use middle or lower temperatures as ABS Support to avoid spilled of ABS.

closed 2021-06-11 by Electromente