aegean-odyssey/mpmd_marlin_1.1.x #78: Question about M92

Electromente, 2021-06-18:
I have a question regarding the M92_XYZ57_E48.gcode , I know my printer is older than 2018, and when I flashed with factory firmware, always review if my printer is 1/8 or 1/16, but once started to use Marlin I've never have corrected the microsteps again, (because I've forgot it).

What is the behavior or improvement that I should observe when using the correct parameter?

The question is related to understand what I should see/review to know if it has gotten better or worse something in the printer after apply it.

NOTE: Now the printer is working almost fine with a little leveling issues in the most external side of the bed.

BackwardEcho, 2021-06-18:
The wrong `M92` would be really obvious if you choose the wrong option. It would either go to fast and jam into the build surface when doing a `G29` probing, or it wont even get there. Any improvement would come from using `114.28` or `114.29` for dimensional accuracy. Something to do with how the firnware rounds numbers giving us the `114.28`. You can get the height of a print to confirm if you've already adjusted the XY accuracy another way.

aegean-odyssey, 2021-08-09:
@BackwardEcho provides a good answer. Converting this issue to a discussion and closing.

closed 2021-08-09 by aegean-odyssey