aegean-odyssey/mpmd_marlin_1.1.x #80: Pronterface Serial Exception

aegean-odyssey, 2021-08-09:
I am getting this issue from Pronterface. I think it might be related to the buffer size or the chip quality, but I may be wrong.

> Exception in thread Thread-2:
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/serial/serialposix.py", line 500, in read
    raise SerialException(
serial.serialutil.SerialException: device reports readiness to read but returned no data (device disconnected or multiple access on port?)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/printrun/printcore.py", line 260, in _readline
    line = self.printer.readline().decode('ascii')
  File "/usr/lib/python3/dist-packages/serial/serialposix.py", line 509, in read
    raise SerialException('read failed: {}'.format(e))
serial.serialutil.SerialException: read failed: device reports readiness to read but returned no data (device disconnected or multiple access on port?)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3.8/threading.py", line 932, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.8/threading.py", line 870, in run
    self._target(*self._args, **self._kwargs)
  File "/usr/lib/python3/dist-packages/printrun/printcore.py", line 347, in _listen
    line = self._readline()
  File "/usr/lib/python3/dist-packages/printrun/printcore.py", line 281, in _readline
    if 'Bad file descriptor' in e.args[1]:
IndexError: tuple index out of range

_Originally posted by @nathanielstenzel in https://github.com/aegean-odyssey/mpmd_marlin_1.1.x/issues/5#issuecomment-894618440_

aegean-odyssey, 2021-08-09:
I think this portion of the error message, _"device disconnected or multiple access on port"_, hints at the root of the issue. I'll try to reproduce the problem on this end. Btw, what firmware version is installed in the minidelta printer?

nathanielstenzel, 2021-08-09:
[image]

aegean-odyssey, 2021-08-15:
In firmware relase `119r15`, we (credit goes to the testers) fixed a USB issue that caused the printer to hang and reboot during a few connection scenarios with a host computer. Pronterface's connection process may be one of these troublesome scenarios.

I ran a quick test with pronterface and `119r15` firmware:
* On a low-end laptop, I  installed pronterface on Debian Bullseye (linux) directly from the Debian repository; 
* connected laptop and printer via a USB cable;
* turned on the printer, launched pronterface;
* clicked on the "connect" button in pronterface.

Pronterface worked without issue. It controlled the printer (move, turn on heaters and fan,...) and read status without a problem. I was able to manually send G/M-code commands as well. When I turned off the printer (removed power), pronterface displayed the error message you reported. 

Please give firmware `119r15` a try, and please report your results.  I _think_ it will fix your issue. If not, I'll see if I can more accurately replicate your issue here.

Thank you for using `mpmd_marlin_1.1.x`.

nathanielstenzel, 2021-08-15:
Thanks. I will try that in a few days.