aegean-odyssey/mpmd_marlin_1.1.x #86: Bizarre Behavior when Calibrating

gordsellar, 2021-09-24:
I'm not sure what I'm doing wrong, but when I try install the latest version of the firmware on my MP Mini Delta, I get to the calibration stage and when I run AUTO_CALIBRATE.gcode, the printer tries to jam the extruder down and to the left (where I'm facing the panel of the machine with the display screen head-on). It jams it into that corner really hard! 

I've looked in the issues (the ones I can see) and don't see anyone describing that. I *can* see that there's a possibility I need to configure things differently based on it being a very early model, but I'm not sure that's the core of the problem at the moment.  

I've tried both -SM0001 and -SM1110 and the printer doesn't seem to be able to Home after either. Not really sure how to proceed: any suggestions?

mulcmu, 2021-09-24:
I had a similar crash that was caused by some old start gcode logic that used `G91` to enable relative positioning but didn't execute `G90` to restore absolute positioning.

Was printer just upgraded from stock firmware or was a prior version of `mpmd_marlin_1.1.x` working previously?

[Wiki](https://github.com/aegean-odyssey/mpmd_marlin_1.1.x/wiki/Which-Firmware%3f) has some steps for determining which variant.  Older machines likely need to run the `M92_XYZ57_E48.gcode` to load the 1/8 stepping values.

gordsellar, 2021-09-25:
Ooops, I missed that note about older printers and needing to run M92_XYZ57_E48.gcode. That seems to have been the issue (I misunderstood the function of the code), and it's calibrating now. Whew! Thanks so much!

closed 2021-09-25 by gordsellar