aegean-odyssey/mpmd_marlin_1.1.x #91: Max bed temperature.

Weberowsky, 2021-11-28:
Hi, why is max bed temp limited? I cant set it above 60 >:C

rgrobe, 2021-11-28:
I regularly print at higher bed temperatures, so it´s not a limitation of the firmware.

However, I assume that you are referring to the printer´s user interface, where the "preheat" function is indeed not allowing more than 60°C.
That is because this user interface (and also the printer´s rudimentary web connectivity) is handled by a separate board with its own CPU and firmware, linked to the mainboard via a serial link.
This subunit is not changed at all with the mpmd_marlin firmware (in fact, the data exchange that originally supported the web interface is now discontinued to save programming space for other functionalities).

Of course, it would be great if you could take on the challenge of reverse engineering and improving this user interface firmware.
If I remember correctly, there is an option to load a new UI firmware using the web interface, Good luck!

PurpleHullPeas, 2021-11-28:
Set the higher temperature in Cura or through terminal. rgrobe did a good job explaining "why".

closed 2021-11-28 by Weberowsky