aegean-odyssey/mpmd_marlin_1.1.x #94: PID tune typo?

BackwardEcho, 2022-02-17:
`PID Autotune finished! Put the last Kp, Ki and Kd constants from below into Configuration.h`
`#define DEFAULT_Kp 32.84`
`#define DEFAULT_Ki 1.53`
`#define DEFAULT_Kd 176.22`
`ok P63 B7`
`echo:Unknown command: "MM105"`

Just finished a PID tune and it occurred to me that `MM105` is a typo, as `M105` is an [actual command](https://marlinfw.org/docs/gcode/M105.html) related to temperature reporting.

aegean-odyssey, 2022-05-03:
I spent some time looking at this bug, and it seems some kind of miscommunication _after_ the PID command finishes. Possibly, the input buffer is garbled (the initial "M" maybe hanging around) or the command is improperly sent from the controlling side. Not quite sure how to proceed on this one.

BackwardEcho, 2022-05-03:
It's not all that important as it doesn't prevent the printer from printing, so no need to proceed if you can't figure it out. But I do feel stupid for not thinking pronterface could be sending that bad command. I mean, it does say "echo:" what should be a reply from the printer, right?

closed 2022-05-03 by BackwardEcho