aegean-odyssey/mpmd_marlin_1.1.x #96: Printer pauses midprint using octopi

THE-BOW, 2022-04-28:
I have an issue printing whatever from octopi.
Printer starts to print over serial without issues (prints fast without stuttering) 
Suddenly the printer pauses and stays for a couple of seconds then it goes on.
Might be caused by small buffer size...

BackwardEcho, 2022-05-01:
Are you on `r16`? I'm having a issue with serial prints myself. I tried to print a couple of tests prints only to have it pause after a while. I don't know if it's because of pronterface or the printer itself, but I just stick with SD card printing no matter what now.

THE-BOW, 2022-05-01:
Thank you for replying.
I switched  to upload directly to SD card in Cura. Takes some seconds more but prints without pausing or stuttering. The SD Upload takes maybe 5 to 10 seconds.

aegean-odyssey, 2022-05-03:
Printing directly from an SD card does eliminate sources for potential problems, and I think it's the best solution.

As far as printing from Octopi, I do know of one user whose four MPMD printers print exclusively via Octoprint on a Raspberry PI. That said, there is a fair amount of cpu overhead when the printer communicates via usb. It's not a buffer size problem, but more of the motion control and usb communication require a high priority when either demand the cpu's attention. Either activity can bog down the cpu.

As I recall, OctoPrint is pretty good at syncing up with the printer to provide g-code commands at a rate to match the printer. The firmware enables Marlin's extended handshaking for this purpose, as well. If the communication link is solid and the OctoPrint side of things is able to keep up (it problably is) and not the cause the pauses, then the printer's cpu is likely "maxxed" out. 

One way to reduce the load of the motion control is too reduce the "segments per seconds" that the firmware tries to generate when computing a path. This is specified in the `S` parameter of the `M665` command.  A number to high and the printer's cpu is bogged down computing line seqments for each motion. Too low and the motion becomes a bit sloppy. I chose the default value, 200, after some testing, but I can't say for sure that it's the best value to use -- it seems to work.

BackwardEcho, 2022-05-03:
If the problem is `M665 S###`, then this was mentioned before on the roadmap guide, [page 41 and 45](https://drive.google.com/file/d/1RYnJSEOnuf3h62FwhQ2lCmzbucvMLhy7/view). 41 specifically mentions octopi printing.

THE-BOW, 2022-05-03:
Thank you all. 
I will give it a try to change the values.  If not i will stay with the "Upload to SD Card option" 
Think we can close the issue ;-)

closed 2022-05-03 by THE-BOW